MichaIng/DietPi #1769: dietpi-update lost connection during first setup, considered update as valid

s0r00t, 2018-05-13:
#### Required Information:
- DietPi Version : v6.6
- SBC Device : Raspberry Pi 1 B (512MB)
- Power supply used : Samsung phone adapter
- SD card used : Lexar 4GB
- Distro : Stretch

#### Expected behaviour:
dietpi-update should not consider the update as valid when one of its downloads failed.

#### Actual behaviour:
dietpi-update considered the update done and rebooted.

#### Steps to reproduce:
Run DietPi for the first time.

#### Did you submit a ```dietpi-bugreport```?
ID is cbe4662e-2484-41f6-8b1c-ec3d15c09b12-0.

Fourdee, 2018-05-13:
@s0r00t 

Many thanks for the report 👍 

I'll take a look.

Fourdee, 2018-05-13:
Update appears to have run fine, no additional packages were required to be updated:
```
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
Although it does appear to be missing the start of the update log.

First run log is also fine, no errors.

---

There is an issue with ethernet dropping connection:
```
[   45.187186] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x41E1
[   54.670968] smsc95xx 1-1.1:1.0 eth0: link down
[  125.785434] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x41E1
[ 1200.097874] smsc95xx 1-1.1:1.0 eth0: link down
[ 1205.916178] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x41E1
[ 1214.745919] smsc95xx 1-1.1:1.0 eth0: link down
[ 1290.204240] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x41E1
[ 4424.410134] smsc95xx 1-1.1:1.0 eth0: link down
###continued
```

This will cause problems during installation, at stages where LAN/WAN connections are in process.

This could be due to either:
- Insufficient power from PSU
- A hardware fault on the RPi itself.

@s0r00t 

I believe the issue is insufficient PSU used on the device. Phone chargers are not designed to provide a stable 5v under load. Please re-write the image and boot again, using a 5v/2A power supply (official RPi PSU is recommended).

Fourdee, 2018-05-13:
@s0r00t 

In DietPi we check the connection prior to running updates and software installs. On a stable connection, this will ensure required data is available for download.

However, should the network drop randomly after the connection tests, it will cause issues and it is not fee-sable for us to constantly monitor the connection, post connection test.

s0r00t, 2018-05-13:
@Fourdee Ethernet was not connected to the Internet, I run both Wifi & Ethernet on my Pi (it acts as a wifi-to-lan bridge).

Fourdee, 2018-05-14:
@s0r00t 

> I run both Wifi & Ethernet on my Pi (it acts as a wifi-to-lan bridge).

As DietPi will only enable either WiFi or Ethernet based on ```dietpi.txt``` settings:
- How are you achieving this dual mode setup, which configuration file + settings used?
- Did you enable WiFi in ```dietpi-config``` 
- Have you tried running the initial setup with only WiFi or Ethernet connected?

s0r00t, 2018-05-16:
Sorry for the lateness of my reply. Not quite used to my new git mail...  

I just enabled both Ethernet & Wifi in dietpi.txt, and followed [this](https://www.raspberrypi.org/forums/viewtopic.php?t=132674).  
Yes, I did.  
In fact, I ran the setup without Ethernet, only on Wifi.

Fourdee, 2018-06-29:
Marking as closed as we have been unable to assist in this case. Is a 2 year old custom script not maintained by us, outside of DietPi scripts/code.

closed 2018-06-29 by Fourdee