MichaIng/DietPi #1770: Can diept  support Amlogic  905D?

zzlinwq, 2018-05-13:
## Creating a bug report/issue:

Can dietpt  support Amlogic  905D?
上海斐讯公司出的一种设备 N1，处理器是 Amlogic  905D。Can dietpt  support Amlogic  905D?
#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)?
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

zzlinwq, 2018-05-13:
I do not know if   Amlogic 905 and 905D are the same?

Fourdee, 2018-05-13:
@zzlinwq 

Hi,

Officially, we do not support android boxes, for the following reasons:
- These tend to be configured specifically for that sole task. 
- The sheer amount of these on the market, and their unique differences, will prevent us from having a single/global image solution.
- We'd need to purchase each device we want to implement, which requires additional funding/donations to pull off. Even then, there is no guarantee it can be achieved.

However, at your own risk and research, if you can install Debian on the system, then DietPi can be installed via PREP: https://github.com/Fourdee/DietPi/issues/1285#issue-280771944

I'll mark this as closed, due the above.

closed 2018-05-13 by Fourdee