MichaIng/DietPi #1772: Dietpi 6.7. on Raspi 3: i2c state will disable after reboot

WTFMaster, 2018-05-13:
Tested on: Raspberry 3, Dietpi 6.7

1) dietpi-config -> 4 Advanced Options - > i2c state -> switch from (disabled) to (enabled)
2) Save and exit -> No reboot
3) Enter dietpi-config again -> Check the settings -> i2c state correct (enabled)
4) Reboot the mache -> i2c state switched back to (disabled)

MichaIng, 2018-05-14:
@WTFMaster
Thanks for your report. I just checked the code, where I couldn't find an issue so far. That pre reboot the correct choice is shown should be a prove that /DietPi/config.txt is adjusted correctly.

What I can think of now, is that on shutdown it is not synced to /boot/config.txt as expected.

Could you run `systemctl restart dietpi-ramdisk` and then `systemctl status dietpi-ramdisk` to check for errors? If non is shown reboot again and see if status survives then.

Fourdee, 2018-05-14:
Fresh install test with RPi 3, unable to replicate, state is saved after reboot.

Further debugging required as per @MichaIng https://github.com/Fourdee/DietPi/issues/1772#issuecomment-388819730

WTFMaster, 2018-05-14:
@MichaIng 
After changing my settings, /DietPi/config.txt looks like this:

#-------i2c-------------
dtparam=i2c_arm=on
dtparam=i2c1=on
i2c_arm_baudrate=100000

After reboot RPI the file looks like this:

#-------i2c-------------
dtparam=i2c_arm=off
dtparam=i2c1=off
i2c_arm_baudrate=100000

It seems, during the reboot the changes are lost and not written to the filesystem. How can I solve this?

WTFMaster, 2018-05-16:
run systemctl restart dietpi-ramdisk and then systemctl status dietpi-ramdisk solved the issue.

The question is: why does not the dietpi-config tool execute this command before rebooting?

Fourdee, 2018-05-17:
@WTFMaster 

> The question is: why does not the dietpi-config tool execute this command before rebooting?

The command is actually a service, which executes during shutdown phase. 
As we are unable to replicate this issue in local testing, we will need to debug with you.

However, before we can start, i need to add additional debugging to our script and log output. Leave it with me for now and i'll reply once its ready for testing.

Fourdee, 2018-05-22:
Ok, full logging output added for v6.9 on all DietPi core boot services. If issues occur in the future we can collate with the logs after user runs ```dietpi-bugreport```
If the logs are clear, we can 99.9% confirm a hardware specific issue of the data not being saved correctly, outside of our control.

I'll mark this as closed, as it wont be until v6.9 that we can continue to debug this with the updated logging.

closed 2018-05-22 by Fourdee