MichaIng/DietPi #1776: SSH | Syntax error: Bad function name

Fourdee, 2018-05-15:
> https://github.com/Fourdee/DietPi/pull/1771#issuecomment-388696242

Unable to replicate with putty/kitty.

Fourdee, 2018-05-15:
@slim0287

Question:
Can you list all the system changes you've made that you can remember (eg: locale set to en_US, timezone changed, installed nextcloud)?

slim0287, 2018-05-15:
I've been using this Pi as a Pi-hole ad blocker and smbd server for a long time so it's a bit challenging to recall.  However I believe the only system changes I made were the locale to en_US and the timezone.

Fourdee, 2018-05-15:
@slim0287 

> en_US and the timezone.

Thanks 👍 

If you get a chance, on the any of the SSH clients with this issue, once connected, please paste results:
```
echo $TERM
```

slim0287, 2018-05-15:
I connected with macOS.  The results were:
```
$ echo $TERM 
xterm-256color
$ echo $LANG
en_US.UTF-8
$ echo $LC_ALL
en_US.UTF-8
```

Any new insights?

Thanks

Fourdee, 2018-05-18:
Commit sent, resolved with: https://github.com/Fourdee/DietPi/commit/6d20b08be0b43ee9ff4a17136ebdb4520e711352

Pending v6.9 release.

Fourdee, 2018-05-18:
Patch test verified. I'll mark this as closed, however, @MichaIng if you want to reopen for other options, please do 👍

Fourdee, 2018-05-18:
User error, reopened for testing.

slim0287, 2018-05-18:
@Fourdee That solved both issues-- the -sh: 146 Syntax error and the sudo su dietpi welcome information.  Thanks!

Slim

Fourdee, 2018-05-20:
@slim0287 

Excellent, many thanks for testing and confirming fix. 👍 
This fix will be applied automatically during v6.9 update (ETA: 1+ week release).

I'll mark this as resolved.

closed 2018-05-20 by Fourdee