MichaIng/DietPi #1778: Suggestion:  integrate NextcloudPi installer for easier deployment of Nextcloud

sunjam, 2018-05-15:
[I just saw this thread comparing the two projects](https://help.nextcloud.com/t/dietpi-lightweight-debian-image-with-automated-fully-featured-nextcloud-installation/31694/2) and see no reason you could not further integrate NextcloudPi to improve Nextcloud integration and deployment in DietPi.

The [NextcloudPi curl script installation or docker image](https://github.com/nextcloud/nextcloudpi) could be easily modified to give you best of both worlds.  This would allow you to inegrate the ncp-config and ncp-web tools to more easily run OCC commands, or justrun your particular Nextcloud scripts from a friendly terminal / SSH / web interface.  @nachoparker is the friendly lead dev.

MichaIng, 2018-05-16:
@sunjam 
Thanks for your idea.

Indeed I was also thinking about this already. Generally you should be able to run NCP install script on any DietPi system, although I didn't try it so far, maybe a task for the evening.
- But as far as I can think of, the way, Nextcloud is implemented, it should break other web ui installations via dietpi-software. We heavily take care, that our implementations are as modular as possible and never do core changes/block core paths (e.g. webroot), so you can install/run everything in parallel.
- Also many system- and Nextcloud configuration options should be doubled with NCP, including data directory, possible backups, HTTPS and much more, I guess, and adjusting via NCP web ui might break the DietPi way and the other way round.

But these are assumptions so far and what I remember from visiting NCP features and code the last time 😉, as said, I need to test.

The docker image of course would work, but if you are already on a SBC and DietPi actively integrates several software besides each other, so that you don't need to run them in separate containers, these would be a large overhead and not optimal, especially performance/resource wise. Generally I am no big friend of containers. For me they should be considered as last resort, if you don't have to take care resource usage and/or need an extremely easy to maintain and stable instance, while not taking care too much about regular updates and/or individual configuration...

To integrate NCP nicely it surely would need some adjustment to reduce/avoid doubled settings and probably automate/set-fix some settings of NCP to match DietPi and/or the other way round.

Let me mark this for investigation. I am surely interested how it works out. Maybe @nachoparker has/had already some thoughts/opinion about this as well?

nachoparker, 2018-05-16:
Thanks both,

I am in favour of two people looking and maintaining one code base nicely rather than two people poorly maintaining two codebases that do the same thing. I think it leads to superior code quality, less duplication of efforts, and safer implementation due to more eyes looking at it.

I would be happy to help, time permitting to make NCP even less intrusive if needed.

> But as far as I can think of, the way, Nextcloud is implemented, it should break other web ui installations via dietpi-software. We heavily take care, that our implementations are as modular as [...]

NCP does a couple modifications to php and mysql only. Nothing that would be too hard to isolate a bit better. Some things like the tmp dir (dietpi leaves it in the SD card, NCP moves it along with the datadir to a USB drive if that option is activated) don't seem easy to completely decouple, neither for dietpi or for NCP.

Of course if there are more web services there is going to be a port clash, but that problem has nothing to do with NCP, because it will happen with any web based services that dietpi is going to install. I don't know how people deal with this in dietpi.

Armbian just included NCP in their software selection (not yet on master), so there's interest to keep things separate, independently of what happens with dietpi

[image]

> The docker image of course would work, but if you are already on a SBC and DietPi actively integrates several software besides each other, so that you don't need to run them in separate containers, [...]

I don't think the docker image fits nicely with dietPi. I mean it is a clean option and indeed it provides isolation, but wouldnt' it be weird if everything else is not containerized and NC is?

Aside from this, in general, I wouldn't disregard docker so quickly for your personal stuff. It is a great step forward in technology. It allows you to install complicated stacks superfast and reliably, and it does *not* have a performance penalty. For instance, it's isolation is perfect for situations like the one where we are at right now, with many webservices interfering each other's configurations... only updating it is a different process and has to be learnt aside from apt-get upgrade.

But that's another topic... xD

In any case, I think that first you should flash the latest NCP image and try it for yourself to understand the added value beyond NC: moving everything to an external hard drive, automount, autoupdates, auto debian upgrades, auto backups, auto SSL, DDNS support, added security, ncp-web, postfix ...

Then you can decide for yourself ;)

Ask anything

[nextcloudplus.com](nextcloudplus.com)

MichaIng, 2018-05-21:
@nachoparker 
Thanks for sharing your ideas. Yes, indeed it's a question if we can get NCP installed in a way, that an existing webserver stack is adjusted with modular configuration files only and a Nextcloud-only web UI, that doesn't include otherwise doubled system settings. Simply a Nextcloud-only version. Also we allow Nginx and Lighttpd as webservers.

Let my now start with NCP installation on my 3 DietPi VMs 😃.
_________________
<details>
<summary>Off-topic about Docker</summary>

@nachoparker 
> docker ... it does not have a performance penalty.

To be true, it was more a logic assumption of me. And if it's about runtime performance only, you seem right: https://stackoverflow.com/questions/21889053/what-is-the-runtime-performance-cost-of-a-docker-container#26149994
- Less abstraction than I thought 👍.
- Though network seems to be slower.

But of course if you run individual webservers, databases, memory caches etc. in separate containers instead of having all your software sharing one of each, you will face a large impact on RAM and disk usage and overall CPU time/usage. The single environments may be nearly as fast as being natively implemented, but the benefits of having all your software within their own environment+dependencies implicates automatically the disadvantages. Of course doubled doesn't mean doubled always, as you can optimize every environment for it's specific use case, disable/remove unneeded modules/parts and of course have just the usage of the related single software inside the container. But two highly optimized databases/webservers/... will always cost more than one that is configured to handle two web applications. 

Besides there are of course several forcing use cases and RAM/disk usage is not always a bottle neck, for me it's also a quite personal taste and finally ideological question. Stop reading here in case, I just love to go into philosophical discussions by times 🤣:
- Having everything within it's own individual environment for me in general is kinda resignation to being able to make things work together, share their needs etc. Of course it's always "easier" to have everything have it's own, optimized in how it's needed. But in the end seen from a wider perspective, sharing goods/resources will mostly be an overall(!) benefit, at least if it's done good, which might need some additional work and thinking. Of course this can be understood in political terms as well, if humanity will be able to finally work together, sharing resources in a fair overall beneficial way, or if finally everyone is looking just onto his/her own (smaller environment), building walls/clear separations, as it's the easier way than talking and finding solutions to work together instead against/aside each other.
- Finally it's a question what currently "works". On Windows I see 10 to 20 separate visual C++ implementations aside each other, as every software brings it's own specific, often very outdated version. I read that this was different in the past, causing huge issues, as Microsoft obviously was not able to keep it backwards compatible for older software titles and software developers were not able to upgrade their software based on the new C++ version. Okay it's just a "small" library here and before taking huge effort somehow guaranteeing old and new software titles work with the most current version, just allow them bring their own...
- But for me, all this, same as docker, is not "step forward" in technology. It's a (indeed well done) step backwards, that make certain stings easier. A step "forward" or better: modern technology design, in my opinion, should always developing into a direction, that implies different use cases, modularity and most compatible interfaces according to open standards to allow it, or parts of it being implemented and used by different software/users in a different way in parallel.
- This finally is, why I was falling in love with Linux (compared to Windows or MacOS), FOSS in general and DietPi in particular, as we are working on running different software on an open standard OS/distribution driven by again very different hardware ❤️.
- On the other hand, I like to have my own server at home handle data myself, instead of sharing server capacity and cloud solutions. Yeah, it's always a question of how the implementation is done and if you want to play and tinker around as well 😉. For development/tinkering of course an isolated environment totally makes sense. I currently run 3 VMs here, ready to install NextcloudPlus 😄!
</details>

MichaIng, 2018-05-21:
### Testing protocol
- On fresh DietPi Stretch VirtualBox VM:
1. `dietpi-software install 76 91` # LAMP stack + Redis
2. `curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash`
   - LAMP stack + Redis want's to be installed again
   - Apache is installed with `php-fpm`, where DietPi installs it with `mod-php`.
   - Some Apache modules are actively disabled<s>, which could cause issues with other software.</s> Some kind of Apache(+PHP) module dependency system is anyway something I have on my list, although many other things with higher priority.
**€: Ah found re-enable code, modules are disabled just during installation 👍.**
   - Apache configured with basic HTTPS functionality (snake oil certs), although within separate configuration/vhost and no forced redirection/HSTS, this should be no problem.
   - MariaDB configuration output seems to expect root password authentication. However since Debian Stretch, unix_socket authentication is default, allowing login only when logged in as unix root user, but without the need to enter password for MariaDB login again. Compatibility with this behaviour assured?
🈯️**€: Ah just found `mysql_secure_installation` script as provided by MariaDB. Nice to have, but on Debian Stretch not needed, as stated above: root user present only with unix_socket authentication configured, which is great for my impression. However running the script does not damage something.**
   - Afterwards NCP installation itself starts, maybe this is where we can start from, which DietPi.
   - Postfix is installed. Would be nice to have modules like this optional, allowing to e.g. skip installation (and of course potential UI elements, depending on it) via command option, global variable or setup configuration file.
   - Redis is being installed at last, consider moving installation to the beginning within/between LAMP stack installation etc. and just do final Nextcloud related needed configuration within final NCP install script. This would also avoid enabling the service, which does not work in combination with DietPi, controlling the services itself.
   - 🈴 Last output: `Failed to stop mysqld.service: Unit mysqld.service not loaded.` The service is called `mysql` (init.d) or optionally `mariadb` (systemd). <s>Is installation aborted here or finished anyway?</s>
**€: Script execution broken at this stage, attempt with workarount below...**
   - 🈴 Installer expects PHP 7.0 to be installed. This will break installation with default PHP version on Jessie (PHP 5) and Buster (PHP 7.2).

<details>
  <summary>Whole terminal output</summary>

```
root@DietPi:~# curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash
>>> WARNING: existing mysqld configuration will be changed <<<
Getting build code...
Ign:1 https://cdn-aws.deb.debian.org/debian stretch InRelease
Hit:2 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease
Hit:3 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease
Hit:4 https://cdn-aws.deb.debian.org/debian stretch-backports InRelease
Hit:5 https://cdn-aws.deb.debian.org/debian stretch Release
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
ca-certificates is already the newest version (20161130+nmu1).
sudo is already the newest version (1.8.19p1-2.1).
wget is already the newest version (1.18-5+deb9u2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
/tmp/nextcloudplus.n9eLwT ~
--2018-05-21 13:36:52--  https://github.com/nextcloud/nextcloudpi/archive/master/latest.tar.gz
Resolving github.com (github.com)... 192.30.253.113, 192.30.253.112
Connecting to github.com (github.com)|192.30.253.113|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/nextcloud/nextcloudpi/tar.gz/master [following]
--2018-05-21 13:36:53--  https://codeload.github.com/nextcloud/nextcloudpi/tar.gz/master
Resolving codeload.github.com (codeload.github.com)... 192.30.253.121, 192.30.253.120
Connecting to codeload.github.com (codeload.github.com)|192.30.253.121|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘STDOUT’

-                                                               [    <=>                                                                                                                                  ] 712.20K   983KB/s    in 0.7s

2018-05-21 13:36:54 (983 KB/s) - written to stdout [729296]

/root

Installing NextCloudPlus
Installing lamp
Ign:1 https://cdn-aws.deb.debian.org/debian stretch InRelease
Hit:2 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease
Hit:3 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease
Hit:4 https://cdn-aws.deb.debian.org/debian stretch-backports InRelease
Hit:5 https://cdn-aws.deb.debian.org/debian stretch Release
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
apt-utils is already the newest version (1.4.8).
cron is already the newest version (3.0pl1-128+deb9u1).
[... 457 lines cut ...]
Executing: /lib/systemd/systemd-sysv-install enable redis-server
Created symlink /etc/systemd/system/redis.service → /lib/systemd/system/redis-server.service.
Failed to stop mysqld.service: Unit mysqld.service not loaded.
root@DietPi:~# htop
[... cut ...]
Executing: /lib/systemd/systemd-sysv-install disable apache2
root@DietPi:~# systemctl disable php7.0-fpm
Synchronizing state of php7.0-fpm.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable php7.0-fpm
root@DietPi:~# htop
root@DietPi:~# systemctl status mysqld
Unit mysqld.service could not be found.
```
</details>
_______________
Workaround to fix mysqld service name:
`sed -i '/systemctl .* mysqld/s/mysqld/mysql/g' etc/ncp-config.d/nc-nextcloud.sh`

nachoparker, 2018-05-24:
great job. Give me a few days to go through these points, as I am now preparing the next release

nachoparker, 2018-06-12:
Hi again,

Apologies for taking so long but I have not had the time to sit down for this before.

Indeed, the approach is different in both projects, and some modifications would need to be made,  which might be fine. Still, we could share the maintenance burden so I think it would be nice for both projects.

If you look at the [code](https://github.com/nextcloud/nextcloudpi/blob/master/install.sh), NCP installation  boils down to the following steps

```
install_script lamp.sh
install_script nc-nextcloud.sh           # install Postfix and Redis
activate_script nc-nextcloud.sh          # install NC itself
install_script ncp.sh                    # install all NCP extras 
install_script nc-init.sh                # preconfigure NC
```
I think that the first logical step would be to skip `ncp.sh` and initially share the rest of the installation between projects. Then we could look at what NCP really adds and how it can be combined with what DietPi does.

Also, NCP only supports Stretch ATM, and the main reason is really small manpower. What is the case with DietPi? do you guys support several different releases?

---

### To your points:

> Apache is installed with php-fpm, where DietPi installs it with mod-php.

This is one thing that would need to be changed on one side... well, not necessarily. Which one is best? did a little search.

https://www.cloudways.com/blog/php-fpm-on-cloud/
http://fastjoomlahost.com/mod_php-fastcgi-php-fpm-server

Thoughts?

> Apache configured with basic HTTPS functionality (snake oil certs), although within separate configuration/vhost and no forced redirection/HSTS, this should be no problem.

HSTS is enabled by default [code](https://github.com/nextcloud/nextcloudpi/blob/master/lamp.sh#L79). Also HTTP2, OCSP Stapling and other stuff.

> MariaDB configuration output seems to expect root password authentication. However since Debian Stretch, unix_socket authentication is default, allowing login only when logged in as unix root user, [...]
u6307€: Ah just found mysql_secure_installation script as provided by MariaDB. Nice to have, but on Debian Stretch not needed, as stated above: root user present only with unix_socket authentication configured, which is great for my impression. However running the script does not damage something.

mysql_secure_installation does other things too, it's nice to keep I think ;)

the unix_socket behaviour should be no problem. I didn't even notice the change and didn't have to adapt it to Stretch.

> Postfix is installed. Would be nice to have modules like this optional, allowing to e.g. skip installation (and of course potential UI elements, depending on it) via command option, global variable or setup configuration file.

It is a different step, ( nc-nextcloud.sh "activation", meaning the configure() step ), so it would be easy to separate, but I wonder what's the benefit of removing it. Does DietPi also install postfix?

> Redis is being installed at last, consider moving installation to the beginning within/between LAMP stack installation etc. and just do final Nextcloud related needed configuration within final NCP [...]

Redis is installed after LAMP, and before Nextcloud (same with Postfix), so I think we agree here. I didn't understand the problem with enabling/disabling the service.
 
> Last output: Failed to stop mysqld.service: Unit mysqld.service not loaded. The service is called mysql (init.d) or optionally mariadb (systemd). Is installation aborted here or finished anyway?

In my Raspbian Stretch all three exist: mysql, mysqld, mariadb, so I am surprised. Are you in Stretch? In any case, no problem to change this.

> Installer expects PHP 7.0 to be installed. This will break installation with default PHP version on Jessie (PHP 5) and Buster (PHP 7.2).

Again, ATM only Stretch is supported. There are people starting to ask for PHP > 7.0, so one option would be to move to PHP 7.2 (by bringing sources from Buster to Stretch). That could potentially cause problems though, for instance that is what I did in Jessie to get PHP 7.0 support, but it would break php-smbclient.

Also, PHP 5.0 is just slow

> Script fails/breaks, if ncp user already exists at: https://github.com/nextcloud/nextcloudpi/blob/master/nextcloudplus.sh#L120

The ncp user is not a login user. It is only used to log in to ncp-web. I think it is better that is separated from the login user, and has no permissions whatsoever. In any case, this is part of `ncp.sh`

In `lamp.sh`, another thing that should be done is to stop modifying `/etc/mysql/mariadb.conf.d/50-server.cnf` and create separate files, like `/etc/mysql/mariadb.conf.d/90-ncp.cnf`. Same for php ini and apache files where possible.

So the "curl installer" is a very simple one in terms of robustness: it expects a clean Stretch system. If there is already a LAMP stack or if you try to install it twice some things might fail. This would be another point for improvement.

testbird, 2018-07-26:
I'd actually like the curl installer to allow "frugal"? installs on **any** preexisting and externally managed debian installations.

The LAMP and system settings would have to be checked and adjusted in a fine grained method if there is no modular conf.d directory.

May this work?:

* Make completed ncp installs detectable by moving a temporary ncp config state directory to a permanent place on success.

* Instead of replacing (default or external) config files, always grep config files for the individual required ncp settings or stanzas and if not present:
  * If it is a required configuration, insert/comment out or change it in the existing files (together with a comment "# Following line was udpated by NCP on $date:".
  * If the config file was already present before installing NCP, list all changes in an install report.
* Keep separate copies of the managed config files and inotify-watch the config files to redo the grep/edit on any external changes, and sending the diff to the admin.

testbird, 2018-07-26:
Well, maybe in the end above "ncp--provisioning-service" could get installed with a .deb package.

install_script lamp.sh
install_script nc-nextcloud.sh           # install Postfix and Redis
install_script ncp.sh                    # install all NCP extras 
   **->** debian package with dependencies and installing the nc-provisioning-service

activate_script nc-nextcloud.sh          # install NC itself
install_script nc-init.sh                # preconfigure NC
   **->** configs watched by ncp-provisioning-service

nachoparker, 2018-07-26:
I don't think I understand the point of this new script. I think that if possible, it would be nice to enhance the current scripts (lamp.sh specially) to

- be more modular, by creating `conf.d`files instead of modifying the global configuration files
- try to minimize changes, because of the fact that it won't be the only thing installed in the system necessarily anymore
- try to detect whenever something might get broken (hard) by the script and make it more robust so that it can be, for instance, run twice in a system without problems. For instance, don't run nc-init if it already exists.
- backup any file that is changed

In any case, I still think that integration should start by the lamp.sh + nc-nextcloud.sh + nc-init.sh, skipping the rest of NCP software. When that is completed we can move on to see how and if we add the rest of the software.

testbird, 2018-07-26:
Hi,
the reason I thought of a new "script" may just be that I may not have understood completely the reasons for the way you had split things up currently.

The idea was just wrapping things into some (service?) file that is started on boot to check the configs and starts watching for changes. (May just call the scripts as split up currently.)

Do you think this would solve your points?: Each action on an existing file would have to be
1) made conditional (skipped if the outcome already exists)
1) copy the old and new version of the file to a backup location
1) and logged.

For particular settings the condition test could, for example, be a regexp grep.

testbird, 2018-07-26:
How would you control skipping the rest of NCP software?

testbird, 2018-07-26:
> We use some config injection function for this: https://github.com/Fourdee/DietPi/blob/testing/dietpi/func/dietpi-globals#L2223

Very nice thing to share, too bad you said "All mine!" :D :-)

> we stay with the default APT version each repo.

Sounds very reasonable, a more generic meta-package? depenency on php would allow installing on stable, testing and unstable, even if initially not supported and working yet, it will allow users to test and fix things.

testbird, 2018-07-26:
@nachoparker Maybe could you introduce a frugal install detection mechanism (not ncp-build?, not armbian?) and a variable that causes only the basic things to get installed.

@MichaIng Maybe could you add the backup creation to G_CONFIG_INJECT() in a way that allows it to also be used in https://github.com/nextcloud/nextcloudpi/blob/master/etc/library.sh (to only make a check and allow skiping the corresponding ncp install step, if the configuration already exists)

With that basic infrastructure in place it could be possible to adapt and test on different systems.

testbird, 2018-07-26:
>>  Does DietPi also install postfix?

> I like... to have installs by default as clean/slim as possible

Maybe using a non-spooling, ram-only forwarder is all that is needed to send out notifications. (Not sure msmpt, esmtp, or ssmtp?)

testbird, 2018-07-27:
I guesss the status that some service or setting was already found (preinstalled) initally will have to get saved, to later skip any corresponding updates.

Is there some specific way you would imagine this to work? Append lines naming preinstalled or external options to some file?

testbird, 2018-07-28:
I have made some things optional in the https://github.com/testbird/nextcloudpi/ branch, and **after** php-redis is installed manually from sid (because it is currently not available in buster) the installer is now able to run through in the freedombox debian-testing image. (https://ftp.freedombox.org/pub/freedombox/testing-latest/freedombox-testing-free_latest_all-amd64.vdi.xz)

However, I am not able to find out why it broke the web access. I don't know enough how all the parts are interrelated. Would be great if you could could improve on my patches.

@MichaIng It would be really great if you could find the time to update your G_CONFIG_INJECT() with prior-state tests and and backup keeping of the prior and the configured state, in a way that it could also be used in the NCP installer.

nachoparker, 2018-08-01:
### nginx

One of the reasons why I use Apache is because of mod_security. This is harder to install in nginx. I definitely don't have the manpower for supporting three different LAMP stacks, but if I had more help it would be doable. For the same reason, I only support Stretch, and again if there was someone adapting the system to Jessie I would gladly merge that.

### HTTPS

In the days of insecure IoT I believe that even in your local network you need HTTPS. Maybe you can skip signing your certificates, but HTTPS is a must. In any case, both Letsencrypt and HTTPS are optional in NCP, but by default NCP works with HTTPS only and you have to disable that manually, which I think it's the right thing to do in this case.

### Postfix

I add postfix because that way the user receives notifications about shared files and other activity in their cloud. It is quite standard in Nextcloud and I think it is nice to include it. It is true that it is a bit heavy for what it does. I would be happy to replace but something lighter, we don't need a full fledged email server to send some informative emails.

Again here our projects have different goals. NC aims to be an all-included ready to use solution, and tries to minimize the amount of obscure steps that a non-technical user has to go through. Most people don't even know what postfix is, but they appreciate that the system sends them emails automatically out of the box. That being said, it could be a good improvement to have some option to disable it, and maybe install it separatedly from NC itself (in a separate step).

### PHP

> For now at least we stay with the default APT version each repo.

Agreed.

### NCP user

Agreed, we should detect this, and probably just throw a big warning in case it already exists. I created https://github.com/nextcloud/nextcloudpi/issues/579

### Config modularity

Created https://github.com/nextcloud/nextcloudpi/issues/580

### More robust installer

In order to be able to run in systems that are not under our control, it would be nice to add a few functions to the [NCP library](https://github.com/nextcloud/nextcloudpi/blob/master/etc/library.sh) and use that during build/update/installation. We can take the config injection function from DietPi, and potentially others as well.

In the past I wanted each single `.sh` file to be as self contained as possible, so I was trying to avoid sourcing other files, but in the new situation at hand I think it would be a nice move. This would make the code more readable as it is full of _seds_ and _greps_, we could log actions better, backup configs and more.

https://github.com/nextcloud/nextcloudpi/issues/581

I think that @testbird agrees on this ;)

---

Sadly I won't have both time and a computer for a few weeks, but it is good that we start creating items. Feel free to open other ones in NCP github.

MichaIng, 2018-08-30:
Updated G_CONFIG_INJECT: https://github.com/Fourdee/DietPi/pull/2042
- Added `grep` error catching of the if statements. This is needed to prevent e.g. adding setting to end of file, if `grep` checks fail due to error instead of string not found.
- Added `BACKUP=1` handling to create backup /path/to/file.bak before editing the file.

Only issue with direct NCP implementation is that it uses `G_DIETPI-NOTIFY` and `G_WHIP_MSG`, which would need to be replaced by `echo -e` and `whiptail --msg`.

Feel free to use and adjust for own needs of course. If you know the settings you want to add, all the error handling could be skipped, to have a lightweight version. I added this just, since we use the function much throughout DietPi and source it to current terminal session, so end users can use it as well as global function. Errors can happen if []$"{}() and such are not escaped properly (see function comments), also since extended regular expressions are handled (grep/sed -E).

kevkha, 2018-09-07:
Is PHP 7 coming soon? Nextcloud 14 requires PHP 7 so checking. Thanks.

MichaIng, 2018-09-07:
@kevkha 
> Is PHP 7 coming soon?

We offer Debian Stretch images, thus PHP7, for all our supported devices, besides Odroid C1.

> Nextcloud 14 requires PHP 7

Are you sure??? Can't believe it, haven't heard of it and it would break compatibility for a very large user space. Do you have a link for this info?
In case this would mean that we need to manually install NC13 on Jessie images and inform users that they can't use Nextcloud 14. Flash new Stretch image would be needed otherwise, or manual steps to dist-upgrade to Debian Stretch or install PHP7 🤔.

MichaIng, 2018-09-10:
https://github.com/nextcloud/server/commit/ace96a406a4fea0dcc07abd003495da34b5fa71c
- Indeed PHP5 support dropped with Nextcloud 14. Not sure what is best to do. I hope on Jessie/PHP5 systems, WebUI update attempt shows meaningful output and is prevented, so we can resolve it by forcing NC13 on Jessie. Otherwise we need to inform Jessie users that they shall not update and remove Nextcloud install option for Jessie to avoid unexpected issues.
Test needed.
@Fourdee 
This should be fixed/worked around before v6.15 release, as Nextcloud 14 is already released. Today official launch is planned, a fresh install already pulls v14, just the WebUI upgrade is not yet offered in my case.

nachoparker, 2018-09-11:
indeed a situation. Until when are you guys going to support Jessie? I guess you will have to support two versions of NC :S

in NCP due to shortage of manpower we only support the latest stable debian

MichaIng, 2018-09-11:
@nachoparker 
At least with PHP5 in use, NC14 update is not offered and we do no NC14-only install steps. So we just need to download/install the latest NC13 on Jessie systems.

The reason to still support Jessie is, besides existing Jessie users from older times, that the Odroid Stretch pre-image we use does not have full GPU support to play e.g. Kodi smoothly. And Kodi is one of the more used software on DietPi systems.
Also we have no Stretch image for Odroid C1 available :thinking:. We can just hope that Meveric finds a fix for his Odroid Stretch images soon or have to disappoint many of our Odroid users, sooner or later.

MichaIng, 2018-09-11:
@kevkha 
Thanks for additional verification. Solution from our side will be as above to simply install https://download.nextcloud.com/server/releases/latest-13.zip on Jessie systems.

I am still surprised by this decision, as e.g. all the RHEL users already had problems to get PHP5.6 when PHP5.4 support was dropped, but the step to PHP7.0 is indeed a larger one, if one wants to stick with LTS distros and their native software repos.

nachoparker, 2018-09-11:
Too bad for boards with shitty support. IIRC all Armbian boards NCP supports are on Stretch already

kevkha, 2018-09-18:
Another though. Switching to use docker container https://ownyourbits.com/2017/02/13/nextcloud-ready-raspberry-pi-image/

MichaIng, 2018-09-19:
@nachoparker 
Odroids are actually well supported. Sadly official images just include Ubuntu and Android 😭. But one developer Meveric offers clean Debian images, including an own large APT software and kernel repository with regular updates.

The missing Kodi (rechecked, it's really just Kodi, Chromium GPU acceleration works) GPU support is the left Stretch-only issue I am aware of.

@kevkha 
We offer docker and everyone is free to install docker containers of course, but we will not add docker container installs to `dietpi-software` or take special care about it. DietPi tries to integrate software hierarchically flat into the system. Docker encapsulates software and has a hierarchically vertical system structure. This has great benefits, e.g. avoiding the need to take care of compatibility and dependencies of different software titles, but the abstraction/encapsulation also has disadvantages of multiple resource (e.g. drive space, ram) usage and in some areas less performance (although not 100% sure which areas, also in comparison to snaps and such). Configuration abilities are mostly reduced as well.

MichaIng, 2019-11-06:
I mark this issue as closed. I think it has been discussed enough that both projects have their own stranges and aims and should be seen as either/or for now. Making DietPi more flexible and compatible with 3rd party installers is a general long term aim that is followed by many small steps with every DietPi update. I'll have a look back at NextCloudPi, as I am following it anyway with great interest, so as fast as we can think of an integration, we'll open it as new software request.

closed 2019-11-06 by MichaIng