MichaIng/DietPi #1780: Home assistant installation fails on Odroid C2

brumouta, 2018-05-16:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | 6.8
- SBC Device | Odroid C2
- Power supply used | iPad charger
- SD card used | Samsung class 10
- Distro | Linux DietPi 3.14.79+ #1 SMP PREEMPT Sun Jul 23 00:04:44 CEST 2017 aarch64 GNU/Linux

#### Additional Information (if applicable):
- Software title | Home assistante
- Can this issue be replicated on a fresh installation of DietPi? | Don't know, since I have only this memory card and it's already working with my media center

#### Expected behaviour:
I try to install home assistant via dietpi-software and it can't be installed because of broken packages

#### Actual behaviour:
DietPi-Software: G_AGI: libssl-dev git cmake libc-ares-dev uuid-dev daemon curl
libgnutls28-dev libgnutlsxx28 nmap net-tools sudo libglib2.0-dev libudev-dev swi
libssl-dev libusb-1.0-0 gcc libssl-dev libffi-dev libbz2-dev zlib1g-dev
libreadline-dev libsqlite3-dev libncurses5-dev libncursesw5-dev
 - Exit code: 100
 - DietPi version: v6.8 | HW_MODEL:12 | HW_ARCH:3 | DISTRO:3
 - Image creator:
 - Pre-image:

Log file contents:
E: Unable to correct problems, you have held broken packages.

#### Steps to reproduce:
With and odroid C2 with plex, sickrage, couchpotato and transmission installed try to install home assistant via dietpi-software

#### Did you submit a ```dietpi-bugreport```?
No

#### Extra details:
I've tried manually installing packages above, but the error persists

Fourdee, 2018-05-17:
@brumouta 

Thanks for the report, i'll take a look 👍

Fourdee, 2018-05-17:
@brumouta 

Lets see if you are running the updated Stretch image, please paste results,
```
cat /etc/os-release
```

brumouta, 2018-05-17:
Mine is Jessie.

PRETTY_NAME="Debian GNU/Linux 8 (jessie)"
NAME="Debian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=debian

Fourdee, 2018-05-17:
@brumouta 

> Jessie

Apologies, we updated our Odroid C2 image a few weeks ago, it now runs Debian Stretch. 

This will offer more recent software installations, and, performance improvements (eg: PHP7.0).

#### Please re-write the latest DietPi image to resolve this issue:
https://dietpi.com/downloads/images/DietPi_OdroidC2-ARMv8-Stretch.7z

I'll flag this as a known issue, however, the C1 is the only Jessie based image we have left over, and, no plans to update the C1 to stretch. Therefore, i'll mark this as closed.

closed 2018-05-17 by Fourdee