MichaIng/DietPi #1784: DietPi-Software: add influxdb and grafana

marcobrianza, 2018-05-19:
There are already several use cases of Diet-Pi for IoT / home automation thanks to softwares such as mosquitto and Node-Red.
The ability to easily log and visualize time base data completes the Diet-Pi strenght in this field.

I'm currently using influxdb as time-series database and grafana as visualization server on Diet-Pi both on Raspberry Pi and X86 virtual machine.
The manual installation is already quite easy, but the dietpi-software integration would make it super smooth for all and will standardise things as data location etc...

Influxdb https://www.influxdata.com/time-series-platform/influxdb/
influxdb is officially supported on X86 and ARM and I'm installing with this:
```
curl -sL https://repos.influxdata.com/influxdb.key | sudo apt-key add -
echo "deb https://repos.influxdata.com/debian stretch stable" | sudo tee /etc/apt/sources.list.d/influxdb.list
sudo apt-get update
sudo apt-get install influxdb
```

automatic service startup is enabled by default

grafana https://grafana.com/

grafana is supported on X86 by the creators and on ARM by side project (https://github.com/fg2it/grafana-on-raspberry/wiki)
on X86 the easyest way is with this:
```
curl -s https://packagecloud.io/install/repositories/grafana/stable/script.deb.sh | sudo bash
sudo apt-get install grafana
```

on ARM there is alreay a very old grafana 2.6 in the default repos that must be avoided.

I install this way:
```
curl https://bintray.com/user/downloadSubjectPublicKey?username=bintray | sudo apt-key add -
echo "deb https://dl.bintray.com/fg2it/deb stretch main" | sudo tee -a /etc/apt/sources.list.d/grafana.list
apt-get update
apt-get install grafana
```

automatic startup must be enable with
```
systemctl daemon-reload
systemctl enable grafana-server
```

I'm ready to contribute to the feature providing some basic configuration of the systems with sample data and to testing it.

Fourdee, 2018-05-20:
@marcobrianza 

Many thanks Marco 👍 

Usually we close all software requests, pending comminuty interest before investigating, however, as you have done a cracking job with the install info, i'll get this added in for v6.9.

> I'm ready to contribute to the feature providing some basic configuration of the systems with sample data and to testing it.

Any assistance with testing/configuration would be great, i'll let you know once its added to ```dietpi-software``` and available for testing.

Fourdee, 2018-05-20:
Notes
- influxdb | ID=74 | https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=120#p12523
- grafana | ID=77 | https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=12524#p12524

Fourdee, 2018-05-20:
@marcobrianza 

Install/uninstall code added, no changes to default install configs yet, could do with your help on that. 

#### Here is how you can test the installation:
https://github.com/Fourdee/DietPi/blob/master/TESTING-BRANCH.md#steps-to-use-the-dietpi-testing-branch

#### I've tested installs on Stretch:
- 🈯️ ARMv7
- 🈯️ ARMv8
- 🈯️ x86_64

Fourdee, 2018-05-21:
To Do:
- Online docs

marcobrianza, 2018-05-21:
for the moment I have succesfully tested  influxdb on x86 and raspberry Pi 1 and 3

**documentation**
influxDB is a database optimized to save time based data as logs or data from a sensor.
The main interface to the database for management and data transfert is http requests that are handled directly by influxdb service (default port being 8006).
The data can be nicely viewed with Grafana.

The package comes with a command line tool 'influx' for database management operations
This tool also uses http so it can manage a database on a remote machine setting the -host option.

official documentation can be found at https://docs.influxdata.com/influxdb/

Some quick tips to start:

create a database:
`influx -execute 'create database mydb'`
or
`curl -i -XPOST http://localhost:8086/query --data-urlencode "q=CREATE DATABASE mydb"`

posting data:
```
curl -i -XPOST 'http://localhost:8086/write?db=mydb' --data-binary 'temperature value=20.12'
curl -i -XPOST 'http://localhost:8086/write?db=mydb' --data-binary 'temperature value=21.13'
curl -i -XPOST 'http://localhost:8086/write?db=mydb' --data-binary 'temperature value=22.42'
```

to show data:
`influx -database mydb -execute 'SELECT * FROM temperature'`
or
`curl -i -XPOST http://localhost:8086/query?db=mydb --data-urlencode "q=SELECT * FROM temperature"`

by default http authentication is disabled, to enable it edit the configuration file
/etc/influxdb/influxdb.conf

[http]
auth-enabled = true

to create users and authorizations from influx CLI
```
influx -username admin -password admin01
>CREATE USER admin WITH PASSWORD 'admin01' WITH ALL PRIVILEGES
>CREATE USER test_user WITH PASSWORD 'test_user01'
>GRANT ALL ON mydb TO test_user
```

**data location:**
data location is by default in /var/lib/influxdb/
I suggest to put it in /mnt/dietpi_userdata 

mkdir /mnt/dietpi_userdata/influxdb
chown influxdb /mnt/dietpi_userdata/influxdb

in file /etc/influxdb/influxdb.conf

change theese paths:

[meta]
dir = "/var/lib/influxdb/meta" to: dir = "/mnt/dietpi_userdata/influxdb/meta"

[data]
 dir = "/var/lib/influxdb/data" to:  dir = "/mnt/dietpi_userdata/influxdb/data"
 wal-dir = "/var/lib/influxdb/wal" to:  wal-dir = "/mnt/dietpi_userdata/influxdb/wal"

marcobrianza, 2018-05-21:
I just checked that grafana has different repos for the different ARM versions. The one I linked yesterday is ok for raspberry 2 and 3 and does not run on raspberry 1.
on ARMv6 this is the config:
`echo "deb https://dl.bintray.com/fg2it/deb-rpi-1b stretch main" | sudo tee -a /etc/apt/sources.list.d/grafana.list`

This also runs on ARMv7 (tested)

There is also one for ARMv8 but cannot test it since I have no such board
`echo "deb https://bintray.com/fg2it/deb-arm64/grafana-on-raspberry/ stretch main" | sudo tee -a /etc/apt/sources.list.d/grafana.list`

Fourdee, 2018-05-22:
@marcobrianza 

Many thanks for the additional testing and info 👍 

> data location is by default in /var/lib/influxdb/
I suggest to put it in /mnt/dietpi_userdata

We'll do a symlink as per grafana install.

> ok for raspberry 2 and 3 and does not run on raspberry 1.

Thanks, i'll go through these and re-check/change as needed.

marcobrianza, 2018-05-22:
some info to create the grafana info post:

grafana runs a stand alone web server, the default port is 3000
http://dietpi.local:3000
default credentials are admin/admin
 
If you have created a database as in the influxDB help follow theese instructions: 
 
from the Home Dashboard click "Create you first data source"
Name: mydb_source
Type: InfluxDB
URL: http://localhost:8086
Database: mydb
User: test_user
Password: test_password01
leave the rest with the default values

click "Save & Test"

from the Home Dashboard click "New dashboard"
click Graph
click the example graph
press 'e' to edit the Data Source and the custom query

Fourdee, 2018-05-23:
@marcobrianza 

I believe we are completed. Tests installs and using your guide was spot on 👍 Great work on this. Email me your address (daniel.knight@dietpi.com) and i'll send over some DietPi stickers:
https://twitter.com/DietPi_/status/998305772227497985

I've made one change to Grafana, the default web interface password is 'dietpi'.

I'd also be grateful if you could run a final test install + use the online documentation, just to verify its completed.
https://github.com/Fourdee/DietPi/issues/1784#issuecomment-391412965

marcobrianza, 2018-05-25:
I should do the final tests by the end of the week. I already got yesterday the dietpi stikers because of a donation :-)

marcobrianza, 2018-05-26:
Hi, I'm still out of office and started to test just on virtualbox:
InfluxDB all correct.

Grafana has an issue in the repo url. I'm attaching the screenshots: the error from the setup, and the sources situation.
[image]

[image]

Fourdee, 2018-05-26:
@marcobrianza 

Confirmed, thanks 👍 

Bug our end, i'll get it fixed:
```
root@DietPi:~# cat /etc/apt/sources.list.d/grafana.list
deb  stretch main
```

Fourdee, 2018-05-26:
@marcobrianza 

Resolved, install now succeeds. Make sure to reinstall the VM, or if you made a backup of system prior to installing InfluxDB/Grafana, you can simply run:
```
G_DEV_1
dietpi-software install 74 77
```

marcobrianza, 2018-05-26:
Yes! I can confirm install works on virtualbox and setup is ok , tomorrow hopefully I will test the raspberries.

marcobrianza, 2018-05-27:
I confirm that setup and software run correctly also on raspberry 1 and 3. I think we are done!

Maybe we should add a note on grafana that binaries are different for the different arm architectures: I'm used to swap SD cards between raspberries without considering the model.
I don't know if there are different builds for other softwares in DietPi and the warning is addressed already.

Thank you for adding influxdb and grafana...and... I'll be back some day with another request :-)
because DietPi is my favourite special purpose linux distribution.

Fourdee, 2018-05-27:
> Maybe we should add a note on grafana that binaries are different for the different arm architectures: I'm used to swap SD cards between raspberries without considering the model.

Yep, added a notice: https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=12524#p12524. Although generally speaking, once software is installed, we do not recommend swapping SD cards across RPi arch's, as some of the software in ```dietpi-software``` is CPU arch specific.

> Thank you for adding influxdb and grafana...and... I'll be back some day with another request :-)
because DietPi is my favourite special purpose linux distribution.

Thank you for providing all the install and documentation info, really appreciate it. Made my life alot easier 😃 
Yep, by all means, if you have anything else you think our DietPi users can benefit from, please let us know.

I'll mark this as completed. Thanks again @marcobrianza 👍

closed 2018-05-27 by Fourdee

francescogiunta, 2021-02-16:
Hi and thank you for the great job! Justo to confirm DietPi+Grafana+InfluxDb+NodeRed+Mosquitto work well even on Raspberry PZero W!
[image]