MichaIng/DietPi #1787: PREP_SYSTEM_FOR_DIETPI.sh exits with error "Failed to get D-Bus connection: No connection to service manager."

Nakda, 2018-05-20:
Just tried to follow the procedure to prepare a freshly installed Debian system for DietPi (https://github.com/Fourdee/DietPi/issues/1285#issue-280771944). 
During the process, it terminated returning the following error:
Failed to get D-Bus connection: No connection to service manager.

SBC Device: Odroid-XU (yes, the first of the name)
Distro: Debian Jessie (https://forum.odroid.com/viewtopic.php?f=62&t=2070)

Fourdee, 2018-05-20:
@Nakda 

Thanks for the report 👍 

> Odroid-XU 

Nice, always wanted one of those, no longer in production.

Ok, any chance you could copy/paste the entire SSH terminal output, when the error occurs? Will assist us in tracking down the area this occurs.

MichaIng, 2018-05-20:
@Nakda @Fourdee 
I am wondering why D-Bus is actually called on `systemctl start` or better why it fails, if D-Bus connections fails. It works perfectly completely without D-Bus.

But first some other questions @Nakda:
- Did I see it right that you ran the preparation script two times, or is it just some copy&paste issue?
- Same about the funny characters, do they show up correctly on real terminal/SSH client, colourized as expected?
- I first thought that `localepurge` might interfere here in a bad way, which then often leads to broken output, as current session locale is reset to POSIX e.g. Could you verify that after PREP stops, `en_GB.UTF-8` is available? Run to check: `locale -a`
- In case try to purge localepurge and run the script again: `apt purge localepurge`

Nakda, 2018-05-20:
@MichaIng 

I ran it once on a fresh install but didn't have logging enable in Putty settings (forgot I could simply copy/paste). So I ran the process a second time, just to ouput the logs and the funny characters from the Putty log file (they were displayed correctly in the terminal).

I just edited my post fixing characters, by simply copy/pasting the terminal output but you were probably already answering. Sorry about that.

MichaIng, 2018-05-20:
@Nakda 
Ah yeah color/font format codes are problematic when forwarded to log. In case of terminal output, it is best to just copy&paste from there, though not always possible, if not accessed via SSH.
👍 Way better readable now, thanks!

Okay, locales/localepurge seems to be not the issue. Purging `dbus` as stated above would be the fast fix to go, but actually dbus "should" cause no errors. If you find time to further debug/investigate:
```
systemctl status dbus
systemctl restart dbus
systemctl daemon-reload
systemctl restart systemd-logind
systemctl status systemd-logind
```
- Just to see output, possible broken dbus service status and check if systemd service handling generally throws these errors or if other services can be treated as expected.

_Not to self: Test `localepurge`, just to be sure, also as "strange behaviour" for dpkg related programs is mentioned in package description: https://packages.debian.org/en/stretch/localepurge_

Fourdee, 2018-05-20:
@MichaIng 

Yep strange one, the only thing I can think is the following somehow requires dbus:
```
WantedBy=local-fs.target
```

Or, SystemD full install is not present on the pre-image? Unsure will need to wait for @Nakda's reply from: https://github.com/Fourdee/DietPi/issues/1787#issuecomment-390505650

MichaIng, 2018-05-20:
@Fourdee 
Ah of course, I just thought about broken dbus, but missing dbus of course could be the issue as well. But actually all my pre-images and all my running systems are dbus-free and I never faces an issue like this when treating systemd units. I add `apt install dbus` to fix test above 😉.

Nakda, 2018-05-20:
@Fourdee @MichaIng 

Thank you both for the suggestions.

Any command starting with "systemctl" did return the same error code "Failed to get D-Bus connection: No connection to service manager." So I proceed to the installation of dbus. After that, I did try again the systemctl related commands, but the output was the same error (even after rebooting, in case that matters).

Here is the output of the installation:
```
root@odroid-jessie:~# apt-get install dbus
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  ant ant-optional g++-4.6 gir1.2-folks-0.6 gir1.2-gee-0.8 gir1.2-gee-1.0
  gnome-session-fallback gstreamer0.10-ffmpeg libamd2.2.0 libapache-pom-java
  libarchive12 libavcodec53 libavformat53 libavutil51 libboost-date-time1.49.0
  libcamel-1.2-33 libclass-isa-perl libcmis-0.2-0 libcolamd2.7.1
  libcommons-beanutils-java libcommons-collections3-java
  libcommons-compress-java libcommons-digester-java libcommons-logging-java
  libcommons-parent-java libdb-java libdb-je-java libdb5.1-java
  libdb5.1-java-jni libdconf0 libdevmapper-event1.02.1 libdrm-nouveau1a
  libebackend-1.2-2 libecal-1.2-11 libedata-cal-1.2-15 libedataserver-1.2-16
  libgcr-3-1 libgd2-xpm libgdu0 libglew1.7 libgnome-bluetooth10
  libgnome-desktop-3-2 libgnutls28 libgweather-3-0 libhsqldb-java
  libicu4j-java libimobiledevice2 libjaxp1.3-java libjline-java libjson0
  libjtidy-java libkms1 libllvm3.2 liblucene2-java liblvm2app2.2 libmozjs10d
  libmpc2 libobrender27 libobt0 libparted0debian1 libpostproc52 libproxy0
  libpthread-stubs0 libregexp-java libreoffice-emailmerge libsgutils2-2
  libstdc++6-4.6-dev libswitch-perl libtiff4-dev libtiffxx0c2 libumfpack5.4.0
  libusbmuxd1 libwebp2 libx264-123 libxalan2-java libxerces2-java
  libxml-commons-external-java libxml-commons-resolver1.1-java
  libxmmsclient-glib1 libxmmsclient6 libxz-java openbox-themes
  python-aptdaemon python-aptdaemon.gtk3widgets python-defer python-gconf
  python-glade2 python-gnome2 python-gnupginterface python-notify
  python-pkg-resources python-pyorbit python-software-properties
  ttf-liberation ttf-sil-gentium-basic udisks xmms2-core xmms2-plugin-alsa
  xmms2-plugin-id3v2 xmms2-plugin-mad xmms2-plugin-vorbis xulrunner-10.0
Use 'apt-get autoremove' to remove them.
The following extra packages will be installed:
  libdbus-1-3 libdbus-1-dev libgcrypt20 libgpg-error0 libsystemd0
Suggested packages:
  rng-tools
The following NEW packages will be installed:
  libgcrypt20 libsystemd0
The following packages will be upgraded:
  dbus libdbus-1-3 libdbus-1-dev libgpg-error0
4 upgraded, 2 newly installed, 0 to remove and 1319 not upgraded.
Need to get 1,122 kB of archives.
After this operation, 800 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://ftp.debian.org/debian/ jessie/main libgpg-error0 armhf 1.17-3 [91.1 kB]
Get:2 http://ftp.debian.org/debian/ jessie/main libgcrypt20 armhf 1.6.3-2+deb8u4 [354 kB]
Get:3 http://ftp.debian.org/debian/ jessie/main libdbus-1-dev armhf 1.8.22-0+deb8u1 [191 kB]
Get:4 http://ftp.debian.org/debian/ jessie/main libdbus-1-3 armhf 1.8.22-0+deb8u1 [150 kB]
Get:5 http://ftp.debian.org/debian/ jessie/main libsystemd0 armhf 215-17+deb8u7 [83.1 kB]
Get:6 http://ftp.debian.org/debian/ jessie/main dbus armhf 1.8.22-0+deb8u1 [253 kB]
Fetched 1,122 kB in 0s (1,507 kB/s)
(Reading database ... 132280 files and directories currently installed.)
Preparing to unpack .../libgpg-error0_1.17-3_armhf.deb ...
Unpacking libgpg-error0:armhf (1.17-3) over (1.12-0.2) ...
Selecting previously unselected package libgcrypt20:armhf.
Preparing to unpack .../libgcrypt20_1.6.3-2+deb8u4_armhf.deb ...
Unpacking libgcrypt20:armhf (1.6.3-2+deb8u4) ...
Preparing to unpack .../libdbus-1-dev_1.8.22-0+deb8u1_armhf.deb ...
Unpacking libdbus-1-dev:armhf (1.8.22-0+deb8u1) over (1.6.18-2) ...
Preparing to unpack .../libdbus-1-3_1.8.22-0+deb8u1_armhf.deb ...
Unpacking libdbus-1-3:armhf (1.8.22-0+deb8u1) over (1.6.18-2) ...
Setting up libgpg-error0:armhf (1.17-3) ...
Setting up libgcrypt20:armhf (1.6.3-2+deb8u4) ...
Processing triggers for libc-bin (2.19-18+deb8u10) ...
[... 12 lines cut ...]
Installing new version of config file /etc/dbus-1/system.conf ...
Installing new version of config file /etc/init.d/dbus ...
insserv: warning: script 'K01exynos5-hwcomposer' missing LSB tags and overrides
insserv: warning: script 'exynos5-hwcomposer' missing LSB tags and overrides
[ ok ] system message bus already started; not starting..
Processing triggers for libc-bin (2.19-18+deb8u10) ...
```

Trying a purge of dbus right after that led to this:
```

root@odroid-jessie:~# apt-get purge dbus
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gsettings-desktop-schemas : Breaks: gnome-settings-daemon (< 3.10.0) but 3.8.5-2 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```

MichaIng, 2018-05-20:
@Nakda 
Thanks for testing. Ah you have a desktop image. Okay the desktops usually need dbus to work. No idea why `systemctl` does as well in your case.
Your commands did upgrade dbus as well as other system packages. What I find strange is, that `libsystemd0` got installed, thus was not installed before, even that it is base dependency of `systemd` itself 🤔. Looks like @Fourdee was right with incomplete systemd image.

So now you should be able to run the `systemctl` commands, perhaps also try a reboot to re-initiate the whole system with systemd. Hmm or better check if `systemd` is actually installed?
`apt install systemd-sysv`

Nakda, 2018-05-20:
@MichaIng 

Didn't know it was a desktop image, sorry if that would have helped. I simply grabbed the one available on the Odroid forum. 

I already did try a reboot without success, so I installed `systemd-sysv` as suggested, but after the reboot the odroid is not visible to the network so I can't access it using SSH. I will probably have to plug it to the TV tomorrow. I will get back at you as soon as possible !

```
root@odroid-jessie:~# apt-get install systemd-sysv
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  ant ant-optional g++-4.6 gir1.2-folks-0.6 gir1.2-gee-0.8 gir1.2-gee-1.0
  gnome-session-fallback gstreamer0.10-ffmpeg libamd2.2.0 libapache-pom-java
  libarchive12 libavcodec53 libavformat53 libavutil51 libboost-date-time1.49.0
  libcamel-1.2-33 libclass-isa-perl libcmis-0.2-0 libcolamd2.7.1
  libcommons-beanutils-java libcommons-collections3-java
  libcommons-compress-java libcommons-digester-java libcommons-logging-java
  libcommons-parent-java libdb-java libdb-je-java libdb5.1-java
  libdb5.1-java-jni libdconf0 libdevmapper-event1.02.1 libdrm-nouveau1a
  libebackend-1.2-2 libecal-1.2-11 libedata-cal-1.2-15 libedataserver-1.2-16
  libgcr-3-1 libgd2-xpm libgdu0 libglew1.7 libgnome-bluetooth10
  libgnome-desktop-3-2 libgnutls28 libgweather-3-0 libhsqldb-java
  libicu4j-java libimobiledevice2 libjaxp1.3-java libjline-java libjson0
  libjtidy-java libkms1 libllvm3.2 liblucene2-java liblvm2app2.2 libmozjs10d
  libmpc2 libobrender27 libobt0 libparted0debian1 libpostproc52 libproxy0
  libpthread-stubs0 libregexp-java libreoffice-emailmerge libsgutils2-2
  libstdc++6-4.6-dev libswitch-perl libsystemd-daemon0 libtiff4-dev
  libtiffxx0c2 libumfpack5.4.0 libusbmuxd1 libwebp2 libx264-123 libxalan2-java
  libxerces2-java libxml-commons-external-java libxml-commons-resolver1.1-java
  libxmmsclient-glib1 libxmmsclient6 libxz-java openbox-themes
  python-aptdaemon python-aptdaemon.gtk3widgets python-defer python-gconf
  python-glade2 python-gnome2 python-gnupginterface python-notify
  python-pkg-resources python-pyorbit python-software-properties
  ttf-liberation ttf-sil-gentium-basic udisks xmms2-core xmms2-plugin-alsa
  xmms2-plugin-id3v2 xmms2-plugin-mad xmms2-plugin-vorbis xulrunner-10.0
Use 'apt-get autoremove' to remove them.
The following extra packages will be installed:
  alsa-base initscripts kmod libglib2.0-0 libglib2.0-bin libglib2.0-dev
  libgudev-1.0-0 libkmod2 libmount1 libpam-systemd libpcre3 libpcre3-dev
  libpcrecpp0 libsmartcols1 libudev1 mount startpar systemd sysvinit-utils
  udev
Suggested packages:
  libglib2.0-doc nfs-common systemd-ui bootlogd sash
The following NEW packages will be installed:
  libsmartcols1 startpar systemd-sysv
The following packages will be upgraded:
  alsa-base initscripts kmod libglib2.0-0 libglib2.0-bin libglib2.0-dev
  libgudev-1.0-0 libkmod2 libmount1 libpam-systemd libpcre3 libpcre3-dev
  libpcrecpp0 libudev1 mount systemd sysvinit-utils udev
18 upgraded, 3 newly installed, 0 to remove and 1301 not upgraded.
Need to get 11.2 MB of archives.
After this operation, 6,573 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://ftp.debian.org/debian/ jessie/main libmount1 armhf 2.25.2-6 [153 kB                                                         ]
Get:2 http://ftp.debian.org/debian/ jessie/main libsmartcols1 armhf 2.25.2-6 [10                                                         4 kB]
Get:3 http://ftp.debian.org/debian/ jessie/main mount armhf 2.25.2-6 [165 kB]
Get:4 http://ftp.debian.org/debian/ jessie/main startpar armhf 0.59-3 [20.2 kB]
Get:5 http://ftp.debian.org/debian/ jessie/main sysvinit-utils armhf 2.88dsf-59                                                          [82.1 kB]
Get:6 http://ftp.debian.org/debian/ jessie/main libpcre3-dev armhf 2:8.35-3.3+de                                                         b8u4 [418 kB]
Get:7 http://ftp.debian.org/debian/ jessie/main libpcrecpp0 armhf 2:8.35-3.3+deb                                                         8u4 [140 kB]
Get:8 http://ftp.debian.org/debian/ jessie/main libpcre3 armhf 2:8.35-3.3+deb8u4                                                          [305 kB]
Get:9 http://ftp.debian.org/debian/ jessie/main libkmod2 armhf 18-3 [41.3 kB]
Get:10 http://ftp.debian.org/debian/ jessie/main alsa-base all 1.0.27+1 [54.3 kB                                                         ]
Get:11 http://ftp.debian.org/debian/ jessie/main kmod armhf 18-3 [78.5 kB]
Get:12 http://ftp.debian.org/debian/ jessie/main systemd-sysv armhf 215-17+deb8u                                                         7 [36.7 kB]
Get:13 http://ftp.debian.org/debian/ jessie/main libpam-systemd armhf 215-17+deb                                                         8u7 [115 kB]
Get:14 http://ftp.debian.org/debian/ jessie/main initscripts armhf 2.88dsf-59 [8                                                         7.0 kB]
Get:15 http://ftp.debian.org/debian/ jessie/main systemd armhf 215-17+deb8u7 [2,                                                         316 kB]
Get:16 http://ftp.debian.org/debian/ jessie/main udev armhf 215-17+deb8u7 [853 k                                                         B]
[... 199 lines cut ...]
Setting up libpam-systemd:armhf (215-17+deb8u7) ...
Setting up libglib2.0-0:armhf (2.42.1-1+b1) ...
Setting up libglib2.0-bin (2.42.1-1+b1) ...
[... cut ...]
Processing triggers for dbus (1.8.22-0+deb8u1) ...
Some new locales have appeared on your system:

tt@iqtelif

They will not be touched until you reconfigure localepurge
with the following command:

    dpkg-reconfigure localepurge

localepurge: Disk space freed in /usr/share/locale: 84 KiB
localepurge: Disk space freed in /usr/share/man: 4 KiB
localepurge: Disk space freed in /usr/share/gnome/help: 0 KiB
localepurge: Disk space freed in /usr/share/omf: 0 KiB

Total disk space freed by localepurge: 88 KiB
```

Nakda, 2018-05-21:
Just gave it a try this morning. Odroid still not visible on the network since I installed sysv, and I got "no signal" when plugging it to the TV. It's strange because it was working before doing a clean install. I did try to edit boot.ini, to make sure it was outputing to HDMI with a correct resolution, but it didn't help.

I'll investigate and let you know.

MichaIng, 2018-05-21:
@Nakda 
Perhaps the first thing, that you should do with the fresh image, before running DietPi-PREP:
- Cleaning the system a bid: `apt-get update && apt-mark auto gnome lxde openbox localepurge && apt-get autoremove --purge`, the more e.g. desktop related packages you can automark/autoremove, the better: `apt-mark showmanual`
- Then upgrading all packages to current version: `apt-get upgrade`
- Including changed dependencies: `apt-get dist-upgrade`
- Cleaning again: `apt-get autoremove --purge`
- Check if systemctl runs as expected: `systemctl status dbus`

DietPi-PREP basically does the same, with the only difference that it assures `en_GB.UTF-8` locales first to guarantee correct output scraping and starts a RAMdisk to add DietPi code for some basic bash message and APT related functions. Nothing special needed for this, just `systemctl` which is a basic functionality of "modern" Debian and should run on every system anyway.
- As your image is 4 years old, there seems something broken in between, or let's say the APT upgrade seems to be not possible without issues with present packages.
- Does `systemctl` actually work on the fresh image?

Nakda, 2018-05-21:
@MichaIng 
Ok, so I flashed again a fresh image, updated the kernel following the official procedure, and `systemctl` is not working.

After reaching the `apt-get dist-upgrade` step you suggested, `systemctl status` returned 

> Failed to get D-Bus connection: Unknown error -1

Should I try `apt install systemd-sysv` once again, since my system is now up-to-date?

MichaIng, 2018-05-21:
@Nakda 
Thanks for testing this. Yes Stretch is preferred, as Jessie support will slowly fade out. The distro upgrade can always cause issues though, similar to the one you faced with systemd, even that it was still Jessie 😉.
But DietPi-PREP at least minimizes risk by purging down to image to a bare minimum. Less packages means less risk that any of them will cause issues/create a non-resolvable dependency loop or something like that. Great that it went well.

If you don't mind and find the time, would you create and share your current upgraded image for other XU owners? I could offer some server for upload in case.

Nakda, 2018-05-22:
Yes, sure! Since I'm new to this, would you mind sharing how to create an image of my system? I can probably find a procedure on the Internet but is a specific tool or parameters preferred?

I'll forward you a link to download my image this week!

MichaIng, 2018-05-22:
Perhaps @Fourdee can add his image creation procedure best? 

I will also add it to: https://github.com/Fourdee/DietPi/issues/1505
And better we enable the github wiki to add info like this, as well as other static info to move then out of "issues"?

Fourdee, 2018-05-22:
@Nakda 

To save to image:
- Run through PREP and complete it
- Once PREP is completed, power off the system
- Read the SDcard to an image file (eg: win32diskimager, or, ```dd if=/dev/sda of=/myimage.img``` (where sda is SD card)
- Save the following script to a file
```
#!/bin/bash
{

	#check packages are installed
	if (( ! $(dpkg -l | grep -ci -m1 'gparted') )); then

		apt-get install gparted gpart zerofree -y

	fi

	#-------------------------------------------------------------------------------
	#MUST LEAVE 50MB+ space or automation autologin may fail due to 0 free space!!!
	#-------------------------------------------------------------------------------

	IMAGE_FP='/mnt/samba/#Backups/_Daniel/Projects/DietPi'

	IMAGE_NAME='DietPi_v6.8_RPi-ARMv6-Stretch'
	IMAGE_NAME+='.img'

	INDEX_BOOT_PART=1 #rock64 = 6 # comment out for 1 FS images and set INDEX_ROOTFS_PART=1
	INDEX_ROOTFS_PART=2 #rock64 = 7

	if [ ! -f "$IMAGE_FP/$IMAGE_NAME" ]; then

		read -p "No file found..."
		exit

	fi

	#resize2fs /dev/loop2p7 500M
	#dumpe2fs -h /dev/loop2p7 | grep 'Block count'
	#Block count * Block size / 1024 / 1024 =MB?

	modprobe loop
	losetup -f
	losetup /dev/loop2 "$IMAGE_FP/$IMAGE_NAME"
	partprobe /dev/loop2

	if [ -n "$INDEX_BOOT_PART" ]; then

		e2fsck -f /dev/loop2p$INDEX_BOOT_PART

	fi
	e2fsck -f /dev/loop2p$INDEX_ROOTFS_PART

	#Optional, copy
	# mkdir -p /mnt/loop2p1
	# mkdir -p /mnt/loop2p2
	# mount /dev/loop2p$INDEX_BOOT_PART /mnt/loop2p1
	# mount /dev/loop2p$INDEX_ROOTFS_PART /mnt/loop2p2

	# echo -e "\ncopying data to $HOME\n"
	# rm -R $HOME/loop2p1
	# rm -R $HOME/loop2p2

	# cp -R /mnt/loop2p1 $HOME/
	# cp -R /mnt/loop2p2 $HOME/
	# rm -R $HOME/loop2p1/lost+found
	# rm -R $HOME/loop2p2/lost+found

	sync

	umount /mnt/loop2p1
	umount /mnt/loop2p2

	#Resize partitions
	echo -e "\Resize partitions\n"
	gparted /dev/loop2

	sync

	#Mount for file changes, if required
	mkdir -p /mnt/loop2p1
	mkdir -p /mnt/loop2p2
	if [ -n "$INDEX_BOOT_PART" ]; then

		mount /dev/loop2p$INDEX_BOOT_PART /mnt/loop2p1

	fi
	mount /dev/loop2p$INDEX_ROOTFS_PART /mnt/loop2p2

	# echo -e "\nCopying data back\n"
	# cp -R $HOME/loop2p1/* /mnt/loop2p1/
	# cp -R $HOME/loop2p2/* /mnt/loop2p2/

	read -p "Partitions mounted to /mnt/loop2px, make changes to files if required..."

	sync

	umount /mnt/loop2p1
	umount /mnt/loop2p2

	#Resize partitions
	# echo -e "Resize partitions"
	# gparted /dev/loop2

	# sync

	read -p "If failed press CTRL+C to exit and prevent further action, else, press any key to continue..."

	if [ -n "$INDEX_BOOT_PART" ]; then

		zerofree -v /dev/loop2p$INDEX_BOOT_PART

	fi
	zerofree -v /dev/loop2p$INDEX_ROOTFS_PART

	losetup -d /dev/loop2

	END_PARTITION=$(( $(fdisk -l "$IMAGE_FP/$IMAGE_NAME" | grep ".img$INDEX_ROOTFS_PART" | awk '{print $3}') + 1 ))
	END_PARTITION=$(( 512 * ( $END_PARTITION + 128 ) )) #34 bytes for secondary GPT + safety net

	truncate --size=$END_PARTITION "$IMAGE_FP/$IMAGE_NAME"

	echo -e "\nCleaning up $HOME/loop2px\n"
	rm -R $HOME/loop2p1
	rm -R $HOME/loop2p2

	read -p "Done, press any key to continue..."

	exit

}

#ROCK 64 after reading the image AND after shrinking
#to fix:
#	GPT PMBR size mismatch (4458495 != 15523839) will be corrected by w(rite).
#	15523806
#RUN
#gdisk "$IMAGE_FP/$IMAGE_NAME"
#	w | y
#
```
- Edit the file, 
- Change ```IMAGE_FP=``` to where the directory of the saved image is (eg: ```/root```)
- Change ```IMAGE_NAME=``` to the name of the image you saved (without .img extension)
- ```chmod +x script```
- Make sure you are in a desktop env (eg: mate/lxde)
- Run script
- Follow on screen instructions
- When GPARTED loads up. Resize the last partition to the smallest size + 50MB
- Exit GPARTED and follow onscreen instructions

Once completed, you should have a image ready for writing/compressing.

Nakda, 2018-05-24:
Thanks @Fourdee ! By chance, do you got any alternative that can be run from a terminal ? I can move the SBC and plug it to the TV if needed, but CLI is more convenient.

Nakda, 2018-05-24:
@MichaIng  @Fourdee 

I played with DietPi for a bit to make sure the system runs well, and it does. 

Since I messed a bit with the system, I wanted to do a clean install before creating an image that I'll share with you. Last time, I didn't have my eMMC adapter I did flash the SD Card. Today, I wanted to follow the same procedure but targeting the eMMC. Apart from the fact that my root partition had been set to 7 GB while my eMMC is 64GB large, I encountered an error during PREP_SYSTEM_FOR_DIETPI.SH.

```
Purging configuration files for ucf (3.0030) ...
Processing triggers for libc-bin (2.19-18+deb8u10) ...
Processing triggers for install-info (5.2.0.dfsg.1-6) ...
Processing triggers for udev (215-17+deb8u7) ...
Processing triggers for ca-certificates (20141019+deb8u3) ...
Updating certificates in /etc/ssl/certs... 0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d....done.
Errors were encountered while processing:
 lightdm
E: Sub-process /usr/bin/dpkg returned an error code (1)

[FAILED] DietPi-PREP | G_AGA
[ INFO ] DietPi-PREP | If problems persist, please report this to DietPi for investigation, including a screenshot of this error! (https://github.com/Fourdee/DietPi/issues).

───────────────────────────┤ DietPi Error Handler: ├────────────────────────────

DietPi-PREP: G_AGA
 - Exit code: 100
 - DietPi version: v6.8 | HW_MODEL:22 | HW_ARCH:2 | DISTRO:3
 - Image creator:
 - Pre-image:

Log file contents:
Removing libxrender1:armhf (1:0.9.8-1+b1) ...^M
Purging configuration files for libxrender1:armhf (1:0.9.8-1+b1) ...^M
Removing perl-modules (5.20.2-3+deb8u9) ...^M
Purging configuration files for perl-modules (5.20.2-3+deb8u9) ...^M
Removing dconf-gsettings-backend:armhf (0.22.0-1) ...^M
Removing dconf-service (0.22.0-1) ...^M
Removing libpango-1.0-0:armhf (1.36.8-3) ...^M
Purging configuration files for libpango-1.0-0:armhf (1.36.8-3) ...^M
Removing fontconfig (2.11.0-6.3+deb8u1) ...^M

                                     <Ok>

────────────────────────────────────────────────────────────────────────────────
```

After selecting "OK" here is the full detail:

```
#### Details:
- Date           | Thu 24 May 13:35:08 CEST 2018
- DietPi version | v6.8
- Img creator    |
- Pre-image      |
- SBC device     | NULL (index=22)
- Distro         | jessie (index=3)
- Command        | G_AGA
- Exit code      | 100
- Software title | DietPi-PREP

#### Expected behaviour:
<!-- What SHOULD be happening? -->

#### Actual behaviour:
<!-- What IS happening? -->

#### Steps to reproduce:
<!-- Explain how to reproduce the issue -->

#### Additional Information:
<!-- if applicable -->

#### Additional logs:

Log file contents:
Removing libxrender1:armhf (1:0.9.8-1+b1) ...
Purging configuration files for libxrender1:armhf (1:0.9.8-1+b1) ...
Removing perl-modules (5.20.2-3+deb8u9) ...
Purging configuration files for perl-modules (5.20.2-3+deb8u9) ...
Removing dconf-gsettings-backend:armhf (0.22.0-1) ...
Removing dconf-service (0.22.0-1) ...
Removing libpango-1.0-0:armhf (1.36.8-3) ...
Purging configuration files for libpango-1.0-0:armhf (1.36.8-3) ...
Removing fontconfig (2.11.0-6.3+deb8u1) ...
Purging configuration files for fontconfig (2.11.0-6.3+deb8u1) ...
Removing libfontconfig1:armhf (2.11.0-6.3+deb8u1) ...
Purging configuration files for libfontconfig1:armhf (2.11.0-6.3+deb8u1) ...
Removing fontconfig-config (2.11.0-6.3+deb8u1) ...
Purging configuration files for fontconfig-config (2.11.0-6.3+deb8u1) ...
Removing fonts-liberation (1.07.4-1) ...
Removing libthai0:armhf (0.1.21-1) ...
Purging configuration files for libthai0:armhf (0.1.21-1) ...
Removing libdatrie1:armhf (0.2.8-1) ...
Purging configuration files for libdatrie1:armhf (0.2.8-1) ...
Removing libdconf1:armhf (0.22.0-1) ...
Purging configuration files for libdconf1:armhf (0.22.0-1) ...
Removing libexpat1:armhf (2.1.0-6+deb8u4) ...
Purging configuration files for libexpat1:armhf (2.1.0-6+deb8u4) ...
Removing libfreetype6:armhf (2.5.2-3+deb8u2) ...
Purging configuration files for libfreetype6:armhf (2.5.2-3+deb8u2) ...
Removing libglib2.0-0:armhf (2.42.1-1+b1) ...
Purging configuration files for libglib2.0-0:armhf (2.42.1-1+b1) ...
Removing libpng12-0:armhf (1.2.50-2+deb8u3) ...
Purging configuration files for libpng12-0:armhf (1.2.50-2+deb8u3) ...
Removing libthai-data (0.1.21-1) ...
Removing libx11-6:armhf (2:1.6.2-3+deb8u1) ...
Purging configuration files for libx11-6:armhf (2:1.6.2-3+deb8u1) ...
Removing libx11-data (2:1.6.2-3+deb8u1) ...
Removing libxcb1:armhf (1.10-3+b1) ...
Purging configuration files for libxcb1:armhf (1.10-3+b1) ...
Removing libxau6:armhf (1:1.0.8-1) ...
Purging configuration files for libxau6:armhf (1:1.0.8-1) ...
Removing libxdmcp6:armhf (1:1.1.1-1+b1) ...
Purging configuration files for libxdmcp6:armhf (1:1.1.1-1+b1) ...
Removing ucf (3.0030) ...
Purging configuration files for ucf (3.0030) ...
Processing triggers for libc-bin (2.19-18+deb8u10) ...
Processing triggers for install-info (5.2.0.dfsg.1-6) ...
Processing triggers for udev (215-17+deb8u7) ...
Processing triggers for ca-certificates (20141019+deb8u3) ...
Updating certificates in /etc/ssl/certs... 0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d....done.
Errors were encountered while processing:
 lightdm
E: Sub-process /usr/bin/dpkg returned an error code (1)

--------------------------------------------------------------------

[FAILED] DietPi-PREP | Unable to continue, DietPi-PREP will now terminate.
```

Fourdee, 2018-05-25:
@Nakda 

> By chance, do you got any alternative that can be run from a terminal ? I can move the SBC and plug it to the TV if needed, but CLI is more convenient.

Not currently, I've not found a way to automate the gparted resize yet.

>  I encountered an error during PREP_SYSTEM_FOR_DIETPI.SH.

Strange, try killing the process then manually removing it:
```
killall -w lightdm
apt-get purge lightdm
```

If not errors, re-reun prep again, should go through.

Nakda, 2018-05-25:
@Fourdee  

Ok, thanks. Will try it tomorrow ! 

Just to be sure, while upgrading I got contextual messages like this one:
https://ibb.co/e4tRvo

Basically it tells me that "update-passwd" has found differences between my system and Debian defaults. In order to make this system image as compatible as possible to everyone needs, is the best practice to answer "yes" or "no". I answered "yes" the first time I did the process because I was worried about compatibility issues, but now I prefer to ask.

Nakda, 2018-05-26:
@Fourdee 

Killing the process and removing it worked! DietPi is successfully installed on the emmc.

I did try to create an image of the system, but I'm stuck following your procedure.

> When GPARTED loads up. Resize the last partition to the smallest size + 50MB

The last partition displayed by gparted is root, so I can't resize it. 
[image]

It does not detect the full capacity of the emmc card (64GB) so does it matter? Should I simply send you the Win32DiskImager image?

Fourdee, 2018-05-26:
@Nakda 

> It does not detect the full capacity of the emmc card (64GB) so even if I could resize there would be no sufficient space.

Strange, our resize script should do this during 1st run of the device. We'll need to debug, please paste results:
```
cat /var/tmp/dietpi/logs/fs_partition_resize.log
```

Nakda, 2018-05-26:
@Fourdee 

Here is the log output:

```
root@DietPi:~# cat /var/tmp/dietpi/logs/fs_partition_resize.log

Welcome to fdisk (util-linux 2.29.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.
```

MichaIng, 2018-05-29:
@Nakda @Fourdee 
Our first run script just extends drives with /dev/mmcblk or /dev/sda name scheme, but yours is /dev/loop?

You can go through the fdisk steps manually: https://github.com/Fourdee/DietPi/blob/update-rootfs-merge/rootfs/var/lib/dietpi/services/fs_partition_resize.sh#L48-L66
But be careful, as this can break all data on the drive.
```
fdisk /dev/loop2                //enters partitioning console
p             //will show you the current partition table
d             //will delete a partition to choose
2             //choose second partition to delete
n             //create new partition
p             //set it up as primary partition (should be also offered as default)
2             //give it partition number 2 (should be also offered as default)
              //It should ask you at which sector the partition should start. The offered default should be located just some sectors behind the end of the first partition, so just hit <return> here.
              //Hit return again, to choose the default last sector, which will be the last available.
              //When manually executed it might be necessary to hit again enter to remove ext4 flag or something.
p             //Just to check again the partition table
w             //Write partition table to disk, if everything is fine.
q             //Exit fdisk console
partprobe /dev/loop2           //Inform kernel about partition changes
resize2fs /dev/loop2p2         //Extend the file system to use full new partition size
```

MichaIng, 2018-06-18:
Further research about `/dev/loop` devices: https://en.wikipedia.org/wiki/Loop_device

Hmm, I am not familiar with those kind of fs devices, but if I understand it correctly, there need to be somewhere image files containing them. Confusing that there are no other devices found (or just not mounted after boot?) that contain those image files. Also I guess in this case they cannot be expended, as the fs size should be fixed within the image file.

MichaIng, 2018-07-29:
Initial issue is solved, right? Drive manager detection and fs rewrite script have been reworked in between. DietPi-PREP now needs systemd(-sysv) as pre-requirement. Images which come without systemd, are either too old or too much customized from default Debian/Raspbian. We do not want to automatically apply an init system change, which thus in case needs to be applied by user first, before running DietPi-PREP.

closed 2018-07-29 by MichaIng