MichaIng/DietPi #1788: DietPi-Globals | G_AGP: Tiny output format issue

MichaIng, 2018-05-20:
Just a tiny visual:

Observed this already on all systems and distros, VMs and RPi, seems to be an issue in how G_DIETPI-NOTIFY handles the APT removal output. See the large gap in front of `After this operation` and doubled `(Reading database ...)`. This exactly shows up the same on every G_AGP:
```
The following packages will be REMOVED:
  libasound2* libasound2-data* libfftw3-single3* libgomp1* libncursesw5*
  libsamplerate0* libtinfo5*
0 upgraded, 0 newly installed, 7 to remove and 0 not upgraded.
                                                              After this operation, 6,826 kB disk space will be freed.
(Reading database ... 17317 files and directories currently installed.)                                               (Reading database ...
```

Fourdee, 2018-05-20:
@MichaIng 

Yep, confirmed the extra space.
```
root@DietPi:~# G_AGP mc
[ INFO ] Install verified: mc
[  OK  ] APT removal for: mc , please wait...
Reading package lists...
Building dependency tree...
Reading state information...
The following packages were automatically installed and are no longer required:
  libglib2.0-0 mc-data
Use 'apt autoremove' to remove them.
The following packages will be REMOVED:
  mc*
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
                                                              After this operati                                                                 on, 1,307 kB disk space will be freed.
(Reading database ... 17357 files and directories currently installed.)
Removing mc (3:4.8.18-1) ...
(Reading database ... 17267 files and directories currently installed.)
Purging configuration files for mc (3:4.8.18-1) ...
```

Reading database double is normal, APT re-reading the database after finishing the command:
```
root@DietPi:~# apt-get purge mc -y
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libglib2.0-0 mc-data
Use 'apt autoremove' to remove them.
The following packages will be REMOVED:
  mc*
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
After this operation, 1,307 kB disk space will be freed.
(Reading database ... 17357 files and directories currently installed.)
Removing mc (3:4.8.18-1) ...
(Reading database ... 17267 files and directories currently installed.)
Purging configuration files for mc (3:4.8.18-1) ...
root@DietPi:~#
```

MichaIng, 2018-05-22:
Hmm of course we do not pipe APT output through G_DIETPI-NOTIFY, just echo colour code around and run non-interactive... Will do some tests about which step causes the format issue.

Otherwise ignore, it's really just a tiny one.

MichaIng, 2018-06-19:
@Fourdee 
Found `tee` to be the problem:
```
root@VM-Stretch:~# apt-get -y install iputils-ping | tee output.txt
Reading package lists...
Building dependency tree...
Reading state information...
Recommended packages:
  libcap2-bin
The following NEW packages will be installed:
  iputils-ping
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
                                                              Need to get 0 B/57.5 kB of archives.
                                                                                                  After this operation, 114 kB of additional disk space will be used.
                                                                                                                                                                     Selecting previously unselected package iputils-ping.
(Reading database ... 16420 files and directories currently installed.)
Preparing to unpack .../iputils-ping_3%3a20161105-1_amd64.deb ...
Unpacking iputils-ping (3:20161105-1) ...
Setting up iputils-ping (3:20161105-1) ...
Setcap is not installed, falling back to setuid
```
- The same format issue with upgrade and purge, in something gets actually removed or installed.
- Color, DEBIAN_FRONTEND/noninteractive, `apt` instead of `apt-get` and other options have no effect.
- Sadly I didn't find something about it via web research, thus no solution besides not using `tee`.
- Only idea I have (besides leaving as is, as it's just minor visual), to completely forward output to file and just show `G_DIETPI-NOTIFY -2` processing message instead.

MichaIng, 2018-11-06:
I opened a ticket on Debian bug tracker: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=913083
- Lets see if the maintainers/devs pay attention on such a minor visual-only issue 🙂. Since we found other `tee` related output issues, perhaps all of them can be solved at once, e.g. by using another terminal dimension estimation method.

MichaIng, 2019-02-24:
Issue actually resolved with apt-get `-q`/`-qq` options (that we use now in most cases), which prevents processing related output:
```
root@VM-Stretch:/tmp# apt-get -qq install net-tools | tee
Selecting previously unselected package net-tools.
(Reading database ... 20225 files and directories currently installed.)
Preparing to unpack .../net-tools_1.60+git20161116.90da8a0-1_amd64.deb ...
Unpacking net-tools (1.60+git20161116.90da8a0-1) ...
Setting up net-tools (1.60+git20161116.90da8a0-1) ...

root@VM-Stretch:/tmp# apt-get -qq purge net-tools | tee
(Reading database ... 20283 files and directories currently installed.)
Removing net-tools (1.60+git20161116.90da8a0-1) ...

root@VM-Stretch:/tmp# apt-get -qq install net-tools
Selecting previously unselected package net-tools.
(Reading database ... 20225 files and directories currently installed.)
Preparing to unpack .../net-tools_1.60+git20161116.90da8a0-1_amd64.deb ...
Unpacking net-tools (1.60+git20161116.90da8a0-1) ...
Setting up net-tools (1.60+git20161116.90da8a0-1) ...

root@VM-Stretch:/tmp# apt-get -qq purge net-tools
(Reading database ... 20283 files and directories currently installed.)
Removing net-tools (1.60+git20161116.90da8a0-1) ...

root@VM-Stretch:/tmp# G_AGI net-tools
[  OK  ] Root access verified.
[ INFO ] APT installation for: net-tools, please wait...
Selecting previously unselected package net-tools.
(Reading database ... 20225 files and directories currently installed.)
Preparing to unpack .../net-tools_1.60+git20161116.90da8a0-1_amd64.deb ...
Unpacking net-tools (1.60+git20161116.90da8a0-1) ...
Setting up net-tools (1.60+git20161116.90da8a0-1) ...
[  OK  ] G_AGI net-tools

root@VM-Stretch:/tmp# G_AGP net-tools
[ INFO ] Install verified: net-tools
[ INFO ] APT removal for: net-tools , please wait...
(Reading database ... 20283 files and directories currently installed.)
Removing net-tools (1.60+git20161116.90da8a0-1) ...
[  OK  ] G_AGP net-tools
```

closed 2019-02-24 by MichaIng