MichaIng/DietPi #1791: RPi + LXDE | LXAppearance hangs on dbus-launch

Fourdee, 2018-05-21:
> https://github.com/Fourdee/DietPi/issues/1558#issuecomment-390332722

MichaIng, 2019-09-17:
Just verified this is still an issue on current Buster image with DietPi v6.26.0 (latest `dev` branch)...
Debugging planned for v6.27.

MichaIng, 2019-10-08:
Not related to #1558, on Buster pcmanfm gets upgraded at latest with next APT upgrade, both issues are present independent of each others...

MichaIng, 2019-10-08:
Okay, works well on local access. I always tested via VNC before.
Confirmed being an issue on VNC virtual desktop only, testing both in parallel.

MichaIng, 2019-10-08:
TigerVNC-only issue, works well on RealVNC virtual desktop.

MichaIng, 2020-09-24:
Report on RPi repo: https://github.com/RPi-Distro/repo/issues/202
However, I'll close the issue here and track all LXDE-on-RPi issues now here instead: #1558

closed 2020-09-24 by MichaIng

MichaIng, 2020-09-30:
Solution: `apt install dbus-user-session` (https://github.com/MichaIng/DietPi/issues/1558#issuecomment-701547904)