MichaIng/DietPi #1792: "Welcome messages" crash SCP

Furfel, 2018-05-23:
#### Required Information:
- DietPi Version 6.7
- SBC Device RPi Zero W
- Power supply used USB 20W PSU
- SD card used Unknown (4D)
- Distro Linux DietPi 4.14.34+ #1110 Mon Apr 16 14:51:42 BST 2018 armv6l GNU/Linux

#### Additional Information (if applicable):
- Software title: SCP
- Can this issue be replicated on a fresh installation of DietPi? Dunno.

#### Expected behaviour:
Sending file happily by scp

#### Actual behaviour:
tput: unknown terminal "unknown"
/DietPi/dietpi/func/dietpi-globals: line 297: ((: lines=(38+6)/ : syntax error: operand expected (error token is "/ ")
tput: unknown terminal "unknown"
\033[K\033[90m[\033[0m\033[32m  OK  \033[0m\033[90m]\033[0m Root access verified.

#### Steps to reproduce:
Try sending file with scp

#### Did you submit a ```dietpi-bugreport```?
No.

#### Extra details:
USB OTG g_ether but don't think this matters

MichaIng, 2018-05-23:
@Furfel 
Thanks for your report.

As you are on version 6.7, can you verify, that the calling code for the message/globals is within `/root/.bashrc`?
```
. /DietPi/dietpi/func/dietpi-globals
/DietPi/dietpi/login
```

I will do myself later, but could you try to add `[ -z "$PS1" ] && return` in front of the DietPi related lines? I had the impression earlier that .bashrc is not only called for interactive terminals.
[ -z "$TERM" ] does not work obviously, as this is already present within globals: https://github.com/Fourdee/DietPi/blob/testing/dietpi/func/dietpi-globals#L30

Furfel, 2018-05-24:
@MichaIng 
Yeah, both login and globals are there, they execute normally when logging via ssh.

@Fourdee 
From: OpenSSH_7.4p1 scp client on Raspbian
To: OpenSSH_7.4p1 server on DietPi 6.7
`echo "hello world" > helloworld.txt`
`scp helloworld.txt dietpi@10.1.1.12:/home/dietpi`

Fourdee, 2018-05-25:
#### Test on v6.9 dev branch ('ll test v6.8 master once I've setup a fresh install machine to test):
- 🈯️ DietPi to DietPi system
```
root@DietPi:~# echo "hello world" > helloworld.txt
root@DietPi:~# scp helloworld.txt dietpi@192.168.0.113:/home/dietpi
The authenticity of host '192.168.0.113 (192.168.0.113)' can't be established.
ECDSA key fingerprint is SHA256:xkbVZ/wCGVnL1MWQZC/S70DZ0DNbt6AKjGe+GkhLtpU.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added '192.168.0.113' (ECDSA) to the list of known hosts.
dietpi@192.168.0.113's password:
helloworld.txt                                100%   12    11.1KB/s   00:00

#DietPi user
dietpi@DietPi:~$ echo "hello world" > helloworld.txt
dietpi@DietPi:~$ scp helloworld.txt dietpi@192.168.0.113:/home/dietpi
The authenticity of host '192.168.0.113 (192.168.0.113)' can't be established.
ECDSA key fingerprint is SHA256:xkbVZ/wCGVnL1MWQZC/S70DZ0DNbt6AKjGe+GkhLtpU.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added '192.168.0.113' (ECDSA) to the list of known hosts.
dietpi@192.168.0.113's password:
helloworld.txt                                100%   12    24.2KB/s   00:00

```

Notes, 
- possible issue with Raspbian as client?
- Resolved for v6.9?

Fourdee, 2018-05-25:
@Furfel 

Any chance you can test current v6.9, to see if the issue is resolved?
- Create a backup ```dietpi-backup 1```
- Update to latest v6.9 testing branch ```G_DEV_1```
- reboot system and retest

MichaIng, 2018-05-27:
@Furfel
We just switched to a new login script method that should not be called on non-interactive shell with no terminal ($TERM) available. This new system on testing branch needs testing with all file transfer software but is meant to solve your issue: https://github.com/Fourdee/DietPi/pull/1798

Fourdee, 2018-05-31:
This should be resolved in v6.9 update, i'll mark as closed. Please reopen if required.

closed 2018-05-31 by Fourdee