MichaIng/DietPi #1794: Rock64 | Implement GPU driver install

Fourdee, 2018-05-24:
> https://dietpi.com/phpbb/viewtopic.php?f=9&t=3604

Fourdee, 2018-05-24:
Packages:
```
xserver-xorg-video-armsoc
libdrm-rockchip1
libmali-rk-utgard-450-r7p0
#vdpauinfo
```

Xorg.conf
```
Section "Device"
        Identifier      "Mali FBDEV"
        Driver          "armsoc"
        Option          "fbdev"                 "/dev/fb0"
        Option          "Fimg2DExa"             "false"
        Option          "DRI2"                  "true"
        Option          "DRI2_PAGE_FLIP"        "false"
        Option          "DRI2_WAIT_VSYNC"       "true"
        Option          "CursorPlaneType"       "2"
EndSection

Section "Screen"
        Identifier      "DefaultScreen"
        Device          "Mali FBDEV"
        DefaultDepth    24
EndSection
```

Fourdee, 2018-05-25:
https://launchpad.net/~ayufan/+archive/ubuntu/rock64-ppa/
🈴 Lacks ARM64 packages

Fourdee, 2018-05-25:
https://github.com/ayufan-rock64/linux-build/issues/111#issuecomment-370261364
- Requires Ubuntu:
- GPU packages not available in http://deb.ayufan.eu/orgs/ayufan-rock64/releases
```
E: Package 'gnomine' has no installation candidate
E: Package 'chromium-browser' has no installation candidate
E: Unable to locate package libmali-rk-utgard-2th-r7p0
E: Unable to locate package libdrm-rockchip1
E: Unable to locate package gstreamer1.0-rockchip1
E: Couldn't find any package by glob 'gstreamer1.0-rockchip1'
E: Couldn't find any package by regex 'gstreamer1.0-rockchip1'
E: Unable to locate package ubuntu-mate-core
E: Unable to locate package ubuntu-mate-desktop
E: Unable to locate package ubuntu-mate-lightdm-theme
E: Unable to locate package ubuntu-mate-wallpapers-xenial
```

Fourdee, 2018-07-09:
Not possible, at least from my end.

Marking as closed.

purist69, 2018-07-22:
Libreelec was able to get this working....

josemahj, 2019-10-04:
there are some new with this?

MichaIng, 2019-10-05:
@josemahj 
Could you please try the following (this is Ayufans Rock64 repo which now has arm64 packages): https://launchpad.net/~ayufan/+archive/ubuntu/rock64-0.9-ppa
```
apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 06AA3A141B8C919401EF634BDAFCA20FBF428671
echo 'deb http://ppa.launchpad.net/ayufan/rock64-0.9-ppa/ubuntu disco main' > /etc/apt/sources.list.d/dietpi-rock64-gpu.list
cat << _EOF_ > /etc/apt/preferences.d/dietpi-rock64-gpu
Package: *
Pin: release o=LP-PPA-ayufan-rock64-0.9-ppa
Pin-Priority: 990
_EOF_
apt-get update
apt install libmali-rk-utgard-450-r7p0
cat _EOF_ > /etc/X11/xorg.conf
Section "Device"
    Identifier  "Rockchip Graphics"
    Driver      "modesetting"
    Option      "AccelMethod"    "glamor"
    Option      "DRI"            "2"
    Option      "Dri2Vsync"      "true"
    Option      "PageFlip"       "false"
EndSection
_EOF_
```
- This installs the base Mali driver for Rock64.

Then it depends on which software you use. E.g. the following should install/reinstall the X.org Xserver with hardware acceleration:
```
apt install --reinstall xserver-common xserver-xorg-core
```
Or simply run `apt full-upgrade` to check for updates of installed packages which should be all presented on Stretch.
Additional GPU-accelerated packages are available as well, check the link above.

josemahj, 2019-10-05:
@MichaIng I get 

```The following packages have unmet dependencies:
 libmali-rk-utgard-450-r7p0 : Depends: libdrm2 (>= 2.3.1) but it is not going to be installed``` when try install driver... 

Then I try libdrm2 and ``` 
The following packages have unmet dependencies:
 libdrm2 : Depends: libc6 (>= 2.29) but 2.24-11+deb9u4 is to be installed   ```

MichaIng, 2019-10-05:
@josemahj 
Thanks, let me have a look, there is a bionic repo as well. `Depends: libc6 (>= 2.17)` that should work.
Please try:
```
echo 'deb http://ppa.launchpad.net/ayufan/rock64-0.8-ppa/ubuntu bionic main' > /etc/apt/sources.list.d/dietpi-rock64-gpu.list
cat << _EOF_ > /etc/apt/preferences.d/dietpi-rock64-gpu
Package: *
Pin: release o=LP-PPA-ayufan-rock64-0.8-ppa
Pin-Priority: 990
_EOF_
apt update
apt install libmali-rk-utgard-450-r7p0
cat _EOF_ > /etc/X11/xorg.conf
Section "Device"
    Identifier  "Rockchip Graphics"
    Driver      "modesetting"
    Option      "AccelMethod"    "glamor"
    Option      "DRI"            "2"
    Option      "Dri2Vsync"      "true"
    Option      "PageFlip"       "false"
EndSection
_EOF_
```
- But I already see that there are other packages that depend on `libc6 >= 2.27`, so e.g. ffmpeg must not be installed from there.

So we can only guarantee that this install works when we provide those packages individually instead of from the repo, so we know the dependencies are fulfilled by Debian Stretch. Finally a new Buster-based Rock64 image is required anyway which at least then fully supports the Bionic repo, sadly still not the Disco repo, since libc6 v2.28 is shipped.

josemahj, 2019-10-07:
fail again
```
root@DietPi:/var/lib/dpkg/info# apt install libmali-rk-utgard-450-r7p0
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias
Leyendo la información de estado... Hecho
Se instalarán los siguientes paquetes NUEVOS:
  libmali-rk-utgard-450-r7p0
0 actualizados, 1 nuevos se instalarán, 0 para eliminar y 3 no actualizados.
Se necesita descargar 0 B/752 kB de archivos.
Se utilizarán 3.176 kB de espacio de disco adicional después de esta operación.
(Leyendo la base de datos ... 28529 ficheros o directorios instalados actualmente.)
Preparando para desempaquetar .../libmali-rk-utgard-450-r7p0_1.6-3ayufan5_arm64.deb ...
Desempaquetando libmali-rk-utgard-450-r7p0:arm64 (1.6-3ayufan5) ...
dpkg: error al procesar el archivo /var/cache/apt/archives/libmali-rk-utgard-450-r7p0_1.6-3ayufan5_arm64.deb (--unpack):
 intentando sobreescribir `/usr/lib/aarch64-linux-gnu/libEGL.so.1', que está también en el paquete libegl1-mesa:arm64 13.0.6-1+b2
Se encontraron errores al procesar:
 /var/cache/apt/archives/libmali-rk-utgard-450-r7p0_1.6-3ayufan5_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2019-10-08:
@josemahj 
Ah damn, this due to a package difference between Ubuntu (what this repo is made for) and Debian. On Ubuntu `libegl1-mesa` is just a transitional package for `libegl1` and itself does not contain any library files. On Debian this change is done since Buster, but not yet on Stretch.

Okay but it's available as transitional package via backports. Please do:
```
apt purge libmali-rk-utgard-450-r7p0
apt install libegl1/stretch-backports
apt install libmali-rk-utgard-450-r7p0
```

Good to know about this package transition, will implement it into dietpi-software as well as the backports install of `libegl1`.

josemahj, 2019-10-08:
clean install v 6.25.3

```
echo 'deb http://ppa.launchpad.net/ayufan/rock64-0.8-ppa/ubuntu bionic main' > /etc/apt/sources.list.d/dietpi-rock64-gpu.list
cat << _EOF_ > /etc/apt/preferences.d/dietpi-rock64-gpu
Package: *
Pin: release o=LP-PPA-ayufan-rock64-0.8-ppa
Pin-Priority: 990
_EOF_
apt update
apt install libmali-rk-utgard-450-r7p0
```
sucessfully installed

`/etc/X11/xorg.conf` `No such file or directory`

MichaIng, 2019-10-08:
@josemahj 
Little typo above, it should be:
```
cat << _EOF_ > /etc/X11/xorg.conf
Section "Device"
    Identifier  "Rockchip Graphics"
    Driver      "modesetting"
    Option      "AccelMethod"    "glamor"
    Option      "DRI"            "2"
    Option      "Dri2Vsync"      "true"
    Option      "PageFlip"       "false"
EndSection
_EOF_
```

josemahj, 2019-10-09:
ok, everything is done. Now, try to install ffmpeg

MichaIng, 2019-10-09:
@josemahj 
I guess ffmpeg from this repo is not possible, again due to incompatibility with libc6 from Debian repo. I guess you need to remove the repo again and go with ffmpeg from Debian. I hope it uses the installed GPU drivers. Would be also interesting in how far the desktop (and other Xserver-based applications) benefits from the driver/xorg.conf.

josemahj, 2020-04-21:
could 
this https://github.com/rockchip-linux/rk-rootfs-build/tree/master/packages/arm64/libmali
or
this https://forum.armbian.com/topic/9310-rk3328-media-script-rock64-renegade/
help?

MichaIng, 2020-04-21:
@josemahj 
The first links does not contain the required Mali 450 (Utgard) drivers, which are only available for armhf: https://github.com/rockchip-linux/rk-rootfs-build/tree/master/packages/armhf/libmali
See my post above, probably the armhf package, after `dpkg --add-architecture armhf`, will work, but requires testing.
The Armbian media script should work, but is very outdated.

gprst, 2020-05-23:
Hi, I would be willing to help for testing, I typed in the `dpkg --add-architecture armhg` command, but I am very new to this kind of architecture-tweaking thing, could you tell me which `.deb` file to install, and how to test if it works ?

MichaIng, 2020-05-25:
@gprst 
The test would be this:
1. Install a desktop, e.g. via `dietpi-software` or manually.
2. Assure glxgears and es2gears are installed: `apt install mesa-utils mesa-utils-extra`
3. Test `glxgears` and `es2gears` for frame rates and it they work at all.
4. Install the mentioned Mali GPU driver:
```
dpkg --add-architecture armhf
cd /tmp
wget https://github.com/rockchip-linux/rk-rootfs-build/raw/master/packages/armhf/libmali/libmali-rk-utgard-450-r7p0_1.7-1_armhf.deb
dpkg -i libmali-rk-utgard-450-r7p0_1.7-1_armhf.deb
rm libmali-rk-utgard-450-r7p0_1.7-1_armhf.deb
```
5. Test again `glxgears` and `es2gears` for frame rates and compare with results above.

gprst, 2020-05-25:
Okay ; so here's how it went :
- I had an average of about 145 FPS with `glxgears` and about 155 FPS with `es2gears` on the first run
- I got the following error when running the second `dpkg` command :
```
dpkg: dependency problems prevent configuration of libmali-rk-utgard-450-r7p0:armhf:
 libmali-rk-utgard-450-r7p0:armhf depends on libc6 (>= 2.4).
 libmali-rk-utgard-450-r7p0:armhf depends on libdrm2 (>= 2.3.1).
 libmali-rk-utgard-450-r7p0:armhf depends on libx11-6 (>= 2:1.4.99.1).
 libmali-rk-utgard-450-r7p0:armhf depends on libxdamage1 (>= 1:1.1).
 libmali-rk-utgard-450-r7p0:armhf depends on libxext6.
 libmali-rk-utgard-450-r7p0:armhf depends on libxfixes3.

dpkg: error processing package libmali-rk-utgard-450-r7p0:armhf (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 libmali-rk-utgard-450-r7p0:armhf
```
- I tried to install `libc6` or `libdrm2` but got told that these packages were already on there newest versions,  2.24-11+deb9u4 for `libc6` and 2.4.74-1 for `libdrm2`, but also on both tries have been suggested to type in `apt --fix-broken install`, so I typed in the command, and the required dependencies got installed ! :tada: 
- on the second run, I got :
  - an average of about 144 FPS with `glxgears`, that then rose up to about 160 FPS, then about 150 FPS, then I stopped
  - an average of about 165 FPS with `es2gears`

I don't know if the frame rate difference is significant, nor if I actually got the Mali GPU driver to work. Could you tell ?

MichaIng, 2020-05-25:
@gprst 
Okay many thanks for testing. To the armhf package seems to depend on the armhf versions of the libraries as well. On your arm64 system the arm64 versions are installed. Quick result is that we have no pre-compiled Mali drivers for Rock64 currently. We could compile them ourself or use the quite outdated version from the Armbian community.

Someone with a bid more experience in Linux graphics and compiling and a Rock64 board would be very helpful. I can only provide the theoretical steps but especially when compiling usually one runs into several errors due to missing development/header packages and such until figuring out everything, especially since RockChip sources are not aimed for Debian only but Ubuntu as well.

MichaIng, 2020-07-11:
Okay, instead of messing around with the proprietary driver, we'll go for FOSS mesa, which starts to support Mali450 with newest version: https://wiki.debian.org/PanfrostLima

I'll have a look if I can compile those for Debian Buster arm64 either for ROCK64 in particular or to include generic support for all our supported SoCs or even x64_86?: https://docs.mesa3d.org/install.html

If anyone can or wants to help, highly welcome.

Here the build log to check/compare compile time options and results: https://buildd.debian.org/status/fetch.php?pkg=mesa&arch=arm64&ver=20.1.2-1&stamp=1593073012&raw=0

Another idea would be to provide an early Bullseye image for that device.

MichaIng, 2020-08-13:
NanoPi NEO3 uses the same SoC, hence requires the same GPU support: http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO3#Hardware_Spec

MichaIng, 2021-06-02:
I'm gonna mark this as closed. Finally with Debian Bullseye, many SoCs will get much better GPU support with Mesa drivers, shipped natively with the Debian repository. Any attempt to use 3rd party drivers/libraries is problematic as it means that matching development headers need to be used as well for software compilation to assure it works correctly. It will solve itself the next months, as we'll ship Bullseye images for all SBCs shortly after its official release.

closed 2021-06-02 by MichaIng