MichaIng/DietPi #1796: Nextcloud certificate end problem

ludji49, 2018-05-26:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version 6.8
- SBC Device (EG: RPi 3) RPi 3
- Power supply used EG: 5v 1A RAVpower
- SD card used  Sandisk ultra 32
- Distro Jessie

#### Additional Information (if applicable):
- Software title?certificate nextcloud 
- Can this issue be replicated on a fresh installation of DietPi :non

#### Expected behaviour:
 nextcloud certificate end problem
#### Actual behaviour:
non

#### Steps to reproduce:
non

#### Did you submit a ```dietpi-bugreport```? non
 non

#### Extra details:

[image]

MichaIng, 2018-05-27:
@ludji49 
You have a `letsencrypt` certificate I assume and as far as I remember from your last issue? 3 months duration fits to this.

Seems like the automated renewal is not active.

Please try to run `dietpi-letsencrypt 1`.

ludji49, 2018-05-27:
here is the result

[image]

MichaIng, 2018-05-27:
@ludji49 
Ah I remember we changed your webserver root last time to have Nextcloud available directly at domain, instead of at domain/nextcloud. Yes as mentioned already that time this causes some incompatibility with DietPi.

But this one is easy to fix. DietPi-LetsEncrypt tries to use /var/www as webroot where yours is at /var/www/nextcloud. Let's try to call certbot directly with this adjudged. On next certificate renewal, thus setting our saved:
`certbot renew -a webroot -w /var/www/nextcloud`

MichaIng, 2018-05-29:
@Ludji49 
Jep looking good. As mentioned, end of Juli the certbot cron job will perform the certificate renewal automatically with the new settings, so nothing more to do about this. 
I will mark as closed.

closed 2018-05-29 by MichaIng