MichaIng/DietPi #1799: Odroid XU4 stretch image - HDMI not working

rasugo, 2018-05-27:
## Creating a bug report/issue:

Downloaded yesterday the image DietPi_OdroidXU4-ARMv7-Stretch and put it on eMMC module and SD card. No HDMI output.

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)?
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

To see the login screen.

#### Actual behaviour:

Black screen.

#### Steps to reproduce:

Restore the image on eMMC module and start oDroid.

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

Tried the same with an SD card ==> same behaviour.
Tried the HDMI_safe setting ==> same behaviour

Restored a previous image DietPi_OdroidXU4-ARMv7-Jessie (downloaded mid-April, version 6.1 of Dietpi) the startup performs as ex expected and I get the login screen.

Fourdee, 2018-05-27:
@rasugo 

Thanks for the report, we'll try to replicate 👍

Fourdee, 2018-05-28:
@rasugo 

Working fine in our local testing
[image]

Please verify:
- HDMI cable working
- on the SD card edit ```/boot/boot.ini``` try DVI mode ```setenv vout "dvi" ```, then boot
- on the SD card edit ```/boot/boot.ini``` try a lower resolution, eg ```setenv videoconfig "drm_kms_helper.edid_firmware=edid/1280x720.bin"```, then boot

rasugo, 2018-05-28:
The HDMI cable is working - using the Jessie image, it works out of the box.
With the above two settings it works !

closed 2018-05-28 by rasugo