MichaIng/DietPi #1800: Missing libjpeg-dev for Octoprint Plugin

ichbestimmtnicht, 2018-05-27:
#### Required Information:
- [ INFO ] DietPi-Update | Current Version : 6.8
- SBC Device  RPi 3b
- Power supply used 5v 2,5A
- SD card used Sandisk ultra 10th grade 32GB
- Distro `Linux mary 4.14.37-v7+ #1111 SMP Thu Apr 26 13:56:59 BST 2018 armv7l GNU/Linux`

#### Additional Information (if applicable):
- Octoprint
- yes 

#### Expected behaviour:
The Plugin get installed

#### Actual behaviour:
`The headers or library files could not be found for jpeg,a required dependency when compiling Pillow from source.Please see the install instructions at:https://pillow.readthedocs.io/en/latest/installation.htmlTraceback (most recent call last):File "<string>", line 1, in <module>File "/tmp/pip-install-91H7ot/Pillow/setup.py", line 806, in <module>raise RequiredDependencyException(msg)__main__.RequiredDependencyException:The headers or library files could not be found for jpeg,a required dependency when compiling Pillow from source.Please see the install instructions at:https://pillow.readthedocs.io/en/latest/installation.html----------------------------------------Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-91H7ot/Pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-W0MWmg/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-91H7ot/Pillow/Fehler!Could not parse output from pip, see plugin_pluginmanager_console.log for generated output`

#### Steps to reproduce:
- setup dietpi 
- with octoprint 
- firsttimesetup octoprint keep everything on.
- try to install Astroprintcloud Plugin through Pluginmanager
#### Did you submit a ```dietpi-bugreport```?
no

#### Extra details:
`sudo apt-get install libjpeg-dev -y` fixed the problem Please ad this to the requirements.

Fourdee, 2018-05-27:
@ichbestimmtnicht 

Thanks for the report 👍 

Although OctoPrint does not require this package to run stock, <1MB for this package, I see no harm in adding this to the install by default, to assist with additional end user plugin installations.
```
root@DietPi:~# apt-get install libjpeg-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following package was automatically installed and is no longer required:
  libopts25
Use 'apt autoremove' to remove it.
The following additional packages will be installed:
  libjpeg62-turbo libjpeg62-turbo-dev
The following NEW packages will be installed:
  libjpeg-dev libjpeg62-turbo libjpeg62-turbo-dev
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 351 kB of archives.
After this operation, 890 kB of additional disk space will be used.
```

Fourdee, 2018-05-27:
```libjpeg-dev``` will be installed automatically for v6.9.

Completed.

closed 2018-05-27 by Fourdee