MichaIng/DietPi #1803: Start failed

valerivp, 2018-05-28:
Orange Pi i96 on Debian after instalation DietPi (with out errors), and reboot wrote:

Please login to continue

But not show command promt

## Creating a bug report/issue:

#### Required Information:
v6.7
Orange Pi i96 Cortex A5

#### Additional Information (if applicable):

#### Expected behaviour:
promt for login

#### Actual behaviour:
nothing

#### Steps to reproduce:
install DietPi on board

#### Did you submit a ```dietpi-bugreport```?
dont know

#### Extra details:
[image]

valerivp, 2018-05-29:
@Fourdee 
it's not work
[image]

Fourdee, 2018-05-30:
@valerivp 

Ah I see. Serial console is disabled by default after 1st run setup, hence the lock.

You'll need to enable serial console in ```/boot/dietpi.txt``` ( and ```/DietPi/dietpi.txt```, to cover if systemD runs stop on a service that isn't in started state, unsure) before 1st run. Edit the file after PREP has finished, and, before you reboot the system.
```
CONFIG_SERIAL_CONSOLE_ENABLE=1
```

msdos, 2019-02-06:
@Fourdee I just had the problem you described in https://github.com/Fourdee/DietPi/issues/1803#issuecomment-393249909. Shoul this be added as a comment somehow in https://github.com/Fourdee/DietPi/blob/eb1be60e89ddc651d80345946c1b7f0585494bcc/dietpi.txt#L178?