MichaIng/DietPi #1810: WiFi HotSpot Issue [dhcpd: receive_packet failed on wlan0: Network is down]

UnrealEngineer, 2018-05-31:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | _v6.8_
- SBC Device | _RPi 3 Model B+ (armv7l)_
- Power supply | _5V, 2.4A_
- SD card | _Transcend 8GB_
- Distro | _Linux DietPi 4.14.34-v7+ 1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux_

#### Expected behaviour:
Stable WiFi AP Network

#### Actual behaviour:
WiFi device gets IP from DHCP but then:
_isc-dhcp-server fails: receive_packet failed on wlan0: Network is down_
No ping from 192.168.42.1 to 192.168.42.11:
_From 192.168.42.1 icmp_seq=1 Destination Host Unreachable_

#### Steps to reproduce:
- install WiFi HotSpot software from ```dietpi-software```
- enable & config WiFi from ```dietpi-config```
- connect device to DietPi WiFi AP

#### Did you submit a ```dietpi-bugreport```?
no

#### Extra details:

isc-dhcp-server logs:
```
May 31 11:03:22 DietPi systemd[1]: Starting LSB: DHCP server...
May 31 11:03:22 DietPi isc-dhcp-server[14402]: DHCPv4 interfaces are no longer set by the INTERFACES variable in
May 31 11:03:22 DietPi isc-dhcp-server[14402]: /etc/default/isc-dhcp-server.  Please use INTERFACESv4 instead.
May 31 11:03:22 DietPi isc-dhcp-server[14402]: Migrating automatically for now, but this will go away in the future.
May 31 11:03:22 DietPi isc-dhcp-server[14402]: Launching IPv4 server only.
May 31 11:03:22 DietPi dhcpd[14416]: Internet Systems Consortium DHCP Server 4.3.5
May 31 11:03:22 DietPi dhcpd[14416]: Copyright 2004-2016 Internet Systems Consortium.
May 31 11:03:22 DietPi dhcpd[14416]: All rights reserved.
May 31 11:03:22 DietPi dhcpd[14416]: For info, please visit https://www.isc.org/software/dhcp/
May 31 11:03:22 DietPi dhcpd[14416]: Wrote 2 leases to leases file.
May 31 11:03:22 DietPi dhcpd[14419]: Server starting service.
May 31 11:03:24 DietPi isc-dhcp-server[14402]: Starting ISC DHCPv4 server: dhcpd.
May 31 11:03:24 DietPi systemd[1]: Started LSB: DHCP server.
May 31 11:03:27 DietPi dhcpd[14419]: receive_packet failed on wlan0: Network is down
May 31 11:03:50 DietPi dhcpd[14419]: receive_packet failed on wlan0: Network is down
May 31 11:04:12 DietPi dhcpd[14419]: receive_packet failed on wlan0: Network is down
May 31 11:04:35 DietPi dhcpd[14419]: receive_packet failed on wlan0: Network is down
May 31 11:04:58 DietPi dhcpd[14419]: receive_packet failed on wlan0: Network is down
May 31 11:05:16 DietPi dhcpd[14419]: DHCPDISCOVER from 12:81:2a:cf:78:e0 (orangepi) via wlan0
May 31 11:05:17 DietPi dhcpd[14419]: DHCPOFFER on 192.168.42.11 to 12:81:2a:cf:78:e0 (orangepi) via wlan0
May 31 11:05:17 DietPi dhcpd[14419]: DHCPREQUEST for 192.168.42.11 (192.168.42.1) from 12:81:2a:cf:78:e0 (orangepi) via wlan0
May 31 11:05:17 DietPi dhcpd[14419]: DHCPACK on 192.168.42.11 to 12:81:2a:cf:78:e0 (orangepi) via wlan0
May 31 11:05:20 DietPi dhcpd[14419]: receive_packet failed on wlan0: Network is down
May 31 11:05:43 DietPi dhcpd[14419]: receive_packet failed on wlan0: Network is down
May 31 11:06:06 DietPi dhcpd[14419]: receive_packet failed on wlan0: Network is down
May 31 11:06:29 DietPi dhcpd[14419]: receive_packet failed on wlan0: Network is down
May 31 11:06:51 DietPi dhcpd[14419]: receive_packet failed on wlan0: Network is down
May 31 11:07:14 DietPi dhcpd[14419]: receive_packet failed on wlan0: Network is down
May 31 11:07:37 DietPi dhcpd[14419]: receive_packet failed on wlan0: Network is down
May 31 11:07:59 DietPi dhcpd[14419]: receive_packet failed on wlan0: Network is down
May 31 11:08:22 DietPi dhcpd[14419]: receive_packet failed on wlan0: Network is down
May 31 11:08:45 DietPi dhcpd[14419]: receive_packet failed on wlan0: Network is down
May 31 11:09:08 DietPi dhcpd[14419]: receive_packet failed on wlan0: Network is down
May 31 11:09:30 DietPi dhcpd[14419]: receive_packet failed on wlan0: Network is down
May 31 11:09:40 DietPi dhcpd[14419]: DHCPREQUEST for 192.168.42.11 from 12:81:2a:cf:78:e0 (orangepi) via wlan0
May 31 11:09:40 DietPi dhcpd[14419]: DHCPACK on 192.168.42.11 to 12:81:2a:cf:78:e0 (orangepi) via wlan0
May 31 11:09:51 DietPi dhcpd[14419]: reuse_lease: lease age 11 (secs) under 25% threshold, reply with unaltered, existing lease for 192.168.42.11
May 31 11:09:51 DietPi dhcpd[14419]: DHCPREQUEST for 192.168.42.11 from 12:81:2a:cf:78:e0 (orangepi) via wlan0
May 31 11:09:51 DietPi dhcpd[14419]: DHCPACK on 192.168.42.11 to 12:81:2a:cf:78:e0 (orangepi) via wlan0
May 31 11:09:53 DietPi dhcpd[14419]: receive_packet failed on wlan0: Network is down
[... 118 lines cut ...]
wpa_pairwise=TKIP
rsn_pairwise=CCMP
ieee80211n=1
-------------------------------------------------------------------------------------------------------------------

root@DietPi:~# cat /etc/iptables.ipv4.nat
# Generated by iptables-save v1.6.0 on Thu May 31 10:06:09 2018
*filter
:INPUT ACCEPT [0:0]
:FORWARD ACCEPT [0:0]
:OUTPUT ACCEPT [0:0]
-A FORWARD -i eth0 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT
-A FORWARD -i wlan0 -o eth0 -j ACCEPT
COMMIT
# Completed on Thu May 31 10:06:09 2018
# Generated by iptables-save v1.6.0 on Thu May 31 10:06:09 2018
*nat
:PREROUTING ACCEPT [0:0]
:INPUT ACCEPT [0:0]
:OUTPUT ACCEPT [2:1038]
:POSTROUTING ACCEPT [2:1038]
-A POSTROUTING -o eth0 -j MASQUERADE
COMMIT
# Completed on Thu May 31 10:06:09 2018
```

Fourdee, 2018-05-31:
@UnrealEngineer 

Thanks for the report 👍 

Onboard WiFi, or, USB?

If USB WiFi, please paste results:
```
lsusb
```

UnrealEngineer, 2018-06-01:
@Fourdee 
I use onboard WiFi.
```dmesg | grep wlan``` returns nothing.

Fourdee, 2018-06-03:
Tests RPi 3B+
- WiFi HotSpot
- Tor Hotspot (https://github.com/Fourdee/DietPi/issues/1692#issuecomment-379542526)

Fourdee, 2018-06-03:
@UnrealEngineer 

There was a minor issue with ```dietpi-config```, however, this won't be the cause of issue. Aside from that, all your config settings and output values match my test system.

Can you please confirm: 
- RPi3B+ is the board used?
- PSU brand/make/model

Please paste the results of the following:
```
dmesg | grep brcm
```

UnrealEngineer, 2018-06-04:
Sorry for waiting. Have no access to internet.
I am using Raspberry Pi 3 Model B+ with Grand-X CH-45 5V 2.4A 2USB PSU
I have made some measurements of power consumption during WiFi device connection: U = 5.05-5.08 V, I = 0.80-0.86 A.
```
Jun 04 08:25:25 DietPi dhcpd[1063]: receive_packet failed on wlan0: Network is down
Jun 04 08:25:32 DietPi dhcpd[1063]: DHCPDISCOVER from 12:81:2a:cf:78:e0 via wlan0
Jun 04 08:25:33 DietPi dhcpd[1063]: DHCPOFFER on 192.168.42.11 to 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:25:34 DietPi dhcpd[1063]: DHCPREQUEST for 192.168.42.11 (192.168.42.1) from 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:25:34 DietPi dhcpd[1063]: DHCPACK on 192.168.42.11 to 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:25:42 DietPi dhcpd[1063]: reuse_lease: lease age 9 (secs) under 25% threshold, reply with unaltered, existing lease for 192.168.42.11
Jun 04 08:25:42 DietPi dhcpd[1063]: DHCPREQUEST for 192.168.42.11 from 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:25:42 DietPi dhcpd[1063]: DHCPACK on 192.168.42.11 to 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:25:48 DietPi dhcpd[1063]: receive_packet failed on wlan0: Network is down
Jun 04 08:26:10 DietPi dhcpd[1063]: receive_packet failed on wlan0: Network is down
Jun 04 08:26:33 DietPi dhcpd[1063]: receive_packet failed on wlan0: Network is down
Jun 04 08:26:56 DietPi dhcpd[1063]: receive_packet failed on wlan0: Network is down
Jun 04 08:27:19 DietPi dhcpd[1063]: receive_packet failed on wlan0: Network is down
Jun 04 08:27:41 DietPi dhcpd[1063]: receive_packet failed on wlan0: Network is down
Jun 04 08:28:04 DietPi dhcpd[1063]: receive_packet failed on wlan0: Network is down
Jun 04 08:28:27 DietPi dhcpd[1063]: receive_packet failed on wlan0: Network is down
Jun 04 08:28:49 DietPi dhcpd[1063]: receive_packet failed on wlan0: Network is down
Jun 04 08:29:12 DietPi dhcpd[1063]: receive_packet failed on wlan0: Network is down
Jun 04 08:29:35 DietPi dhcpd[1063]: receive_packet failed on wlan0: Network is down
Jun 04 08:29:40 DietPi dhcpd[1063]: DHCPREQUEST for 192.168.42.11 from 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:29:40 DietPi dhcpd[1063]: DHCPACK on 192.168.42.11 to 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:29:45 DietPi dhcpd[1063]: reuse_lease: lease age 5 (secs) under 25% threshold, reply with unaltered, existing lease for 192.168.42.11
Jun 04 08:29:45 DietPi dhcpd[1063]: DHCPREQUEST for 192.168.42.11 from 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:29:45 DietPi dhcpd[1063]: DHCPACK on 192.168.42.11 to 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:29:57 DietPi dhcpd[1063]: reuse_lease: lease age 17 (secs) under 25% threshold, reply with unaltered, existing lease for 192.168.42.11
Jun 04 08:29:57 DietPi dhcpd[1063]: DHCPREQUEST for 192.168.42.11 from 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:29:57 DietPi dhcpd[1063]: DHCPACK on 192.168.42.11 to 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:29:57 DietPi dhcpd[1063]: receive_packet failed on wlan0: Network is down
Jun 04 08:30:20 DietPi dhcpd[1063]: receive_packet failed on wlan0: Network is down
Jun 04 08:30:39 DietPi dhcpd[1063]: reuse_lease: lease age 59 (secs) under 25% threshold, reply with unaltered, existing lease for 192.168.42.11
Jun 04 08:30:39 DietPi dhcpd[1063]: DHCPREQUEST for 192.168.42.11 from 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:30:39 DietPi dhcpd[1063]: DHCPACK on 192.168.42.11 to 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:30:43 DietPi dhcpd[1063]: receive_packet failed on wlan0: Network is down
```

```
root@DietPi:~# dmesg | grep brcm
[    3.394448] brcmfmac: F1 signature read @0x18000000=0x15264345
[    3.403949] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43455-sdio.bin for chip 0x004345(17221) rev 0x000006
[    3.404789] usbcore: registered new interface driver brcmfmac
[    3.715153] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[    3.715834] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 9.10.105 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2018-03-09 18:56:28
[    4.077173] brcmfmac: power management disabled
[    4.225743] brcmfmac: power management disabled
[   15.449119] brcmfmac: power management disabled

root@DietPi:~# ifdown wlan0
root@DietPi:~# ifup wlan0
ifup: interface wlan0 already configured
```

Fourdee, 2018-06-04:
@UnrealEngineer 

Thanks 👍 

Very strange.

Ok, so everything appears to be functional, until the OrangePi obtains an IP address, then something is causing the wlan0 adapter to fail (OPi sending invalid packets causing the failure?).
```
Jun 04 08:25:32 DietPi dhcpd[1063]: DHCPDISCOVER from 12:81:2a:cf:78:e0 via wlan0
Jun 04 08:25:33 DietPi dhcpd[1063]: DHCPOFFER on 192.168.42.11 to 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:25:34 DietPi dhcpd[1063]: DHCPREQUEST for 192.168.42.11 (192.168.42.1) from 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:25:34 DietPi dhcpd[1063]: DHCPACK on 192.168.42.11 to 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:25:42 DietPi dhcpd[1063]: reuse_lease: lease age 9 (secs) under 25% threshold, reply with unaltered, existing lease for 192.168.42.11
Jun 04 08:25:42 DietPi dhcpd[1063]: DHCPREQUEST for 192.168.42.11 from 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:25:42 DietPi dhcpd[1063]: DHCPACK on 192.168.42.11 to 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:25:48 DietPi dhcpd[1063]: receive_packet failed on wlan0: Network is down
```

- We need to check ```hostapd``` for status/errors when ``` receive_packet failed on wlan0: Network is down``` is reported
-- ```systemctl status hostapd -l```
- Also, does the issue occur with a different device connected to hotspot, and, OrangePi disabled (eg: smartphone which I used in my local testing)?

UnrealEngineer, 2018-06-05:
Such situation with all WiFi devices: OPi, smartphone, laptop and ESP8266 chip.

```
isc-dhcp-server.service
Jun 05 07:59:33 DietPi dhcpd[1083]: receive_packet failed on wlan0: Network is down
Jun 05 07:59:56 DietPi dhcpd[1083]: receive_packet failed on wlan0: Network is down
Jun 05 08:00:03 DietPi dhcpd[1083]: DHCPDISCOVER from 12:81:2a:cf:78:e0 via wlan0
Jun 05 08:00:04 DietPi dhcpd[1083]: DHCPOFFER on 192.168.42.11 to 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 05 08:00:04 DietPi dhcpd[1083]: Wrote 2 leases to leases file.
Jun 05 08:00:04 DietPi dhcpd[1083]: DHCPREQUEST for 192.168.42.11 (192.168.42.1) from 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 05 08:00:04 DietPi dhcpd[1083]: DHCPACK on 192.168.42.11 to 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 05 08:00:18 DietPi dhcpd[1083]: receive_packet failed on wlan0: Network is down
Jun 05 08:00:41 DietPi dhcpd[1083]: receive_packet failed on wlan0: Network is down

hostapd.service
Jun 05 07:59:21 DietPi hostapd[12591]: Starting advanced IEEE 802.11 management: hostapd.
Jun 05 07:59:21 DietPi systemd[1]: Started LSB: Advanced IEEE 802.11 management daemon.
Jun 05 08:00:03 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 IEEE 802.11: associated
Jun 05 08:00:03 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 RADIUS: starting accounting session E4C67DC0-00000000
Jun 05 08:00:03 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 WPA: pairwise key handshake completed (RSN)
Jun 05 08:00:18 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 IEEE 802.11: disassociated
Jun 05 08:00:18 DietPi hostapd[12596]: wlan0: STA c9:03:da:07:c9:03 IEEE 802.11: disassociated
Jun 05 08:00:31 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 IEEE 802.11: disassociated
Jun 05 08:00:33 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 IEEE 802.11: associated
Jun 05 08:00:33 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 RADIUS: starting accounting session E4C67DC0-00000001
Jun 05 08:00:33 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 WPA: pairwise key handshake completed (RSN)
Jun 05 08:00:40 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 IEEE 802.11: disassociated
Jun 05 08:00:41 DietPi hostapd[12596]: wlan0: STA e3:01:e3:01:e3:01 IEEE 802.11: disassociated
Jun 05 08:00:52 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 IEEE 802.11: disassociated
Jun 05 08:00:54 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 IEEE 802.11: associated
Jun 05 08:00:54 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 RADIUS: starting accounting session E4C67DC0-00000002
Jun 05 08:00:54 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 WPA: pairwise key handshake completed (RSN)
Jun 05 08:01:03 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 IEEE 802.11: disassociated
Jun 05 08:01:04 DietPi hostapd[12596]: wlan0: STA e1:01:e1:01:e0:01 IEEE 802.11: disassociated
Jun 05 08:01:06 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 IEEE 802.11: disassociated
Jun 05 08:01:08 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 IEEE 802.11: associated
Jun 05 08:01:08 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 RADIUS: starting accounting session E4C67DC0-00000003
Jun 05 08:01:08 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 WPA: pairwise key handshake completed (RSN)
Jun 05 08:01:26 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 IEEE 802.11: disassociated
Jun 05 08:01:26 DietPi hostapd[12596]: wlan0: STA d0:03:f5:c7:d0:03 IEEE 802.11: disassociated
Jun 05 08:01:44 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 IEEE 802.11: disassociated
Jun 05 08:01:46 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 IEEE 802.11: associated
Jun 05 08:01:46 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 RADIUS: starting accounting session E4C67DC0-00000004
Jun 05 08:01:46 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 WPA: pairwise key handshake completed (RSN)
Jun 05 08:01:49 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 IEEE 802.11: disassociated
Jun 05 08:01:49 DietPi hostapd[12596]: wlan0: STA e3:01:e3:01:e3:01 IEEE 802.11: disassociated
```

Fourdee, 2018-06-05:
@UnrealEngineer 

OrangePi device seems to be the trigger, power off the OrangePi and see if the issue reoccurs:
```
Jun 05 08:00:03 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 IEEE 802.11: associated
Jun 05 08:00:03 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 RADIUS: starting accounting session E4C67DC0-00000000
Jun 05 08:00:03 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 WPA: pairwise key handshake completed (RSN)
Jun 05 08:00:18 DietPi hostapd[12596]: wlan0: STA 12:81:2a:cf:78:e0 IEEE 802.11: disassociated

---

Jun 04 08:25:42 DietPi dhcpd[1063]: DHCPACK on 192.168.42.11 to 12:81:2a:cf:78:e0 (orangepi) via wlan0
Jun 04 08:25:48 DietPi dhcpd[1063]: receive_packet failed on wlan0: Network is down
```

UnrealEngineer, 2018-06-06:
Logs after reboot. OrangePi and other devices disconnected. Than I connected by smartphone and situations was the same.
```
root@DietPi:~# journalctl -f -n 100 -u hostapd.service
-- Logs begin at Wed 2018-06-06 14:03:41 EEST. --
Jun 06 14:04:05 DietPi systemd[1]: Starting LSB: Advanced IEEE 802.11 management daemon...
Jun 06 14:04:06 DietPi hostapd[1197]: Starting advanced IEEE 802.11 management: hostapd.
Jun 06 14:04:06 DietPi systemd[1]: Started LSB: Advanced IEEE 802.11 management daemon.
Jun 06 14:15:01 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: associated
Jun 06 14:15:01 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b RADIUS: starting accounting session D69407       8A-00000000
Jun 06 14:15:01 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b WPA: pairwise key handshake completed (RSN       )
Jun 06 14:15:02 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: disassociated
Jun 06 14:15:02 DietPi hostapd[1226]: wlan0: STA c9:03:d7:a7:c9:03 IEEE 802.11: disassociated
Jun 06 14:15:06 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: disassociated
Jun 06 14:15:07 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: disassociated
Jun 06 14:15:13 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: associated
Jun 06 14:15:13 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b RADIUS: starting accounting session D69407       8A-00000001
Jun 06 14:15:13 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b WPA: pairwise key handshake completed (RSN       )
Jun 06 14:15:25 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: disassociated
Jun 06 14:15:25 DietPi hostapd[1226]: wlan0: STA e3:01:e3:01:e3:01 IEEE 802.11: disassociated
Jun 06 14:15:36 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: disassociated
Jun 06 14:15:36 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: disassociated
Jun 06 14:15:36 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: disassociated
Jun 06 14:15:39 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: associated
Jun 06 14:15:39 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b RADIUS: starting accounting session D69407       8A-00000002
Jun 06 14:15:39 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b WPA: pairwise key handshake completed (RSN       )
Jun 06 14:15:47 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: disassociated
Jun 06 14:15:48 DietPi hostapd[1226]: wlan0: STA e1:01:e1:01:e0:01 IEEE 802.11: disassociated
Jun 06 14:15:49 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: disassociated
Jun 06 14:15:52 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: associated
Jun 06 14:15:52 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b RADIUS: starting accounting session D69407       8A-00000003
Jun 06 14:15:52 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b WPA: pairwise key handshake completed (RSN       )
Jun 06 14:16:10 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: disassociated
Jun 06 14:16:10 DietPi hostapd[1226]: wlan0: STA ce:03:c6:47:ce:03 IEEE 802.11: disassociated
Jun 06 14:16:12 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: disassociated
Jun 06 14:16:12 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: disassociated
Jun 06 14:16:12 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: disassociated
Jun 06 14:16:15 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: associated
Jun 06 14:16:15 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b RADIUS: starting accounting session D69407       8A-00000004
Jun 06 14:16:15 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b WPA: pairwise key handshake completed (RSN       )
Jun 06 14:16:33 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: disassociated
Jun 06 14:16:33 DietPi hostapd[1226]: wlan0: STA e3:01:e3:01:e3:01 IEEE 802.11: disassociated
Jun 06 14:16:34 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: disassociated
Jun 06 14:16:34 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: disassociated
Jun 06 14:16:37 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: associated
Jun 06 14:16:37 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b RADIUS: starting accounting session D69407       8A-00000005
Jun 06 14:16:37 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b WPA: pairwise key handshake completed (RSN       )
Jun 06 14:16:41 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: disassociated
Jun 06 14:16:41 DietPi hostapd[1226]: wlan0: STA 10:2a:b3:7f:0f:0b IEEE 802.11: disassociated

root@DietPi:~# journalctl -f -n 100 -u isc-dhcp-server.service
-- Logs begin at Wed 2018-06-06 14:03:41 EEST. --
Jun 06 14:03:50 DietPi systemd[1]: Starting LSB: DHCP server...
Jun 06 14:03:51 DietPi isc-dhcp-server[1041]: DHCPv4 interfaces are no longer set by the INTERFACES variable        in
Jun 06 14:03:51 DietPi isc-dhcp-server[1041]: /etc/default/isc-dhcp-server.  Please use INTERFACESv4 instead.
Jun 06 14:03:51 DietPi isc-dhcp-server[1041]: Migrating automatically for now, but this will go away in the f       uture.
Jun 06 14:03:51 DietPi isc-dhcp-server[1041]: Launching IPv4 server only.
Jun 06 14:03:51 DietPi dhcpd[1057]: Internet Systems Consortium DHCP Server 4.3.5
Jun 06 14:03:51 DietPi dhcpd[1057]: Copyright 2004-2016 Internet Systems Consortium.
[... cut ...]
Jun 06 14:18:49 DietPi dhcpd[1066]: receive_packet failed on wlan0: Network is down
Jun 06 14:19:12 DietPi dhcpd[1066]: receive_packet failed on wlan0: Network is down
Jun 06 14:19:35 DietPi dhcpd[1066]: receive_packet failed on wlan0: Network is down
Jun 06 14:19:58 DietPi dhcpd[1066]: receive_packet failed on wlan0: Network is down
Jun 06 14:20:20 DietPi dhcpd[1066]: receive_packet failed on wlan0: Network is down
Jun 06 14:20:43 DietPi dhcpd[1066]: receive_packet failed on wlan0: Network is down
```

UnrealEngineer, 2018-06-14:
I've tested WiFi HotSpot with new RPi 3B+ and everything was fine! Seems it is some hardware issue with first one RPi.

closed 2018-06-14 by UnrealEngineer

4hidessh, 2021-03-21:
thank to help