MichaIng/DietPi #1811: Can't use Pihole and Baikal on the same dietpi-system

Krawei, 2018-06-01:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version 6.8
- SBC Device: RPi 3
- Power supply used: 5v 1A Power supply
- SD card used: Sandisk ultra
- Distro: Stretch

#### Additional Information (if applicable):
- PiHole
- Baikal

#### Expected behaviour:
http://<IP of RPi>/admin => Pihole Web GUI
http://<IP of Rpi>/baikal/admin => Baikal Web GUI

#### Actual behaviour:
HTTP-Request to the Baikal-URL mentioned above => Delivery of the Pihole standard blocking page

#### Steps to reproduce:
Install DietPi
Install Pihole
Install Baikal
Try to access the baikal page

#### Did you submit a ```dietpi-bugreport```?
no

#### Extra details:
When requesting the Baikal-URL, I see the Pihole "standard blocking page". It seems that there's kinda "redirect everything except Pihole-Web-GUI-Request to Pihole blocking page". But there aren't any redirect orders in the lighttp.conf.

Weird.

MichaIng, 2018-06-01:
@Krawei
Thanks for your report. I will try to reproduce.
It the blocking page colorized or just a white pain text page?

We will get this fixed in case for v6.9 soon

Krawei, 2018-06-01:
@MichaIng : It is the colorized one. I'll try to upload a screenshot later.

Fourdee, 2018-06-01:
Testing notes:
- Unable to replicate on ARMv8 (cant see it being any different for ARMv7), both pages load fine.

@Krawei 

Did you install PiHole and Baikal through ```dietpi-software```?

MichaIng, 2018-06-01:
@Krawei @Fourdee 
I couldn't replicate as well on x86 VM. First installed pi-hole, baikal afterwards.

Could you also show result of `cat /etc/lighttpd/lighttpd.conf`? It seems like backup and recover of DietPi's lighttpd.conf didn't went well and the pi-hole edited version is in place, which blocks all other web applications: https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L4476-L4499
So yeah, it's important you install pi-hole via dietpi-software.

Krawei, 2018-06-04:
@Fourdee : I've installed pihole and baikal via dietpie-software and following the instructions from your site: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=50#p1502   

When I'm trying to access the baikal admin page, my dietpi serves this site:

[image] (this site also returns when I'm trying to access http://192.168.2.2/phpinfo.php).

My lighttpd.conf:

https://privatebin.net/?e4a8033af3854dcc#lFtsZOh3QSp25uFqWds+vLDxxnjy94UFlhKzkw7Q1D4=

Hope you find my issue :).

Fourdee, 2018-06-04:
@Krawei 

Yep, PiHole has overwritten the lighttpd config file. We have code in the ```dietpi-software``` install part which prevents this, so unsure as to why this occurred for you.

The next PiHole release will allow us to avoid this issue.

#### In the mean time, to resolve, run the following commands:
```
dietpi-services stop
wget https://dietpi.com/downloads/testing/lighttpd.conf -O /etc/lighttpd/lighttpd.conf
dietpi-services start
```

Fourdee, 2018-06-13:
Marking as closed in favor of https://github.com/Fourdee/DietPi/issues/1696.

Please reopen if required.

closed 2018-06-13 by Fourdee