MichaIng/DietPi #1814: Odroid C2 | Kodi stuttering

Fourdee, 2018-06-02:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)?
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

Fourdee, 2018-06-03:
Fresh install Odroid C2 Stretch + Kodi and default options, samba server source:
- 🈯️ 720p 60Hz is fine
- 🈺 1080p 60Hz slight stuttering x264
- 1080p 24Hz, excessive vsync tearing.

Interesting, Kodi 16 under Jessie was fine from what I remember.

@oparoz 

Until a solution is found, a workaround is to try the Jessie image with Kodi 16:
https://dietpi.com/downloads/images/DietPi_OdroidC2-ARMv8-Jessie.7z

MichaIng, 2018-06-03:
@Fourdee 
Hmm, wasn't there something with GPU support not yet fully implemented with Odroid Stretch? Okay seems resolved, but I remember reading that the solution was a bid hackish, maybe not providing same performance. Need again a closer look into the odroid forum: https://github.com/Fourdee/DietPi/issues/1552

Gabba35, 2018-06-10:
Having the same problem. Xorg uses >100% cpu and the temp has been hitting the +80C routinely. Sometimes when idle. Had a similar problem with Sonarr, Jackett and Mono.

MichaIng, 2018-11-29:
@Fourdee
Finally Stretch C2 Kodi issue seems fixed by Meveric: https://dietpi.com/phpbb/viewtopic.php?f=11&t=5282&p=15568&sid=16232ea848e9194e0db86b2fcdb96e5c#p15545

Perhaps we need to adjust our install script accordingly?

Fourdee, 2018-11-29:
@MichaIng 

> Update should be fine, it's part of the kodi-odroid package itself

No change needed. Updated APT package resolves issue ^^.

MichaIng, 2018-11-29:
@Fourdee 
Perfect. If we now can resolve the XU4 Xorg issue (https://github.com/Fourdee/DietPi/issues/2130), I would say we can take off Jessie images, or move them to some "legacy" subdir.

From my point of view, we can also disable the Jessie option within DietPi-PREP (still allow dist upgrade from base Jessie to DietPi Stretch). And encourage existing Jessie users to switch to Stretch.

When all Jessie image/creation sources are off, we can watch install counts on survey page and by times drop support completely. Would allow massive code simplifications.

MichaIng, 2018-12-06:
@Fourdee
I believe we can close this, or do you want to run further tests?

Fourdee, 2019-01-02:
@MichaIng 

> I believe we can close this, or do you want to run further tests?

Yep its resolved, verified with setting up my dads Odroid C2 + Kodi system 👍

closed 2019-01-02 by Fourdee