MichaIng/DietPi #1815: 从Debian系统安装DIEPT不能成功

zzlinwq, 2018-06-02:
## Creating a bug report/issue:
Please excuse my poor English
机器是amlogic S905设备，
Linux amlogic 3.14.29 #137 SMP PREEMPT Thu May 17 09:26:05 MSK 2018 aarch64 GNU/Linux
Welcome to ARMBIAN 5.44 user-built Debian GNU/Linux 9 (stretch) 3.14.29 
是按下面帖子命令操作的
https://github.com/Fourdee/DietPi/issues/1285#issue-280771944
命令如下：
export GIT_BRANCH='master' 
#export GIT_BRANCH='testing' #DEV only

wget https://raw.githubusercontent.com/Fourdee/DietPi/$GIT_BRANCH/PREP_SYSTEM_FOR_DIETPI.sh -O PREP_SYSTEM_FOR_DIETPI.sh
chmod +x PREP_SYSTEM_FOR_DIETPI.sh
sudo GIT_BRANCH=$GIT_BRANCH ./PREP_SYSTEM_FOR_DIETPI.sh
但是结果不能成功：
[FAILED] DietPi-PREP | Unable to continue, DietPi-PREP will now terminate.
是什么问题？如何解决？

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)?
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

zzlinwq, 2018-06-02:
DietPi Error Handler:
DietPi-PREP: systemctl stop dietpi-ramlog
  Exit code: 5
  DietPi version: v6.8 I HW_ MODEL :0 I HW_ ARCH:3 I DISTRO:4Image creator:Pre-image:
Log file contents:
Failed to stop dietpi-ramlog.service: Unit dietpi- ramlog.service not loaded.
If problems persist, please report this to DietPi for investigation, including ascreenshot of this error! (https :l/github .com/Fourdee/DietPi/ issues).
Unable to continue, DietPi-PREP will now terminate.
  <0k>

Fourdee, 2018-06-02:
@zzlinwq 

Looks like an issue pre-hw selection.

Could be locale did not get applied, please paste results:
```
locale
```

And lets check service it mentions, please paste results:
```
systemctl status dietpi-ramlog -l
```

- Which image ```.img``` are you using (before converting it to DietPi)?
-

zzlinwq, 2018-06-02:
Which image .img are you using (before converting it to DietPi)?Armbian_5.44_S9xxx_Debian_stretch_3.14.29_icewm_20180515.img
@Fourdee

zzlinwq, 2018-06-02:
Which device is this running on (eg: Odroid C2)? 斐讯天天链N1
http://www.phicomm.com/cn/index.php/Products/lian_spec/cateid/27/id/136.html

zzlinwq, 2018-06-02:
[image]

https://user-images.githubusercontent.com/36132083/40873952-b9c24f84-669b-11e8-800a-d2ff102e44dc.png

MichaIng, 2018-06-02:
@zzlinwq @Fourdee
Hmm, maybe somehow DietPi system was detected?
Does dietpi-ramdisk exist? `systemctl status dietpi-ramdisk`

zzlinwq, 2018-06-02:
[image]
root@amlogic:~# systemctl status dietpi-ramdisk
● dietpi-ramdisk.service - DietPi-RAMdisk
   Loaded: loaded (/etc/systemd/system/dietpi-ramdisk.service; enabled; vendor preset: enabled)
   Active: active (exited) since Sat 2018-06-02 01:34:31 UTC; 13h ago
  Process: 5894 ExecStart=/bin/bash -c /boot/dietpi/dietpi-ramdisk 0 &>> /var/tmp/dietpi/logs/dietpi-ramdisk.log (code=exited, status=0/SUCCESS)
  Process: 5891 ExecStartPre=/bin/mkdir -p /var/tmp/dietpi/logs (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/dietpi-ramdisk.service

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

zzlinwq, 2018-06-02:
@MichaIng 
root@amlogic:~# systemctl status dietpi-ramdisk
● dietpi-ramdisk.service - DietPi-RAMdisk
   Loaded: loaded (/etc/systemd/system/dietpi-ramdisk.service; enabled; vendor preset: enabled)
   Active: active (exited) since Sat 2018-06-02 01:34:31 UTC; 13h ago
  Process: 5894 ExecStart=/bin/bash -c /boot/dietpi/dietpi-ramdisk 0 &>> /var/tmp/dietpi/logs/dietpi-ramdisk.log (code=exited, status=0/SUCCESS)
  Process: 5891 ExecStartPre=/bin/mkdir -p /var/tmp/dietpi/logs (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/dietpi-ramdisk.service

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

Fourdee, 2018-06-02:
@zzlinwq 

Thanks

Try:
```
rm /etc/systemd/system/dietpi-ramdisk.service
systemctl daemon-reload
rm -R /DietPi
```

Then re-run PREP commands (make sure to copy and paste this entire code in SSH window).
```
export GIT_BRANCH='master'

wget https://raw.githubusercontent.com/Fourdee/DietPi/$GIT_BRANCH/PREP_SYSTEM_FOR_DIETPI.sh -O PREP_SYSTEM_FOR_DIETPI.sh
chmod +x PREP_SYSTEM_FOR_DIETPI.sh
sudo GIT_BRANCH=$GIT_BRANCH ./PREP_SYSTEM_FOR_DIETPI.sh
```
If error occurs, please try to paste the entire contents of the SSH window. We can use to debug.

zzlinwq, 2018-06-03:
[image]

zzlinwq, 2018-06-03:
[image]

zzlinwq, 2018-06-03:
[image]

zzlinwq, 2018-06-03:
[image]
[image]
[image]

zzlinwq, 2018-06-03:
[FAILED] DietPi-PREP | Unable to continue, DietPi-PREP will now terminate.
@Fourdee @MichaIng

Fourdee, 2018-06-03:
@zzlinwq 

Many thanks 👍 

Ok, apologies, I forgot to tell you to stop the service, before removal :)
- Reboot the system
- Rerun PREP again

zzlinwq, 2018-06-03:
[image]
[image]

zzlinwq, 2018-06-03:
Now I reboot   the system and Rerun PREP again
@Fourdee

zzlinwq, 2018-06-03:
[FAILED] DietPi-PREP | Unable to continue, DietPi-PREP will now terminate.

MichaIng, 2018-06-04:
@zzlinwq 
Thanks for try and report.

Could you try to install the related APT package manually and report back the terminal output: `apt install firmware-realtek`
I see your device might not need this firmware, but for compatibility reasons we install it on devices, as it enables support for several ethernet, wifi and bluetooth dongles.

If it remains really impossible to install it, you could remove this installation from DietPi preparation script: `sed -i '/G_AGI firmware-realtek/d' PREP_SYSTEM_FOR_DIETPI.sh`
But it would be good to investigate WHY this fails, as this should not happen, regardless of if firmware will be used or not.

Fourdee, 2018-06-04:
@MichaIng 

> But it would be good to investigate WHY this fails, as this should not happen, no matter if firmware will be used or not.

If I remember correctly, this occurs as it overwrites one of armbian's firmware .deb packages and causes an error (incompatible).
So either the armbian firmware needs to be removed, or, disable install of debian firmware.

But what confuses me is why ```dietpi-ramdisk``` exists, found the image http://soft.junlin.li/Armbian_5.44_S9xxx_Debian_stretch_3.14.29_icewm_20180515.img%20%281%29_20180531224519.xz. Unable to find it in ```/lib/systemd``` or ```/etc/systemd```

zzlinwq, 2018-06-05:
[image]
[image]

zzlinwq, 2018-06-05:
[image]

zzlinwq, 2018-06-05:
[image]
@ @MichaIng @Fourdee

zzlinwq, 2018-06-05:
Still not successful

zzlinwq, 2018-06-05:
[image]
[image]
成功了吗？ @Fourdee @MichaIng

MichaIng, 2018-06-05:
@zzlinwq 
Ah I was afraid first as you removed system core packages first, but seems the installer went through finally?

Strange, you are logged in as root, but script asks for root permissions? Could you run and paste the following?
`echo $UID`

@Fourdee
Puhh shall we try to resolve this issue for armbian images? Check for existing firmware and apt-get install firmware error outside of G_AGI handling and in case skip installation and set package on hold and as manual installed? Crazy hackaround, I wonder why armbian does not set it on hold, as every apt upgrade will fail....

zzlinwq, 2018-06-05:
I logged in as dietpi
[image]
@MichaIng

zzlinwq, 2018-06-05:
[image]

zzlinwq, 2018-06-05:
I try  logged in as rootroot,but  always prompt password error. What is password?

zzlinwq, 2018-06-05:
Why doesn't WIFI work?

Fourdee, 2018-06-05:
@MichaIng 

> Puhh shall we try to resolve this issue for armbian images? 

I'am tempted to do the following:
```
if [[ -f '/boot/armbianEnv.txt' ]]; then

   exit

fi
```

Fourdee, 2018-06-05:
> I try logged in as root, but always prompt password error. What is password?

dietpi

> Why doesn't WIFI work?

Enabled in ```dietpi-config```?

Whats results of:
```
lsmod
```

zzlinwq, 2018-06-10:
Why doesn't WIFI work?

Enabled in dietpi-config? It is enabled.

Whats results of:

lsmod
root@DietPi6689:~# lsmod
Module                  Size  Used by
brcmfmac              171742  0
brcmutil                5091  1 brcmfmac
cfg80211              415017  1 brcmfmac
aml_nftl_dev           95108  0
root@DietPi6689:~# 
@Fourdee

zzlinwq, 2018-06-10:
[image]
[image]

zzlinwq, 2018-06-10:
[image]

Fourdee, 2018-06-13:
@zzlinwq 

Apologies for the delay.

In DietPi WiFi relies on functional working drivers on the pre-image, and, knowing the exact WiFi hardware chipset used on the device.

Check dmesg for any errors relating to WiFi:
```
dmesg | grep brcm
dmesg | grep wlan
```

zzlinwq, 2018-06-13:
[图片]

zzlinwq, 2018-06-13:
roo t@DietPi6689:~# dmesg | grep brcm
roo t@DietPi6689:~# dmesg |  grep wlan
2.240105] Wifi: bcmdhd_ init_ wlan_ mem: bcmdhd_ init_ wlan_ mem: WIFI ME
I Allocated
@Fourdee

Fourdee, 2018-07-09:
@zzlinwq 

Apologies for the delay.

Unfortunately, we can only support WiFi devices on the boards we provide images for.

closed 2018-07-09 by Fourdee