MichaIng/DietPi #1816: General | Remove config.txt on non RPi systems

MichaIng, 2018-06-03:
This also needs adjustment of some scripts that expect/scrape this file (indeed unnecessarily): https://github.com/Fourdee/DietPi/blob/testing/PREP_SYSTEM_FOR_DIETPI.sh#L522

MichaIng, 2018-06-20:
Finds within code:
- 🈯️ **/DietPi/dietpi/boot**: Only touched within `if (( $G_HW_MODEL < 10 )); then`
- 🈯️ **/DietPi/dietpi/dietpi-config**: Only touched within `if (( $G_HW_MODEL < 10 )); then` or Menus that are just available for RPi.
- 🈺 **/DietPi/dietpi/func/dietpi-set_hardware**: Non-ramdisk `config.txt` declaration should be placed within RPi model check: https://github.com/Fourdee/DietPi/blob/testing/dietpi/func/dietpi-set_hardware#L56-L62

MichaIng, 2018-06-22:
PR up for testing: https://github.com/Fourdee/DietPi/pull/1863

Fourdee, 2018-06-23:
REF, pending testing: https://github.com/Fourdee/DietPi/pull/1863#issuecomment-399697869

Fourdee, 2018-06-30:
@MichaIng 

Great work on this 👍 No errors in my testing.

I believe we can mark as closed?

MichaIng, 2018-07-02:
@Fourdee 
Jep, although I am still thinking that reverting config.txt gpu_mem to gpu_mem_XXX should be done. It is also failsafe in case user did manual add one of those. But I sleep now, will do this tomorrow.

closed 2018-07-02 by MichaIng