MichaIng/DietPi #1818: DietPi-Config | Fan control

MichaIng, 2018-06-03:
At least for Odroid XU4/HC1/HC2 this is quite easy: https://dietpi.com/phpbb/viewtopic.php?f=9&t=3845&p=12839#p12839
For RPi there was some similarly easy way.

Would be nice to have it inside DietPi-Config where applicable/easy to accomplish.

enwi, 2018-06-25:
@MichaIng You do know that the HC1 and HC2 do not have any connections for a fan?

MichaIng, 2018-06-25:
@enwi
Good hint. Although the beard itself seems to have, according to forum user.

As HC1/2 is the same board, thus we cannot estimate any difference, AFAIK, we can still add the option. Of course it does not have any effect if no fan is attached.

And theoretically you could attach one with some case modification ;).

enwi, 2018-06-25:
@MichaIng Yeah that's what I did. I am using a USB fan, that I've attached to the metal extrusion. But to not confuse anybody there is no other way of adding a fan to HC1/2.

enwi, 2018-07-31:
@MichaIng 
On the HC1 and HC2 you can mount a fan to the metal extrusion. The XU4 either comes with a fan mounted or with a larger heatsink. All other Odroids except C0 have a heatsink where you can probably attach a fan afaik.

> @MichaIng You do know that the HC1 and HC2 do not have any connections for a fan?

Also  with that I meant that there is no way you can connect any cable to the board to control the fan since it only has a connection for a serial interface and a battery. See [this ](https://www.hardkernel.com/main/products/prdt_info.php?g_code=G150229074080&tab_idx=2)link.

MichaIng, 2018-07-31:
@enwi 
Many thanks for the infos.
Ah, so you do not control your fan in this case, leaving it at static (max) speed?

Just checked. Yeah, I thought XU4/HC1/HC2 are exactly the same, but the board layout/features are not.
XU4 comes with separate fan connector, HC1/2 do not. There is a similar connector, but that's the RTC backup battery 😉.
For other boards, there are USB fans offered. Needs check how/if they can be controlled.

For RPi there are several guides for using GPIO pins for fan control, but this needs a separate controller board then. We could add a default (software) interface and configuration for this to the script. But this needs someone who uses this setup and knows a bid how similar control (defined temperature limits, where the fan runs at defined speeds) can be achieved.

For general use, this turns out way more complicated then I though 🤣. So far this then just works for XU4 natively. At least I hope that other boards with this fan connector work similar or identical. Especially for native PCs this could be also interesting.
However, is still fun to write this, using our most current coding knowledge and methods to create menus, error handling etc.

enwi, 2018-08-01:
@MichaIng 
That's correct, the fan has two speed settings that can be selected with a switch and it is running at the low setting, which is plenty to cool the HC1.

Fourdee, 2018-08-06:
Testing:
- Native PC / Intel
```
[  OK  ] DietPi-Set_CPU_fan | Root access verified.
/DietPi/dietpi/func/dietpi-set_cpu_fan: line 47: /sys/devices/platform/pwm-fan/hwmon/hwmon0/automatic: No such file or directory
/sys/devices/platform/pwm-fan/hwmon/hwmon0/automatic: 
/sys/devices/virtual/thermal/thermal_zone0/trip_point_0_temp: 100000
/DietPi/dietpi/func/dietpi-set_cpu_fan: line 49: /sys/devices/platform/pwm-fan/hwmon/hwmon0/fan_speed: No such file or directory
/sys/devices/platform/pwm-fan/hwmon/hwmon0/fan_speed: 
/DietPi/dietpi/func/dietpi-set_cpu_fan: line 50: /sys/devices/platform/pwm-fan/hwmon/hwmon0/pwm1: No such file or directory
/sys/devices/platform/pwm-fan/hwmon/hwmon0/pwm1: 
[FAILED] DietPi-Set_CPU_fan | CPU fan control is not available on your device. Aborting...
```

MichaIng, 2018-08-06:
@Fourdee 
Oh my god, every manufacturer has it's own fan interface it seems. Yours for IBM/Lenovo/ThinkPad I guess... 🤔 

What is inside there?
```
cat /sys/devices/platform/thinkpad_hwmon/fan1_input
cat /sys/module/thinkpad_acpi/parameters/fan_control
```

> level:		auto

I guess auto means temp controlled, but can we adjust this, somehow attach temp to fan speeds?
_________________
Okay was reading the link a bid. There are several interfaces on how to control the fan on IBM/Lenovo/ThinkPad. What is most similar to XU4 is the following:
<details>
<summary>Sysfs</summary>

```
Sysfs notes:

The sysfs interface follows the hwmon subsystem guidelines for the most
part, and the exception is the fan safety watchdog.

Writes to any of the sysfs attributes may return the EINVAL error if
that operation is not supported in a given ThinkPad or if the parameter
is out-of-bounds, and EPERM if it is forbidden.  They may also return
EINTR (interrupted system call), and EIO (I/O error while trying to talk
to the firmware).

Features not yet implemented by the driver return ENOSYS.

hwmon device attribute pwm1_enable:
	0: PWM offline (fan is set to full-speed mode)
	1: Manual PWM control (use pwm1 to set fan level)
	2: Hardware PWM control (EC "auto" mode)
	3: reserved (Software PWM control, not implemented yet)

	Modes 0 and 2 are not supported by all ThinkPads, and the
	driver is not always able to detect this.  If it does know a
	mode is unsupported, it will return -EINVAL.

hwmon device attribute pwm1:
	Fan level, scaled from the firmware values of 0-7 to the hwmon
	scale of 0-255.  0 means fan stopped, 255 means highest normal
	speed (level 7).

	This attribute only commands the fan if pmw1_enable is set to 1
	(manual PWM control).

hwmon device attribute fan1_input:
	Fan tachometer reading, in RPM.  May go stale on certain
	ThinkPads while the EC transitions the PWM to offline mode,
	which can take up to two minutes.  May return rubbish on older
	ThinkPads.

hwmon device attribute fan2_input:
	Fan tachometer reading, in RPM, for the secondary fan.
	Available only on some ThinkPads.  If the secondary fan is
	not installed, will always read 0.

hwmon driver attribute fan_watchdog:
	Fan safety watchdog timer interval, in seconds.  Minimum is
	1 second, maximum is 120 seconds.  0 disables the watchdog.

To stop the fan: set pwm1 to zero, and pwm1_enable to 1.

To start the fan in a safe mode: set pwm1_enable to 2.  If that fails
with EINVAL, try to set pwm1_enable to 1 and pwm1 to at least 128 (255
would be the safest choice, though).
```
</details>
Most importantly:

```
NOTE NOTE NOTE: fan control operations are disabled by default for
safety reasons.  To enable them, the module parameter "fan_control=1"
must be given to thinkpad-acpi.
```
- `/sys/bus/platform/drivers/thinkpad_hwmon/pwm1_enable` set `0` for **static full speed**, `1` for **manual PWM control**, `2` for **auto (thermal based) mode**, which cannot be further controlled, at least according to the above info.
- `/sys/bus/platform/drivers/thinkpad_hwmon/pwm1` to set fan speed via values `[0-255]` (same as on XU4) for **manual PWM control** mode.
- `/sys/devices/platform/thinkpad_hwmon/fan1_input` read current speed in RPM, could be interesting for some kind of live monitor, although I am not sure how to add this to a whiptail menu: background process that rewrites a certain part of the screen, but how to know this part, breaks interactive input etc?
- `/sys/bus/platform/drivers/thinkpad_hwmon/fan_watchdog`, by default, after changing the above controls, will check after 120 seconds if the fan speed is safe and in case enables fan or rises fan speed, if I got it right? Thus if a static fan speed wants to be assured, a value of `0` needs to be given to disable the watchdog, other values define the amount of seconds, after which the watchdog checks fan speed against temperature again.
- NB: The fan speed will slowly ramp to the desired value, up to 120s, if I got it right.
- NB:
```
The sysfs interface tries to blend in the generic Linux sysfs subsystems
and classes as much as possible.  Since some of these subsystems are not
yet ready or stabilized, it is expected that this interface will change,
and any and all userspace programs must deal with it.
```

So we can hope, that for most x86_64 systems, the above interface is at least similar. I am thinking about installing DietPi on a separate partition/drive of my laptop 🤔. HP Probook would be it.

Fourdee, 2018-08-06:
@MichaIng 

```
root@DietPi:~# cat /sys/devices/platform/thinkpad_hwmon/fan1_input
0
root@DietPi:~# cat /sys/module/thinkpad_acpi/parameters/fan_control
N

root@DietPi:~# echo 1 > /sys/module/thinkpad_acpi/parameters/fan_control
bash: /sys/module/thinkpad_acpi/parameters/fan_control: Permission denied

```

MichaIng, 2018-08-06:
@Fourdee 
Does this work?
`echo 0 > /sys/bus/platform/drivers/thinkpad_hwmon/pwm1_enable`
Fan slowly speeds up until full speed...
```
echo 1 > /sys/bus/platform/drivers/thinkpad_hwmon/pwm1_enable
echo 0 > /sys/bus/platform/drivers/thinkpad_hwmon/pwm1
```
Fan slowly speeds down until stop...
...possibly after some time, it speeds up again due to watchdog. In this case:
```
echo 0 > /sys/bus/platform/drivers/thinkpad_hwmon/fan_watchdog
echo 1 > /sys/bus/platform/drivers/thinkpad_hwmon/pwm1_enable
echo 0 > /sys/bus/platform/drivers/thinkpad_hwmon/pwm1
```
...should lead to fan stays off.
`echo 2 > /sys/bus/platform/drivers/thinkpad_hwmon/pwm1_enable`
Fan, depending on temperature, turns on and auto adjusts speed to temp/usage.
`echo 1 > /sys/bus/platform/drivers/thinkpad_hwmon/fan_watchdog`
Just to be failsafe, when enabling auto mode anyway.

MichaIng, 2018-08-13:
@Fourdee 
Let's move to v6.14.
This is no urgent thing and if release, it should be more than XU4-only 😉. Initial branch is still in place, so we can re-add it at any time.

MichaIng, 2018-11-20:
Moved from own branch to main branch line `.meta/` dir: https://github.com/Fourdee/DietPi/commit/2d9e9e3bd40daa15a10075490baedd4386bf7242

MichaIng, 2020-08-29:
ROCKPro64 uses `/sys/class/hwmon/hwmon3/`: https://dietpi.com/phpbb/viewtopic.php?p=26996#p26996
Not sure if there is a generic way to detect the correct tuneables, probably we need to implement it per-device 😞.

cocoflan, 2021-04-12:
Some days ago i tested an I2C cooling had, works perfectly on Dietpi, just needed some finetuning to get it work.

Python or C based control settings.

https://github.com/YahboomTechnology/Raspberry-Pi-RGB-Cooling-HAT

MichaIng, 2021-06-02:
Pinging @Rick847 here and #4454, with the idea to make this more generic, toggling a GPIO-attached fan based on CPU temperature, where no native PWM fan socket + kernel interface is available.

Joulinar, 2021-06-03:
this one we have as well https://github.com/MichaIng/DietPi/issues/4003

MichaIng, 2021-06-03:
True. Again very different (much simpler) to configure, but when we write such a script, why not adding every case of fan controls into it. But I think I'll add the RPi case fan to dietpi-config > Advanced Options earlier, as it is all together just a few lines of code, not comparable to the other solutions. Also while the others work OOTB, so can be live-controlled, for RPi fan changes via dtoverlay, a reboot is always required.

MichaIng, 2026-11-14:
Partly done in #8269, for RPi only so far. Adding support for generic sysfs PWM control will eventually follow.