MichaIng/DietPi #1821: Request manual resize

symbios24, 2018-06-04:
Hello, is it possible to include with DietPi pre-installed the raspi-config tool?

Thanks

MichaIng, 2018-06-04:
@symbios24 
Thanks for your request.

`dietpi-config` should include all features of `raspi-config` and more. For this reason we do not include the latter to reduce redundancy and possible incompatibility of both tools adjusting the same files etc.

If you miss any feature or something is done possibly better in `raspi-config`, please let us know and we will update the behaviour of `dietpi-config` accordingly.

Within issue title you name the manual partition resize I guess? This should be done automatically on 1st boot of DietPi, so nothing to do manually any more.  Did this not work somehow? In this case please post output of `fdisk -l /dev/mmcblk0` and content of `/var/tmp/dietpi/logs/fs_partition_resize.log`.

467815891a, 2018-06-06:
However，I want dietpi-config add "extend rootfs" options like raspi-config.
I use `dd` command to backup the whole sdcard and restore to a new sdcard.
Now I am having trouble in extend rootfs to fullfill new sdcard.

MichaIng, 2018-06-06:
@467815891a 
Ah yeah when switching disk this makes totally sense.
You can currently run the service manually, but be sure the backup on 2nd SD is up-to-date:
```
/DietPi/dietpi/dietpi-services stop
chmod +x /var/lib/dietpi/fs_partition_resize.sh
/var/lib/dietpi/fs_partition_resize.sh
/DietPi/dietpi/dietpi-services start
```

I added this to v6.10 milestone. Will add a button to dietpi-drive_manager for file system expansion.

467815891a, 2018-06-07:
@MichaIng Thanks so much😁

DaCodaHUB, 2018-06-26:
I tried the commands but I got the error "Re-reading the partition table failed.: Device or resource busy". Can anyone help?

Fourdee, 2018-06-26:
@DaCodaHUB 

Hi,
Which disk are you attempting to resize? RootFS? 
```
df -h
```

The error is due to the partition table resize, which will not work on a live system with RootFS mounted, although, the filesystem resize should work.

Either way, the resize should not fail on first run. The only factor that might cause a failed resize on RootFS is a hardware issue.

Fourdee, 2018-06-26:
@DaCodaHUB 

Use the following to resize RootFS during boot:
```
systemctl unmask dietpi-fs_partition_resize.service
systemctl enable dietpi-fs_partition_resize.service
reboot
```

Fourdee, 2018-06-26:
@MichaIng 

I'll pick this one up, as still working on https://github.com/Fourdee/DietPi/issues/1858

MichaIng, 2018-06-28:
I will close this issue in favour of the one linked by Fourdee and as he already added the feature 😃.

closed 2018-06-28 by MichaIng