MichaIng/DietPi #1823: annoying DPMS cannnot turn off in Chromium autostart mode

467815891a, 2018-06-05:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version =v6.8
- SBC Device =RPi 2 Model B
- Power supply used (5V2A)
- SD card used (Sandisk ultra)
- Distro (Sretch )

#### Additional Information (if applicable):
I install Chromium whithout any Desktop Evironment.
Autostart run script for Kiosk mode，but DPMS still working.
I cannot disable DPMS.
I have added 
`xset s off
xset -dpms
xset s noblank` 
to /etc/X11/xinit/xinitrc
But it turns out not working...
Without LightDM,how can I disable dpms?

Fourdee, 2018-06-05:
@467815891a 

Thanks for the report 👍  we'll open for investigations.

Fourdee, 2018-06-05:
Notes:
Check DPMS config:
```
xset q
```

Fourdee, 2018-06-05:
@467815891a 

Solution, copy and paste all
```
mkdir -p /etc/X11/xorg.conf.d
cat << _EOF_ > /etc/X11/xorg.conf.d/99-dietpi-dpms_off.conf
Section "Monitor"
    Identifier "TV"
    Option "DPMS" "false"
EndSection

Section "ServerLayout"
    Identifier "ServerLayout0"
    Option "StandbyTime" "0"
    Option "SuspendTime" "0"
    Option "OffTime"     "0"
    Option "BlankTime"   "0"
EndSection
_EOF_
```

I'll apply this as default in v6.9 when xserver is installed.

Fourdee, 2018-06-05:
Resolved for v6.9

closed 2018-06-05 by Fourdee