MichaIng/DietPi #1824: DietPi and Squeezelite -- No Sound

Timo2708, 2018-06-05:
I have been hugely enjoying the Dietpi on my new Allo DigiOne — at first as Roon Bridge, and then playing around with GMRender and BubbleUpnP. All fine!

Next I wanted to experiment with Squeezelite. I installed LMS on my Qnap, and iPeng on the iPad. The LMS on the QNAP appears in iPeng, and also the DietPi-Squeezelite as player. But I don’t get any sound — my streamer/DAC (Naim ND5XS) doesn’t seem to receive any signal from Squeezelite/Allo DigiOne.

Any idea what I might be doing wrong? 

Many thanks,
Timo

Fourdee, 2018-06-05:
@Timo2708 

Hi,

Thanks for the report 👍 

Are you using the Allo GUI, or stock DietPi install?

When streaming to squeezelite:
- If using the Allo GUI
-- Try changing the output bitdepth to 32bit on squeezelite
- If using stock DietPi + Manual install
-- paste the results of the following command
```
systemctl status squeezelite -l
```

Fourdee, 2018-07-09:
No response and unable to replicate, marking as closed.

Assuming incorrect output bit-depth is set.

closed 2018-07-09 by Fourdee