MichaIng/DietPi #1827: META | Public DietPi-Survey result web page

MichaIng, 2018-06-07:
As we restart the new updated safer and more transparent DietPi-Survey (https://github.com/Fourdee/DietPi/pull/1822), I though it would be nice to have a public result page to review current statistics. So users know exactly what we do with the data and we can also easily access them without the need to do manual extraction, sorting and evaluation.

I though about some cron job, that extracts the data, sorts them and writes them back to some simple HTML file that can be accessed publicly.

Extremely nice would be of course a ready to use PHP table script with sorting, filtering and search function, but I couldn't find one through quick www research. The latter one would e.g. need the data to be regularly read and saved into a MariaDB table from where the PHP script can read. If someone knows how to achieve this or has fun to write such script, would be welcome.

The installed software part could be a bid tricky, as the software IDs change by times. I thought about another (array) file for each DietPi version that connects software IDs with software title. 

This is how the survey files currently look like:
```
-------------------------
DietPi-Survey v$SURVEY_VERSION
-------------------------
Upload Count   : $SURVEY_SENTCOUNT
DietPi Version : $DIETPI_VERSION
Hardware Index : $G_HW_MODEL
Hardware Name  : $G_HW_MODEL_DESCRIPTION
Distro Index   : $G_DISTRO
Autoboot Index : $(</DietPi/dietpi/.dietpi-autostart_index)
Hostname       : $(</etc/hostname)
-------------------------
DietPi-Software Installed
-------------------------
$(grep '=2' /DietPi/dietpi/.installed)
```
Ah @Fourdee we should remove hostname for anonymity? Can't think of value we get through this.

Fourdee, 2018-06-07:
@MichaIng 

> Ah @Fourdee we should remove hostname for anonymity? Can't think of value we get through this.

Yep, we can remove it, no benefit to us knowing that information.

MichaIng, 2018-06-08:
I played around a bid and created a simple result page: https://dietpi.com/survey/

Ugly I know, and there is still a glitch with some software titles (bottom) being somehow broken over two lines. Ah now that I write it, I think I know what's the issue about it: `sed -i 's/ /_/`**g**`' "$file`
`€: solved`

NB:
- The DietPi-Survey data was reset due to security concerns with the previous solution. We made this now bullet prove and want to handle it as transparent as possible. Don't mind to give feedback, so we handle your data as secure as possible!
- This means the data shows v6.9 DietPi only. On update, every user is ask to join or opt-out. It is even possible, if opted in once, to clear the uploaded data file again via `dietpi-survey` menu. As the public upload user has no read nor file list permissions, it is not possible to remove the file entirely. However it is interesting as well to know how much users requested to clear their data to check acceptance (see result page). But again, the files itself are empty (0 bytes) then.

I will go and upload the result page creation script to a separate GitHub branch, in case anybody wan't to make it more beautiful 🙂.

Fourdee, 2018-06-08:
@MichaIng 

Love this, great work 👍 

Was having a play with 'Opted out':
```
        if [[ ! -s $file ]]; then

                ((SURVEY_COUNT_EMPTY++))
        device_name='Unknown_(Opted_Out)'
        ((aDEVICE_NAME[$device_name]++))
                continue

        fi
```

Do you think we need it?

[image]

MichaIng, 2018-06-09:
@Fourdee 
Not sure, we would need to add dummy names for all infos then, for consistency?

Currently, summarizing the counts of each info type matches `Available unique survey files`.
`Purged unique survey files` gives the info about how much users actively opted out and wanted to have their data purged in case. Not sure if it makes sense to add this number to each category again as unknown 😉. Maybe a better naming could make things clearer again.

But still we do not have the total users count, as if users just opt out without purge request, we have no file at all.

Hmm actually would be nice to have the overall count as well, maybe opt out users just by default always send empty files? Otherwise if I don't want to send data, I also would not want to have empty files sent, I would want to have the whole process skipped.

Fourdee, 2018-06-09:
@MichaIng 

> Hmm actually would be nice to have the overall count as well, maybe opt out users just by default always send empty files? Otherwise if I don't want to send data, I also would not want to have empty files sent, I would want to have the whole process skipped.

I'd like to keep the initial send, even if the user opts out and its cleared. Allows us to at least have a total count of unique installs from no data.

> But still we do not have the total users count, as if users just opt out without purge request, we have no file at all.

Good point, maybe would could simply remove ```'1' 'No, I want to opt out DietPi-Survey.'```?

Fourdee, 2018-06-09:
@MichaIng 

> Not sure if it makes sense to add this number to each category again as unknown 😉. Maybe a better naming could make things clearer again.

Yeah, just had the idea of making each table add up to the total. May even make users change their mind to OPT IN lol (we can but hope).

Fourdee, 2018-06-09:
@MichaIng 

https://twitter.com/DietPi_/status/1005586292950884352

Fourdee, 2018-06-09:
@MichaIng 

Ok, the way I want this is:
- If no save file exists, by default set to opted in and send. This ensures automated installations are opted in by default
- If interactive installation, user will then be able to change to mode 2 or 0.

MichaIng, 2018-06-10:
@Fourdee 
- By default opt out I was also thinking about the GDPR.
- I am not sure about if/how this applies for FOSS software and anonymous data, but the aim, that users always first need to actively accept, before any data is send, is I want to have it for any (commercial) software and services I use, thus default opt out. But that's my opinion 😉.
- The current survey forces interactive menu on every first run to create settings file then. Thus even if on automated first run and update no interactive menu is possible (G_USER_INPUTS=0), then the next dietpi-software or dietpi-update will prompt the window, so it's hard to completely avoid the decision 😉.
- Yeah just allowing opt in and purge would at least allow us to have an overall count, I guess that's okay 👍.

Fourdee, 2018-06-10:
@MichaIng 

Yep, see what you mean, getting a little lengthy to process:
```
root@DietPi:~# time /etc/cron.minutely/survey-report

real    0m48.765s
user    0m2.152s
sys     0m3.872s
```

RAM should be fine.

Thinking we need to reduce the pipe's, awk and sed's, might speed it up?

Another option is we could change the survey formatting, make them all variables and simply load the file?
eg:

```
#-------------------------
dietpi_survey_version[$uuid]=5
#-------------------------
upload_count[$uuid]=1
dietpi_version_core[$uuid]=6
#etc
```

We should aim to do the minimal processing client side, take load off the server.

MichaIng, 2018-06-13:
> 1000 unique installations on v6.9+ 👍

Fourdee, 2018-06-14:
@MichaIng 

I was thinking something along the lines of:
```
#!/bin/bash
index_input=$1
dietpi_survey_version[$index_input]=5
upload_count[$index_input]=$uploadcount
g_hw_model[$index_input]=2
```

Then we could simply do the following in the server script:
```
local index=0
for file in /home/dietpi-survey/survey/*.txt
do

. $file $index
((index++))

done
```

Then all the arrays are loaded. The issue is with installed software, might need to covert it to a 2d array on the client before sending, eg:
```
root@DietPi:~# grep '=2' /DietPi/dietpi/.installed | sed 's/E\[/E\[$index_input\]\[/'
aSOFTWARE_INSTALL_STATE[$index_input][0]=2
aSOFTWARE_INSTALL_STATE[$index_input][103]=2
aSOFTWARE_INSTALL_STATE[$index_input][105]=2
```

Or we could create a new input for ```dietpi-software list_survey``` and print only the following:
```
openssh client
dietpi-ramlog
```

or, the following then convert as needed during sourcing on server
```
root@DietPi:~# dietpi-software list | grep ' =2 ' | awk '{print $2}' | sed 's/^/software_index_installed=/'                                                  software_index_installed=0
software_index_installed=103
software_index_installed=105
```

MichaIng, 2018-06-14:
@Fourdee
To have the files source-able should already speed up creation indeed. But as said with:
`g_hw_model[\$index_input]=2`
if we want to have as now a count of how many users user model 2, we would still need an array conversion like
`for i in ${g_hw_model[@]}; do ((aHW_MODEL[$i]++))`
To have `((aHW_MODEL[$G_HW_MODEL_NAME]++))` directly within the survey files, would allow us to skip this conversion as well and does not have any effect on client side creation speed.

Btw the index does not need to exist to allow ++. If it doesn't exist, it will be created with value 1. See current survey_report script, where I removed the initial index creation.

MichaIng, 2018-06-16:
@Fourdee 
Replaced `sed` and `awk` by bash internal variable manipulation ${VAR##* : } etc: http://wiki.bash-hackers.org/syntax/pe
```
root@DietPi:~# time /etc/cron.minutely/survey-report

real    0m37.277s
user    0m3.428s
sys     0m4.516s
```
😃

Fourdee, 2018-06-16:
@MichaIng 

RAM test, I believe its hitting swapfile.
```
root@DietPi:~# time /etc/cron.minutely/survey-report_4D_testing

real    0m35.267s
user    0m3.196s
sys     0m4.244s
```

MichaIng, 2018-06-16:
@Fourdee 
I watched last cron job execution, RAM usage is quite low, luckily 😅. Also CPUs are not at 100%. Nice 19 is also good there to in case prioritize web access.
I think it would be worse, if we'd really source every file into it's own array first, instead of add info to counting array and reset the other variables again.
On the other hand...
```
root@DietPi:~# G_TREESIZE /home/dietpi-survey/survey/
[  OK  ] Root access verified.
5.5 MB /home/dietpi-survey/survey/
```
Can the RAM usage be higher than what all the files actually contain (including comments)?

MichaIng, 2018-06-19:
@Fourdee 
Tested your 4D version:
```
root@DietPi:~# time /etc/cron.minutely/survey-report_4D_testing

real    0m38.168s
user    0m3.492s
sys     0m4.092s
root@DietPi:~# time /etc/cron.minutely/survey-report

real    0m44.668s
user    0m3.908s
sys     0m5.372s
root@DietPi:~# nano /etc/cron.minutely/survey-report
root@DietPi:~# time /etc/cron.minutely/survey-report
rm: cannot remove '/tmp/dietpi-survey': No such file or directory

real    0m50.749s
user    0m4.560s
sys     0m6.132s
```
- The last test was with my first RAM approach: Copying every file separately into RAM and removing/replacing within loop. But the many single copy calls seem to take more time here, than grep from outside of RAM. Copying everything at once on the other hand indeed speeds up the process a bid 👍.
- Hmm made further tests, where the difference it not too large any more, but overall significantly.
______________________
Associative arrays (`declare -A aTEST`, supports string indexes):
- First I thought that they don't support spaces within index, but they do.
- You only need to take care when looping though the indexes to set the array into double quotes. Otherwise, and I would name this a bug, the loop index will go through each space separated string within the indexes as well:
```
root@DietPi:~# declare -A aTEST
root@DietPi:~# aTEST[test ing ahhh]=test1
root@DietPi:~# aTEST[hmm mal sehn]=test2
root@DietPi:~# test='jo schaun wir mal'
root@DietPi:~# aTEST[$test]=test3
root@DietPi:~# echo ${!aTEST[@]}
test ing ahhh hmm mal sehn jo schaun wir mal
root@DietPi:~# echo "${!aTEST[@]}"
test ing ahhh hmm mal sehn jo schaun wir mal
root@DietPi:~# echo "${aTEST[@]}"
test1 test2 test3
root@DietPi:~# for i in ${!aTEST[@]}; do echo $i; done
test
ing
ahhh
hmm
mal
sehn
jo
schaun
wir
mal
root@DietPi:~# for i in ${!aTEST[*]}; do echo $i; done
test
ing
ahhh
hmm
mal
sehn
jo
schaun
wir
mal
root@DietPi:~# for i in ${!aTEST[*]}; do echo "$i"; done
test
ing
ahhh
hmm
mal
sehn
jo
schaun
wir
mal
root@DietPi:~# for i in "${!aTEST[*]}"; do echo "$i"; done
test ing ahhh hmm mal sehn jo schaun wir mal
root@DietPi:~# for i in "${!aTEST[@]}"; do echo "$i"; done
test ing ahhh
hmm mal sehn
jo schaun wir mal
root@DietPi:~# for i in "${!aTEST[@]}"; do echo $i; done
test ing ahhh
hmm mal sehn
jo schaun wir mal
```
- I changed to more beautiful fonts as well.

Fourdee, 2018-06-20:
@MichaIng 

Love the dark theme and new fonts, nice 👍

MichaIng, 2018-06-23:
@Fourdee 
As we anyway change the survey, we can think of further or not needed entries:
- Upload count: Don't see too much interest in this. The only thing is that we can derive a bid the activity of users, how often new software is installed and/or played around. If we want to keep it, we need to increase the number +1 within survey file creation. I see now why you in between changed initial value to 1 instead of 0 😉. However since all sent files now have the value 1 too low, I think it is easier to just add +1 on file creation. Also the number within config file then stays correct. If we keep it, as well we would need to change the way it is presented at survey report, otherwise the table will become way too large.
- On top of DietPi version we could add the used branch. Usually one could derive it from version number, but also it plays a role if/when the update is done. To keep users privacy a bid, although this should be very rare cases, when using an own branch (owner != Fourdee), we could just use `custom/own/unofficial branch/fork`. I don't think it will be much, but could be still interesting, if existent.
- Survey version: Can be derived as well directly from DietPi version. Interesting just for internal use to in case to correct info scraping/file handling. Although if we go with sourceable files, I would just check for `#!/bin/bash` and in other cases, as said we can also just handle based on DietPi version.
- Interesting could be as well the one or the other chosen setting/dietpi.txt entry: How much users use first run automation, use local or public time servers (after we implemented it) and such things.

Fourdee, 2018-06-23:
@MichaIng 

> Interesting could be as well the one or the other chosen setting/dietpi.txt entry: How much users use first run automation, use local or public time servers (after we implemented it) and such things.

This I like!

Automated install data is a must 👍

MichaIng, 2018-07-02:
@Fourdee 
Will do testing now, handling is already partly added to survey-report script. As you check the current survey script, the loops already contain those array number increases of type: `((aSURVEY_VERSION[6]++))`
When sourcing the file, this is not done automatically to the right array. Values look correct. Translation of arrays to html file does not change, as the arrays are the same that are already created for v6.9.

MichaIng, 2018-07-06:
@Fourdee 
So far it works quite well. Only thing is that the list is quite long already, much to scroll. Great would be to have separate pages for the different counts/tables, perhaps some overview page/frame with links.
The ongoing upload count table would be also not as problematic any more. Currently I would vote to show it somehow different instead of an own row for every single count.

Perfect would be to have it with similar style than https://dietpi.com/dietpi-software.html, integrated into dietpi.com main page.

userdeveloper98, 2018-07-06:
Hey guys, I have something for you !
Check it out: https://files.fm/u/skzgfequ 
Extract, run (`java -jar dietPiSurveyDemo.jar`), open http://localhost:8080/survey.xhtml. 
(You need java installed and port 8080 free)
Hope you will like it :+1:

Fourdee, 2018-07-07:
@userdeveloper98 

I tried it using remote connection, hangs on loading white page.

Works great on localhost, looks good 👍 :
[image]

How are you scraping the data? Directly from https://dietpi.com/dietpi-software.html?

userdeveloper98, 2018-07-07:
@Fourdee 

> How are you scraping the data?.

I used hardcoded values just for the sake of demo...
If you like it we can do a real one...

Fourdee, 2018-07-07:
@userdeveloper98 

> I used hardcoded values just for the sake of demo...
If you like it we can do a real one...

By all means, if this is something you want to do and works well, we can add it to the server.
Ideally, if you can scrape from https://dietpi.com/dietpi-software.html directly, it would be best, although, if you do go ahead, we could export the data values as a .txt download?

The only issue is I have 0 knowledge of Java, so all maintenance of the page would be down to you.

MichaIng, 2018-07-07:
@Fourdee
You mean scraping from dietpi.com/survey right? ;)

@userdeveloper98 
Looks really nice, great work. Remote access from VM host to VM guest worked all here.
Ideally of course would be it running on dietpi.com and directly reading data from survey upload directory to replace current static dietpi.com/survey

Only concerns: It runs as java, so the server needs to have java installed and run it. Took 500M RAM here on my VM. This is a quarter of whole dietpi.com RAM and perhaps too much, just for a meta page. Or is there a way to have it somehow run in a more lightweight state? Would be of course preferred to have it using the available apache webserver, but I guess it is not possible the way it is currently.

And jep, also no deeper java knowledge here as well, but we can help to provide example data,  in the way we have it on the server.

userdeveloper98, 2018-07-09:
@MichaIng  @Fourdee 

> Ideally, of course, would be it running on dietpi.com and directly reading data from survey upload directory to replace current static dietpi.com/survey

Agree, where is survey upload directory?

> Took 500M RAM here on my VM. This is a quarter of whole dietpi.com RAM and perhaps too much, just for a meta page. Or is there a way to have it somehow run in a more lightweight state?

It's a very dirty draft, it contains a lot of things that are not needed. 
Also, by default, it takes as much as it can for the cache, but it can be fine-tuned to specific requirements 
A more lightweight version will be built once we agree on all details and preferences 😉

> And jep, also no deeper java knowledge here as well, but we can help to provide example data, in the way we have it on the server.

> although, if you do go ahead, we could export the data values as a .txt  download?

It would be great!
It potentially can update itself very efficient. (ex: only when file changes were detected.)

I would prefer to parse a .txt file or .csv or even better a DB query.
Scraping HTML is very error prone and minor changes in HTML will require a lot of maintenance work.

------------------------------------
<sup>P.S (I didn't investigate a lot how the survey works now)
Just a thought.. what do you think if we will assign a unique id for each user to maintain survey reports accurate (avoid duplicates, track history, etc...). 
This should be something that will give confidence that it is really unique and there is nothing related to personal data.
A perfect candidate for this is the MAC address in my opinion. 
It is related to the hardware so it will not change after reinstalls or user configurations. 
And in addition, just to avoid leaking private data, we will hash MAC address with a strong algorithm.
In this way, we will have a unique, depersonalized id for each user that will not change after reinstall or configurations. 
</sup>

MichaIng, 2018-07-09:
@userdeveloper98 
> Agree, where is survey upload directory?

- `/home/dietpi-survey/survey/`
- Not sure what is possible with the embedded webserver. Ideally would be, the webserver being somehow uncoupled from report page creation, e.g. background process needs read access of course, but webserver user just having access to the produced database/webpage.

> something that will give confidence that it is really unique and there is nothing related to personal data

- DietPi-Survey and Bugreport use an unique ID in hardware ID style, check `/DietPi/dietpi/.hw_model` to get yours. If you check survey creation, this ID is used as upload file name and by this you can uniquely identify also the data within upload folder.
- It is anonymous, not even related to any real hardware ID (created randomly on DietPi firstrun). But as we are currently facing this security and privacy topic, maybe it is best to translate this ID internally to a simply index to show on the page, just that no one can see an ID that is present on your own DietPi 😉 .
- This ID is even better than MAC address, since MAC can change, when changing network device (USB ethernet for Pi Zero e.g.). Only negative is that changes on reinstalling DietPi on the same machine. There MAC (or better a definitely not changeable core hardware ID would be better) 🤔.

With v6.11 we optimized the survey file scheme for our current report creation, thus it is a bid cryptic, containing bash array creation/upcounting code.
When starting `dietpi-survey` your personal survey file will be shown. Most array indexes are in combination with array name self explained, others are embedded into another array that translates the index into string (autostart index, software index). These might be a bid tricky, as especially software indexes are added and removed on DietPi updates, also autostart indexes might change, however not very often. So you need a version specific translation, which is currently done by the inner array within the survey file.

I will upload the current survey creation file. @Fourdee Is there a way to upload (meta) files to github that must not be downloaded via git archive download or clone, perhaps all files within `.github` or are just all files with leading dot ignored?
Hmm, actually easy to test with dedicated branch 😄.

Fourdee, 2018-07-09:
@MichaIng 

>  @Fourdee Is there a way to upload (meta) files to github that must not be downloaded via git archive download or clone, perhaps all files within .github or are just all files with leading dot ignored?
Hmm, actually easy to test with dedicated branch 

No idea, buts its a good idea 👍

userdeveloper98, 2018-07-10:
@MichaIng 
> > Agree,  where is survey upload directory?

>  /home/dietpi-survey/survey/

I mean the folder where all concatenated survey reports are, from all users.
I saw a sample when I opted in, but I don't really know how all reports together will look like.
Is that somehow parsed/formatted matched to UniqID or just appended, etc... 

I need a sample of the file which application will need to read -> parse -> draw charts.
Ideally, it will be the latest version of the file with all users info.

> maybe it is best to translate this ID internally to a simply index to show on the page, just that no one can see an ID that is present on your own DietPi 

Will be done 👍 .

> better a definitely not changeable core hardware ID would be better

I see this as a better option because after re-install you don't really now, is this a new user or an upgraded one? Few re-installs and you will get unrealistic statistics.

MichaIng, 2018-07-10:
@userdeveloper98 
This gives you the file name of your upload, based on DietPi created hardware ID: `echo $(sed -n 5p /DietPi/dietpi/.hw_model).txt`

Those uploaded files are placed within the mentioned folder and not changed there. They are just parsed/sourced by the survey-report script to create the static html page.

I just searched a bid about retrieving a bullet prove unique hardware ID, and there simply doesn't seem to be a solution:
- MAC addresses can be changed and they might come from an (external) network card/dongle, which can be replaced.
- `dmidecode` seems to give hardware IDs as well, some they are sometimes by manufacturer just created of MAC address + leading or trailing zeros and/or finally by the scheme cannot be totally unique (duplications statistically). And `dmidecode` is not available/working on all systems as well.
- https://askubuntu.com/questions/54329/how-to-generate-unique-hardware-hash
- https://stackoverflow.com/questions/328936/getting-a-unique-id-from-a-unix-like-system

If we found a totally unique ID that is attached to the motherboard and/or CPU and cannot be manipulated software wise, we could hash this again and then use it for uploads. But as it doesn't seem so trivial, and as well would lead to another dietpi-survey reset (which we just had), I would for now stay with the generated ID. The data is/will be incorrect for completely fresh DietPi installs, but I think this is something we can live with. The for us important ratio between used devices and software titles will be precise enough the check whether development into the one or the other direction will be beneficial for more users.

My idea anyway is, do e.g. drop/handle differently files that were not updated since a certain time, 3 months e.g.. Most likely devices are inactive (or reflashed), if dietpi-update and dietpi-software was never called within 3 months. To use file ages even more precise, we could call dietpi-survey as well within e.g. monthly cron job and then use 1 month as a drop age.

Fourdee, 2018-07-10:
> Do you otherwise agree to just use .meta or any other folder in repo root to place files such as survey-report or https://github.com/Fourdee/DietPi/blob/dietpi-cloud-migration/dietpi-cloud-migration? A separate branch or even repo is totally overkill. But don't want to mess up the repo either.

Yep, by all means 👍 
As long as the folder is not installed on DietPi systems, or, can be removed during the ramdisk update.

MichaIng, 2018-07-11:
@userdeveloper98 
Here the current script we use: https://github.com/Fourdee/DietPi/blob/testing/.meta/survey_report

userdeveloper98, 2018-07-12:
@MichaIng 
Thanks a lot for the script!
Now it's crystal clear how everything works. 😃

MichaIng, 2018-07-17:
@Fourdee 
What do you think about weekly or monthly `dietpi-survey 1` executions to allow us removed obsolete uploads?

MichaIng, 2018-07-17:
@Fourdee 
Jep, also sending empty file if opt out. I don't see another way to safely identify obsolete uploads.

Fourdee, 2018-07-17:
@MichaIng 

> Jep, also sending empty file if opt out. I don't see another way to safely identify obsolete uploads.

Yep, same here. We lack any method of unlinking the device, when its no longer in use. The only thing I can think of is a simple "This system is no longer going to be used" option, which would remove the survey, however, relies on the user doing it, and, not abusing it.

MichaIng, 2018-07-17:
@Fourdee 
Nothing we can rely on.
I mean it's not a really bid issue, the ratios between used devices and software will be not affected too much, and we could simply assume a max time before marking the related upload as obsolete, e.g. 3 months (?), so old files will not grow unlimited. But for statistic and development reasons and the possibilities with @userdeveloper98 's solution, we should not remove, but move them into an "old" folder for now.
Still some time until survey v5 reaches 3 months anyway 😉.

userdeveloper98, 2018-07-26:
@MichaIng  @Fourdee 
I need your feedback on this: https://files.fm/u/wpvsdrc5 
It is another draft that comes with a set of survey samples.
You can repoint via properties file to the real folder and check if it is parsing properly the files.
**Parsing is generic**, you can adjust the template to parse **any text file**. 
So that changing in future file structure will not require code changes.
Let me know if you want specific charts/filters/searches etc..

Fourdee, 2018-07-26:
Notes:
```
Open: [http://localhost:8888/survey.xhtml]. And enjoy :)
```

@userdeveloper98 

Good progress 👍 Scraping seems to work well.

Not sure if you can use custom fonts with Java? But if you can, the font DietPi logo is based on is:
https://cooltext.com/Download-Font-Michroma

Might work well for the UI?

---

In regards to the "Devices and installed versions barchart". Personally, i'd simplify this and have only "Install count per device"?

In the survey database tab, some suggestions:
- Remove the unique HW_ID/Filename column, as we dont want users to be able to match the UUID to any data. Even though the UUID is random, there no benefit to having it displayed (IMO)
- The CPU Arch dropdown appears to show incorrect devices for the selected item.

userdeveloper98, 2018-07-28:
@Fourdee 
Thanks for the feedback.
New version released: [[link](https://files.fm/u/5hp326jq)]
- Added Michroma font for the title, for everything else, applied Roboto as on dietpi.com
- Removed UUID column 
- Fixed CPU Arch dropdown filter
- Added favicon for the page
- Added Dietpi logo on to of the page
- Changed slightly background color

Fourdee, 2018-07-28:
@userdeveloper98 

Nice progress 👍 

[image]

Some issues/feedback if needed:
- CPU arch selection has incorrect results for me
[image]
- On the main barchart for "devices and installed version". I strongly feel a simple "Device install count" (without versions) would make this cleaner for the end user to read (less busy)?
- Something failed when running the java program (could be due to running it on a Odroid C2 with JRE8?)
```
.SurveyLoaderImpl  : Mathing software records from survey files
2018-07-28 22:09:55.057 ERROR 12164 --- [Pool-1-worker-4] c.o.d.s.i.file.dao.DpiFileReaderImpl     : Ups ! Something wrong on reading file [./data/surveyFiles/PaxHeader]. Cause:[java.io.IOException: Is a directory]
2018-07-28 22:10:11.846 ERROR 12164 --- [           main] c.o.d.s.i.file.loaders.SurveyLoaderImpl  : Exception on paralel processing of files. Cause: {}

java.util.concurrent.ExecutionException: java.lang.NullPointerException
        at java.util.concurrent.ForkJoinTask.get(ForkJoinTask.java:1006) ~[na:1.8.0_171]
        at com.osc98.dpisurvey.service.implementation.file.loaders.SurveyLoaderImpl.insertDoubleParalell(SurveyLoaderImpl.java:83) [classes!/:0.0.2-SNAPSHOT]
        at com.osc98.dpisurvey.service.implementation.file.loaders.SurveyLoaderImpl.insertRecords(SurveyLoaderImpl.java:51) [classes!/:0.0.2-SNAPSHOT]
        at com.osc98.dpisurvey.service.implementation.file.loaders.SurveyLoaderImpl$$FastClassBySpringCGLIB$$739cd35e.invoke(<generated>) [classes!/:0.0.2-SNAPSHOT]
        at org.springframework.cglib.proxy.MethodProxy.invoke(MethodProxy.java:204) [spring-core-5.0.6.RELEASE.jar!/:5.0.6.RELEASE]
        at org.springframework.aop.framework.CglibAopProxy$CglibMethodInvocation.invokeJoinpoint(CglibAopProxy.java:746) [spring-aop-5.0.6.RELEASE.jar!/:5.0.6.RELEASE]
        at org.springframework.aop.framework.ReflectiveMethodInvocation.proceed(ReflectiveMethodInvocation.java:163) [spring-aop-5.0.6.RELEASE.jar!/:5.0.6.RELEASE]
        at org.springframework.transaction.interceptor.TransactionAspectSupport.invokeWithinTransaction(TransactionAspectSupport.java:294) ~[spring-tx-5.0.6.RELEASE.jar!/:5.0.6.RELEASE]
        at org.springframework.transaction.interceptor.TransactionInterceptor.invoke(TransactionInterceptor.java:98) ~[spring-tx-5.0.6.RELEASE.jar!/:5.0.6.RELEASE]
        at org.springframework.aop.framework.ReflectiveMethodInvocation.proceed(ReflectiveMethodInvocation.java:185) [spring-aop-5.0.6.RELEASE.jar!/:5.0.6.RELEASE]
        at org.springframework.aop.interceptor.ExposeInvocationInterceptor.invoke(ExposeInvocationInterceptor.java:92) ~[spring-aop-5.0.6.RELEASE.jar!/:5.0.6.RELEASE]
        at org.springframework.aop.framework.ReflectiveMethodInvocation.proceed(ReflectiveMethodInvocation.java:185) [spring-aop-5.0.6.RELEASE.jar!/:5.0.6.RELEASE]
        at org.springframework.aop.framework.CglibAopProxy$DynamicAdvisedInterceptor.intercept(CglibAopProxy.java:688) ~[spring-aop-5.0.6.RELEASE.jar!/:5.0.6.RELEASE]
        at com.osc98.dpisurvey.service.implementation.file.loaders.SurveyLoaderImpl$$EnhancerBySpringCGLIB$$f72fa546.insertRecords(<generated>) ~[classes!/:0.0.2-SNAPSHOT]
        at com.osc98.dpisurvey.service.implementation.RecordsControlServiceImpl.fillDBSurvey(RecordsControlServiceImpl.java:60) ~[classes!/:0.0.2-SNAPSHOT]
        at com.osc98.dpisurvey.service.implementation.RecordsControlServiceImpl$$FastClassBySpringCGLIB$$659370eb.invoke(<generated>) ~[classes!/:0.0.2-SNAPSHOT]
        at org.springframework.cglib.proxy.MethodProxy.invoke(MethodProxy.java:204) [spring-core-5.0.6.RELEASE.jar!/:5.0.6.RELEASE]
        at org.springframework.aop.framework.CglibAopProxy$CglibMethodInvocation.invokeJoinpoint(CglibAopProxy.java:746) [spring-aop-5.0.6.RELEASE.jar!/:5.0.6.RELEASE]
        at org.springframework.aop.framework.ReflectiveMethodInvocation.proceed(ReflectiveMethodInvocation.java:163) [spring-aop-5.0.6.RELEASE.jar!/:5.0.6.RELEASE]
        at org.springframework.aop.aspectj.MethodInvocationProceedingJoinPoint.proceed(MethodInvocationProceedingJoinPoint.java:88) ~[spring-aop-5.0.6.RELEASE.jar!/:5.0.6.RELEASE]
        at com.osc98.dpisurvey.aop.PerformanceMonitor.logExecutionTime(PerformanceMonitor.java:21) ~[classes!/:0.0.2-SNAPSHOT]
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[na:1.8.0_171]
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[na:1.8.0_171]
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[na:1.8.0_171]
        at java.lang.reflect.Method.invoke(Method.java:498) ~[na:1.8.0_171]
        at org.springframework.aop.aspectj.AbstractAspectJAdvice.invokeAdviceMethodWithGivenArgs(AbstractAspectJAdvice.java:644) ~[spring-aop-5.0.6.RELEASE.jar!/:5.0.6.RELEASE]
        at org.springframework.aop.aspectj.AbstractAspectJAdvice.invokeAdviceMethod(AbstractAspectJAdvice.java:633) ~[spring-aop-5.0.6.RELEASE.jar!/:5.0.6.RELEASE]
        at org.springframework.aop.aspectj.AspectJAroundAdvice.invoke(AspectJAroundAdvice.java:70) ~[spring-aop-5.0.6.RELEASE.jar!/:5.0.6.RELEASE]
        at org.springframework.aop.framework.ReflectiveMethodInvocation.proceed(ReflectiveMethodInvocation.java:174) [spring-aop-5.0.6.RELEASE.jar!/:5.0.6.RELEASE]
        at org.springframework.transaction.interceptor.TransactionAspectSupport.invokeWithinTransaction(TransactionAspectSupport.java:294) ~[spring-tx-5.0.6.RELEASE.jar!/:5.0.6.RELEASE]
[... cut ...]
        at java.util.stream.ForEachOps$ForEachOp$OfRef.accept(ForEachOps.java:184) ~[na:1.8.0_171]
        at java.util.Spliterators$ArraySpliterator.forEachRemaining(Spliterators.java:948) ~[na:1.8.0_171]
        at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:481) ~[na:1.8.0_171]
        at java.util.stream.ForEachOps$ForEachTask.compute(ForEachOps.java:291) ~[na:1.8.0_171]
        at java.util.concurrent.CountedCompleter.exec(CountedCompleter.java:731) ~[na:1.8.0_171]
        ... 4 common frames omitted

```

userdeveloper98, 2018-07-29:
Hi @Fourdee 

> On the main barchart for "devices and installed version". I strongly feel a simple "Device install count" (without versions) would make this cleaner for the end user to read (less busy)?

- **Done**

> Something failed when running the java program

(Seems like inside surveyFiles folder where another folder, while it was expected only files :disappointed: `on reading file [./data/surveyFiles/PaxHeader]. Cause:[java.io.IOException: Is a directory]` )
- **Fixed** 

>  CPU arch selection has incorrect results for me

Can't find what it can be :cry:  Is this happening on provided samples or on real data?
(samples are generated absolutely random, so you may see for example that RasspberyPi have 8 cores while in reality, this model have 4 cores, the same go for cpu arch)

  [AnotherVersion](https://files.fm/u/gxactz9x#_)

userdeveloper98, 2018-08-01:
[v 1.0.0](https://files.fm/u/q36jxqg6) Should be ready to be deployed as it is, next iteration will be minor performance optimizations.
New features:
- Automatically update records on file change
    - on survey file update, will reload and update the table and chart automatically
    - on softwareList.properties update, will restart and rebuild all database
- UI Improvements (UI block, new tab transition animation)
- Minor performance optimizations
- Minor bug fixes
- Added Last Update timestamp on the bottom of the page (for each new/updated survey file)
- Changed mapping to /survey instead of survey.xhtml
- Removed "TODO" tabs

Fourdee, 2018-08-01:
@userdeveloper98 

Nice one, good progress 👍 

Some feedback from my end:
- What does the ```,``` and the additional number mean in this. Is it a total of ```3```?
[image]
- "Device install count", would be great if this could be automatically ordered ascending? EG: Highest install count devices at the top.
- Some minor offset alignment issues under Chrome (100% zoom):
[image]
- Running under 90% zoom seems to resolve
- The dark grey text on black background is really hard to read (at least for me). Would it be possible to make this a lighter grey overall?
- Would be great if we could use the same Font/colour of "Survey dashboard" title, for all headers/titles on the page? eg: 
[image]

userdeveloper98, 2018-08-02:
@MichaIng 
[v1.0.1](https://files.fm/u/8mqrutu9)
- Fixed toolTip, instead of `3,21` it will show `Count: 3`
- Device install count is now sorted (Highest install count devices at the top)
- Fixed alignment issues. Now charts are responsive to screen size.
- Chart tick labels are now almost white :smiley:  `#D2D2D2`. Looks nicer.
- Changed the font and colors of chart titles. (Michroma, `#C2DE60`)

userdeveloper98, 2018-08-04:
Hi @Fourdee 
Thanks a lot for the feedback!
[v1.0.2](https://files.fm/u/s7yhn2xe)
- Fixed "The top item does not display the install count when hovered over"
- Changed "Dietpi Versions" to "DietPi versions"
- Install count chart horizontal tick now will display only integers. 
- Hopefully fixed "margins" issues with device Install count. (At least for Chromium v67 and Firefox v61)
- DietPi Versions bar chart now has the tooltip.

```
c.o.d.s.i.file.dao.SurveyFileReaderImpl  : UPS ! templateLine:[((aSOFTWARE[${aSOFTWARE_NAME6_11[<softwareIndex>]}]++))]. surveyLine[((aSOFTWARE[${aSOFTWARE_NAME6_11[]}]++))]. Cause:[java.lang.NumberFormatException: For input string: ""]
c.o.d.s.i.file.dao.SurveyFileReaderImpl  : UPS ! templateLine:[((aSOFTWARE[${aSOFTWARE_NAME6_11[<softwareIndex>]}]++))]. surveyLine[((aSOFTWARE[${aSOFTWARE_NAME6_11[]}]++))]. Cause:[java.lang.NumberFormatException: For input string: ""]
```
As I can see from logs, the issue is caused by outdated survey file.
It is `((aSOFTWARE[${aSOFTWARE_NAME6_11[]}]++))` < no number inside [ ].
While we are expecting `((aSOFTWARE[${aSOFTWARE_NAME6_11[99]}]++))` < For example 99 
This is the reason for `NumberFormatException: For input string: ""`

> These seem to create blank entries in the database tab

- Now we will skip db insert for all records that don't have **DietPi Version** or **Device Name** fields
    - An error will be raised in logs on each skip `Record is considered empty and will be skipped.`

> Does the program automatically re-scan/update the survey database during runtime? 
- Yes! the program monitors all files inside the `dpi.survey.surveysDirectory.directory` and instantly update the record in DB, so you will have always the latest information.
- If you change something in `config` folder ex: `dpi.survey.softwareList.file` or `dpi.survey.surveyTemplate.file` the program will automatically drop all database, restart and rescan everything from scratch.

[image]
- Have you updated the `dpi.survey.softwareList.file` file (looks like default one) ? The number from : ((aSOFTWARE[${aSOFTWARE_NAME6_11[**5**]}]++)) should be translated to a software title via mapping from the mentioned `dpi.survey.softwareList.file`.

Fourdee, 2018-08-05:
Notes: 
- Current software list in format:
```
0=OpenSSH Client
1=Samba Client
2=NULL
3=MC
4=ViFM
5=ALSA
6=Xserver
7=FFmpeg
8=Java
9=Node.js
10=iftop
11=IPTraf
12=Iperf
13=MTR-Tiny
14=nLoad
15=tcpdump
16=Build-Essentials
17=Git Client
18=Emacs
19=Jed
20=Vim
21=Vim-Tiny
22=QuiteRSS
23=LXDE
24=MATE
25=XFCE
26=GNUStep
27=TightVNC Server
28=VNC4 Server
29=XRDP
30=NoMachine
31=Kodi
32=YMPD
33=AirSonic
34=SubSonic 6
35=SqueezeBox
36=SqueezeLite
37=Shairport Sync
38=BruteFIR
39=ReadyMedia
40=Ampache
41=Emby Server
42=Plex Media Server
43=Murmur
44=Transmission
45=Deluge
46=qBitTorrent
47=ownCloud
48=Pydio
49=Gogs
50=Syncthing
51=OpenTyrian
52=Cuberite
53=MineOS
54=Forums
55=Wordpress
56=Image Gallery
57=BaiKal
58=OpenBazaar
59=RPi Cam Control
60=WiFi Hotspot
61=Tor Hotspot
62=DietPi-Cloudshell
63=LinuxDash
64=PhpSysInfo
65=NetData
66=RPi-Monitor
67=NoIp
68=Remot3.it
69=RPi.GPIO
70=WiringPi
71=WebIOPi
72=I2c
73=Fail2Ban
74=InfluxDB
75=LASP
76=LAAP
77=Grafana
78=LESP
79=LEAP
80=Ubooquity
81=LLSP
82=LLAP
83=Apache2
84=Lighttpd
85=Nginx
86=Roon Extension Manager
87=SQlite
88=MariaDB
89=PHP
90=phpMyAdmin
91=Redis
92=CertBot
93=Pi-hole
94=ProFTP
95=vsFTPD
96=Samba
97=OpenVPN
98=HaProxy
99=EmonPi
100=PiJuice
101=Log Rotate
102=Rsyslog
103=DietPi-Ramlog
104=Dropbear
105=OpenSSH Server
106=Lidarr
107=rTorrent
108=AmiBerry
109=NFS
110=NFS Client
111=UrBackup server
112=DXX-Rebirth
113=Chromium
114=Nextcloud
115=Webmin
116=SickRage
117=PiVPN
118=Mopidy
119=CAVA
120=RealVNC Server
121=Roon Bridge
122=Node-Red
123=Mosquitto
124=NAA daemon
125=Tomcat8
126=LibSSL1.0.0
127=NeoVim
128=MPD
129=O!MPD
130=Python Pip
131=Blynk Server
132=Aria2
133=YaCy
134=Tonido
135=IceCast
136=MotionEye
137=CloudPrint
138=VirtualHere
139=SABnzbd
140=SDL2
141=Spotify Connect Web
142=CouchPotato
143=Koel
144=Sonarr
145=Radarr
146=Tautulli
147=Jackett
148=NULL
149=NZBget
150=Mono
151=Nvidia
152=Avahi-Daemon
153=OctoPrint
154=Roon Server
155=HTPC Manager
156=Steam
157=Home Assistant
158=Minio
159=Allo
160=Allo_update
161=FuguHub
162=Docker
163=Gmediarender
164=Nukkit
165=Gitea
166=PI-SPC
167=Raspotify
168=NULL
169=Google AIY
```

MichaIng, 2018-08-05:
https://github.com/Fourdee/DietPi/blob/testing/.meta/survey_report bzw. also shows the lists for all versions since v6.9. The arrays for follow up versions are created based on the previous one with made changes applied incrementally. With this the IDs can be correctly linked for each DietPi version.
v6.13 changes are no yet implemented of course 😉

Fourdee, 2018-08-05:
@userdeveloper98 

Legend, great work 👍 

[image]

Ok, i'll leave this running for a few days on the test system http://dietpi-4d.ddns.net:8888/survey, if no issues, i'll add to our main server.

---

Some suggestions for future features:
- I think having the below information would be great on the main page. As more than 50% of users opt out, would be nice to have a total install count?
[image]
- Would be also great if we could have a bar chart for installed software count, eg:
[image]
- And you should add "Created by your_name" to the page, credit where credit is due :)
- Minor, the logo really needs a white background?
[image]

Fourdee, 2018-08-05:
@MichaIng @userdeveloper98 

> The arrays for follow up versions are created based on the previous one with made changes applied incrementally. With this the IDs can be correctly linked for each DietPi version.
v6.13 changes are no yet implemented of course

This is the current scrape ```surveyTemplate.properties```:
```
#!/bin/bash
# -------------------------
((aSURVEY_VERSION[<surveyVersion>]++))
# ------------------------
((aDIETPI_VERSION[<dietpiVersion>]++))
((aDEVICE_NAME[<deviceName>]++))
((aCPU_ARCH[<cpuArch>]++))
((aCPU_COUNT[<cpuCount>]++))
((aDISTRO_VERSION[<distroVersion>]++))
((aAUTOSTART_OPTION[<autoStartOption>]++))
((aAUTO_SETUP_AUTOMATED[<autoSetupAutomated>]++))
((aNETWORK_INTERFACE[<networkInterface>]++))
# ------------------------
# DietPi-Software Installe
# ------------------------
((aSOFTWARE[${aSOFTWARE_NAME6_11[<softwareIndex>]}]++))
```

Not sure what would need to be changed here?

userdeveloper98, 2018-08-05:
@Fourdee  @MichaIng 

`((aSOFTWARE[${aSOFTWARE_NAME6_11[<softwareIndex>]}]++))` vs
`((aSOFTWARE[${aSOFTWARE_NAME6_12[<softwareIndex>]}]++))`  It's not a problem at all.
Just ad new line at the bottom of the file and program will understand from where to read new softwareIndex.
It will look for `SOFTWARE_NAME6_12` or `SOFTWARE_NAME6_13` or `SOFTWARE_NAME6_XX` and match with appropriate. No huge maintenance work needed.
___

> Not sure what would need to be changed here?

Add:
`((aSOFTWARE[${aSOFTWARE_NAME6_11[<softwareIndex>]}]++))`
`((aSOFTWARE[${aSOFTWARE_NAME6_12[<softwareIndex>]}]++))`
`((aSOFTWARE[${aSOFTWARE_NAME6_13[<softwareIndex>]}]++))`
`((aSOFTWARE[${aSOFTWARE_NAME6_14[<softwareIndex>]}]++))`
etc.. 
___
Regarding software list, 
```
0=OpenSSH Client
1=Samba Client
etc..
```
Let's just agree to have the same list for all versions. 

1. if a new software will appear 
    -just increment the index and add the new title. 
2. If something will be deleted
    -do not delete it from the list. otherwise, we will break backward compatibility :smile: .

Having a separate list of `index=title` for each version is overhead in my opinion.

Another option (mapping free) will be to have title directly in survey file ex: `((aSOFTWARE[${aSOFTWARE_NAME6_12[<softwareTitle>]}]++))` but this is a bit complicated as I understand.

___

- opt-out/opt-in chart, I think a pie chart will be perfect for this type of information?
    - a circle divided into 2 colors, similar as for Cpu Arch
-  installed software count, the same format as for device install count?
- logo, looks better ? :smile: 
[image]

MichaIng, 2018-08-05:
@userdeveloper98 
> Let's just agree to have the same list for all versions.

This is exactly what I mean. The current survey_report indeed creates own list arrays for each DietPi version. Otherwise you will have quite large wrong entries, as IDs by times also switched to lower unused ones. So I really see no different solution, for up to current version at least, than having separate lists, e.g.
```
((aSOFTWARE[${aSOFTWARE_NAME6_11[<softwareIndex6_11>]}]++))
((aSOFTWARE[${aSOFTWARE_NAME6_12[<softwareIndex6_12>]}]++))
((aSOFTWARE[${aSOFTWARE_NAME6_13[<softwareIndex6_13>]}]++))
((aSOFTWARE[${aSOFTWARE_NAME6_14[<softwareIndex6_14>]}]++))
```
and then translating those variables with the related lists.

Adding the ID to name conversion to the upload script, will shift the overhead to the end user, currently way less efficiently. Currently a `dietpi-software list` call would be needed, which takes quite a while, being a no go from my point of view. Possibly users would opt out just because sending the report takes so much longer, and as well currently calling `dietpi-software list` within `dietpi-survey` which is called again within `dietpi-software` install, causes most likely issues due to $FP_WORK_DIR handling. My aim is to not allow running the same script two times (including nested call), so tmp file handling can never interfere.

To never again reuse software IDs, but instead always count up, as well shifts overhead to end user, as this would mean a growing amount of empty arrays, being initiated and looped through. By times development on old software stops, concepts being dropped, security holes appear etc, many reasons to drop software from out list. We definitely need to be allowed to reuse those empty IDs.

To keep the needed amount of lists under control, we could simply drop reading data from old DietPi version uploads, or completely remove/move the files.

userdeveloper98, 2018-08-05:
@Fourdee  @MichaIng 
How do we count opted-out users? Do they upload an empty file?

MichaIng, 2018-08-05:
@userdeveloper98 
> Do they upload an empty file?

Exactly 😃.

userdeveloper98, 2018-08-09:
[v1.0.3-SNAPSHOT](https://files.fm/u/4dgzrpnq)
- Opted in/out users chart

Fourdee, 2018-08-10:
@userdeveloper98 

Awesome, love the addition 👍 Although hate the results lol, only 1/4 opted in ☹️ 

[image]

Ok, give me a few days, will get this put on our dietpi.com server

Setting v6.13 so this doesnt get lost in pile.

MichaIng, 2018-08-10:
@userdeveloper98 
You do not read v6.9 survey files (with old structure, see OT), right? Would be also okay for me. I watch 0-5 of them being updated every day, but believe that most are not active any more.

However the opt out/in balance is quite different: https://dietpi.com/survey/

Total user count | 16922
---- | -----
Users opted in | 5974
Users opted out | 10948

Did you check, that `Opted Out` is really `Total minus Opted In`? 🙂 Remember I mixed this up at first.

userdeveloper98, 2018-08-12:
@MichaIng 
Give me an example of  v6.9 survey file and I'll tell you how to update template file, it will be readable instantly, the balance will be restored :)

[19 more replies]

closed 2018-10-06 by Fourdee