MichaIng/DietPi #1829: Image | NanoPC T4

Fourdee, 2018-06-08:
Many thanks to FriendlyElec for sending us sample dev boards!

http://www.friendlyarm.com/index.php?route=product/product&product_id=225
[image]

Fourdee, 2018-06-10:
[image]

Fourdee, 2018-06-16:
Unpack image to allow loopback mounting:
```
imgRePackerRK.exe rootfs.img
```

Repack:
```
imgRePackerRK system.img.dump
```

Fourdee, 2018-06-18:
Notes 150000 baud:

#### Windows:
[image]

#### Linux:

Fourdee, 2018-06-19:
> 1500000

Fourdee, 2018-06-19:
### Benchmarks:
[image]

Fourdee, 2018-06-19:
Seems starting the serial service, kills it:
```
[  OK  ] Started Serial Getty on ttyFIQ0.
[  OK  ] Started Getty on tty1.
[  OK  ] Started Serial Getty on ttyS2.

 4 ▒▒▒▒▒─▒▒[
            [e[0e8
                   4▒───▒[
                           m |4)
                                ───▒0
                                      m mt ao
                                              5▒▒───

                                                     di
                                                        gt

DbnUix D Q
D i
```

Fourdee, 2018-06-19:
Finally up and running:
https://twitter.com/DietPi_/status/1008993035823304709

#### ToDO:
- Instructions to write to EMMC
- dietpi.com addition
#### ToDo, post release (can be resolved with v6.10)
- Test audio
- Test GPU

Fourdee, 2018-06-19:
# Writing DietPi to the onboard EMMC:
#### Requires:
- NanoPC-T4
- USB-C Cable
- Ethernet connected (or WiFi can be setup with serial console during 1st run setup)
- A Windows system (to use the provided easy write tool with USB-C, Linux options require command line)

### Download and extract the image:
URL

### Writing the images to the EMMC
- Power off the NanoPC
- Plug in the USB-C cable and power cable (do not power on). Plug in the other end of the USB-C cable into your Windows system.
- Launch the program ```AndroidTool.exe```
-- Now we need to enter into MASKROM mode:
-- Hold down the ```boot``` and ```power``` button's on the NanoPC, for roughly 2-5 seconds.
-- When you see a ```MASKROM``` message appear in the ```AndroidTool.exe``` window, release the buttons.
-- Select ```Run```

When the transfer is completed, the system will automatically reboot and begin the 1st run setup.

Fourdee, 2018-06-19:
https://github.com/rockchip-linux/rk-rootfs-build/tree/master/packages/armhf/libmali
https://github.com/rockchip-linux/rk-rootfs-build/blob/master/overlay/etc/init.d/rockchip.sh#L29-L31
```
dpkg --add-architecture armhf
G_AGUP
```

Fourdee, 2018-06-29:
Completed (for now), GPU support will come at a later date (hopefully).

brianlinuxing, 2018-07-03:
Fantastic!

carlosedp, 2018-07-16:
I just flashed this image into my NanoPC-T4 but the included kernel has no modules (no Iptables, OpenVSwitch). How can I update to a new or alternate kernel that contains the modules?

I tried to install "linux-image-arm64", it installs and places the `vmlinuz, initrd` files into` /boot` but the board doesn't boot from these images. What's the correct procedure?

Thanks

Fourdee, 2018-07-17:
@carlosedp 

Thanks for the report 👍 

I'll take a look at our image and pre-image used. Will reply once I have an answer.

carlosedp, 2018-07-17:
I have recompiled the 4.4 kernel from FriendlyArm and enabled/compiled a bunch of modules that are needed for Docker.

I can upload them if you want. It's a matter of replacing kernel.img and adding the modules (from a tar.gz) to the rootfs.img.

Fourdee, 2018-07-17:
@carlosedp 

> I can upload them if you want. It's a matter of replacing kernel.img and adding the modules (from a tar.gz) to the rootfs.img.

Would be excellent if you could? I can update the image files here and re-upload.

Any chance you enabled CIFS kernel option for your kernel? On the pre-image we used from FriendlyARM, it was not enabled ☹️

carlosedp, 2018-07-17:
I've uploaded the files to: https://we.tl/n7JXaEHKAH

Here is the list of modules I compiled. If you need any additional ones let me know so I can rebuild it:

```
$ find /lib/modules/4.4.126/
/lib/modules/4.4.126/
/lib/modules/4.4.126/modules.builtin
/lib/modules/4.4.126/modules.devname
/lib/modules/4.4.126/modules.dep.bin
/lib/modules/4.4.126/modules.symbols
/lib/modules/4.4.126/build
/lib/modules/4.4.126/modules.builtin.bin
/lib/modules/4.4.126/source
/lib/modules/4.4.126/modules.alias
/lib/modules/4.4.126/modules.symbols.bin
/lib/modules/4.4.126/modules.order
/lib/modules/4.4.126/modules.dep
/lib/modules/4.4.126/modules.softdep
/lib/modules/4.4.126/kernel
/lib/modules/4.4.126/kernel/lib
/lib/modules/4.4.126/kernel/lib/ts_bm.ko
/lib/modules/4.4.126/kernel/lib/ts_kmp.ko
/lib/modules/4.4.126/kernel/lib/ts_fsm.ko
/lib/modules/4.4.126/kernel/crypto
/lib/modules/4.4.126/kernel/crypto/md4.ko
/lib/modules/4.4.126/kernel/crypto/des_generic.ko
/lib/modules/4.4.126/kernel/fs
/lib/modules/4.4.126/kernel/fs/cifs
/lib/modules/4.4.126/kernel/fs/cifs/cifs.ko
/lib/modules/4.4.126/kernel/fs/fscache
/lib/modules/4.4.126/kernel/fs/fscache/fscache.ko
/lib/modules/4.4.126/kernel/fs/configfs
/lib/modules/4.4.126/kernel/fs/configfs/configfs.ko
/lib/modules/4.4.126/kernel/fs/overlayfs
/lib/modules/4.4.126/kernel/fs/overlayfs/overlay.ko
/lib/modules/4.4.126/kernel/fs/autofs4
/lib/modules/4.4.126/kernel/fs/autofs4/autofs4.ko
/lib/modules/4.4.126/kernel/fs/cachefiles
/lib/modules/4.4.126/kernel/fs/cachefiles/cachefiles.ko
/lib/modules/4.4.126/kernel/fs/ceph
/lib/modules/4.4.126/kernel/fs/ceph/ceph.ko
/lib/modules/4.4.126/kernel/fs/nfs
/lib/modules/4.4.126/kernel/fs/nfs/flexfilelayout
/lib/modules/4.4.126/kernel/fs/nfs/flexfilelayout/nfs_layout_flexfiles.ko
/lib/modules/4.4.126/kernel/drivers
/lib/modules/4.4.126/kernel/drivers/usb
/lib/modules/4.4.126/kernel/drivers/usb/gadget
/lib/modules/4.4.126/kernel/drivers/usb/gadget/function
/lib/modules/4.4.126/kernel/drivers/usb/gadget/function/usb_f_fs.ko
/lib/modules/4.4.126/kernel/drivers/usb/gadget/function/usb_f_acm.ko
/lib/modules/4.4.126/kernel/drivers/usb/gadget/function/u_serial.ko
/lib/modules/4.4.126/kernel/drivers/usb/gadget/function/usb_f_mass_storage.ko
/lib/modules/4.4.126/kernel/drivers/usb/gadget/libcomposite.ko
/lib/modules/4.4.126/kernel/drivers/vhost
/lib/modules/4.4.126/kernel/drivers/vhost/vringh.ko
/lib/modules/4.4.126/kernel/drivers/vhost/vhost.ko
/lib/modules/4.4.126/kernel/drivers/vhost/vhost_net.ko
/lib/modules/4.4.126/kernel/drivers/gpu
/lib/modules/4.4.126/kernel/drivers/gpu/arm
/lib/modules/4.4.126/kernel/drivers/gpu/arm/mali400
/lib/modules/4.4.126/kernel/drivers/gpu/arm/mali400/mali
/lib/modules/4.4.126/kernel/drivers/gpu/arm/mali400/mali/mali.ko
/lib/modules/4.4.126/kernel/net
/lib/modules/4.4.126/kernel/net/llc
/lib/modules/4.4.126/kernel/net/llc/llc2.ko
/lib/modules/4.4.126/kernel/net/llc/llc.ko
/lib/modules/4.4.126/kernel/net/hsr
/lib/modules/4.4.126/kernel/net/hsr/hsr.ko
/lib/modules/4.4.126/kernel/net/dsa
[... 186 lines cut ...]
/lib/modules/4.4.126/kernel/net/netfilter/nf_conntrack_broadcast.ko
/lib/modules/4.4.126/kernel/net/netfilter/xt_quota.ko
/lib/modules/4.4.126/kernel/net/netfilter/nf_nat_redirect.ko
/lib/modules/4.4.126/kernel/net/netfilter/nf_synproxy_core.ko
/lib/modules/4.4.126/kernel/net/netfilter/xt_multiport.ko
/lib/modules/4.4.126/kernel/net/netfilter/xt_cgroup.ko
/lib/modules/4.4.126/kernel/net/netfilter/xt_nfacct.ko
/lib/modules/4.4.126/kernel/net/netfilter/xt_state.ko
/lib/modules/4.4.126/kernel/net/netfilter/xt_mac.ko
/lib/modules/4.4.126/kernel/net/netfilter/xt_policy.ko
/lib/modules/4.4.126/kernel/net/netfilter/nf_conntrack_sane.ko
/lib/modules/4.4.126/kernel/net/netfilter/nf_conntrack_proto_dccp.ko
/lib/modules/4.4.126/kernel/net/netfilter/xt_HMARK.ko
/lib/modules/4.4.126/kernel/net/netfilter/nf_log_common.ko
/lib/modules/4.4.126/kernel/net/netfilter/xt_TEE.ko
/lib/modules/4.4.126/kernel/net/netfilter/nft_compat.ko
/lib/modules/4.4.126/kernel/net/netfilter/nft_nat.ko
/lib/modules/4.4.126/kernel/net/ceph
/lib/modules/4.4.126/kernel/net/ceph/libceph.ko
/lib/modules/4.4.126/kernel/net/bridge
/lib/modules/4.4.126/kernel/net/bridge/br_netfilter.ko
/lib/modules/4.4.126/kernel/net/bridge/bridge.ko
/lib/modules/4.4.126/modules.alias.bin

```

Fourdee, 2018-07-20:
@carlosedp 

> Fine! Just saw I didn't enable memory cgroups on the kernel compilation. The default from FriendlyArm didn't have it either... Do you see anything other missing so I can add?

From my side, only CIFS is critical for our user base.

carlosedp, 2018-07-20:
Done, after a ton of testing I've compiled a Kernel that contains most needed modules for Docker, Kubernetes, Weave overlay network, NFS, CIFS and many others.

```
$ sudo modprobe cifs
carlosedp@NanoT4-1:~ (kubearm:kube-system) $ lsmod
Module                  Size  Used by
cifs                  573440  0
xt_NFLOG               16384  1
veth                   16384  0
dummy                  16384  0
...
```

The files can be downloaded on https://we.tl/E2d2n7RqSD and replaced on the DietPi package. 

Also I found out that the `/etc/machine-id` that's on the rootfs is static and not generated dynamically. In my case, Weave Net (an overlay network for Kubernetes) uses this to generate it's unique ID and since they are equal, it's not that unique :) More info [here](https://www.thegeekdiary.com/centos-rhel-7-how-to-change-the-machine-id/).

The ideal is to have this file auto-generated only on first config.

carlosedp, 2018-07-21:
Actually I successfully created the image for the Firefly. Compiled the Firefly kernel to it with all parameters from the NanoT4 (modules and all). I tried with the FriendlyARM kernel but if froze on heavy network test, probably some DTB messed up.

I successfully flashed it all with DietPi Rootfs and it's working perfectly. If you want I can sent the package to you. Let me do a couple more tests.

Will try to compile the newer 4.4.138 kernel from Rockchip now.

carlosedp, 2018-07-21:
@Fourdee , just please test with your cases before using this image. There might be a couple of things I didn't test on video subsystems or WiFi. Let me know if you find something wrong.

Fourdee, 2018-07-23:
Hmmm, i'am missing something here in regards to rootfs.img creation.

I tried created a new EXT4 image with MBR, added the files, 

```
dd if=/dev/zero of=rootfs.img.iso bs=1G count=1
mkfs.ext4 rootfs.img.iso
```

then packing fails
```
C:\Users\4D>C:\Users\4D\Desktop\DietPi_NanoPCT4-ARMv8-Stretch\Image_and_Tools\img_pack\imgRePackerRK.exe C:\Users\4D\Desktop\DietPi_NanoPCT4-ARMv8-Stretch\Image_and_Tools\rootfs.img.iso

        imgRepackerRK (version 1.06 windows)
        Rockchip firmware batch/update images unpacker/packer

        (c) RedScorpio, Moscow, 2013-2017
            RedScorpio@land.ru

        Detected OS:    Windows 8 Home [build 9200] x64
        Real OS:        Windows 10 Home [build 16299] x64
        ==========================[ START ]==========================

        --- Firmware unpacking ---

        2-nd layer file unpacking
        ext4 image format detected
                Unsupported format (ext4 image)
        Unpacking filed
```

I'll need to do some reading on this (if documentation exists?).

carlosedp, 2018-07-23:
Is this how you created it before? I don't know.. never used these tools.

When I created a rootfs, I used this:

 ```
dd if=/dev/zero of=linuxroot.img bs=1M count=2048
sudo  mkfs.ext4  linuxroot.img
mkdir  rootfs
sudo mount linuxroot.img rootfs/
sudo cp -rfp temp/*  rootfs/
sudo umount rootfs/
e2fsck -p -f linuxroot.img
resize2fs  -M linuxroot.img
```

Got it from http://en.t-firefly.com/doc/product/info/id/291.html

Fourdee, 2018-07-24:
- 🈯️ CIFS test 
- 🈴  WiFi test | Scan, very intermittent, some times fails to pull any results due to driver/HW issue with new kernel?
```
[  322.720581] CFG80211-ERROR) wl_escan_handler : escan is not ready ndev ffffffc0f1ead000 drv_status 0x0 e_type 69 e_states 8

root@DietPi:~# iwlist wlan0 scan
wlan0     No scan resultt
```

@carlosedp 

Aside from the issue with WiFi, great work, very nice, thank you 👍 
Ideally, we need to find out why the scan is failing 80% of the time.

#### I've updated the image, based on your kernel for testing: https://dietpi.com/downloads/testing/DietPi_NanoPCT4-ARMv8-Stretch.7z

carlosedp, 2018-07-24:
I just enabled the wlan0 interface and did the scan a couple times. I'd say that from 10 times it didn't find SSIDs in 3-4. I don't know if it's something on the driver in this kernel or if it happened before. TBH I don't use wlan and never enabled it.

I researched a bit and the WLAN module is made by Rockchip itself (http://rockchip.fr/ampak/AP6356_datasheet_V1.0_07252014.pdf) and the driver is in their kernel tree. 

I compared the wlan driver trees between the FriendlyARM and Rockchip kernels and there are not many changes, they even disabled something related to sleep like described on this commit(https://github.com/rockchip-linux/kernel/commit/243f6ead0f388afb0a010d5524bdafd421848acc):

```
$ diff -ENwbur  ~/ARM/friendlyarm-kernel/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/ ~/ARM/rockchip-kernel/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd
diff -ENwbur /home/carlosedp/ARM/friendlyarm-kernel/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_config.c /home/carlosedp/ARM/rockchip-kernel/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_config.c
--- /home/carlosedp/ARM/friendlyarm-kernel/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_config.c        2018-07-17 11:30:14.524986506 -0500
+++ /home/carlosedp/ARM/rockchip-kernel/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_config.c   2018-07-21 15:39:32.938030000 -0500
@@ -2623,6 +2623,8 @@
                conf->txglomsize = SDPCM_MAXGLOM_SIZE;
        conf->deferred_tx_len = conf->txglomsize;
 #endif
+       if (conf->chip == BCM4354_CHIP_ID && conf->chiprev == 1)
+               dhd_slpauto = 0;

        return 0;
 }
diff -ENwbur /home/carlosedp/ARM/friendlyarm-kernel/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_gpio.c /home/carlosedp/ARM/rockchip-kernel/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_gpio.c
--- /home/carlosedp/ARM/friendlyarm-kernel/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_gpio.c  2018-07-17 11:30:14.524986506 -0500
+++ /home/carlosedp/ARM/rockchip-kernel/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_gpio.c     2018-07-21 15:39:32.938030000 -0500
@@ -244,6 +244,7 @@
 #ifdef CUSTOMER_OOB
        int host_oob_irq = -1;
        uint host_oob_irq_flags = 0;
+       int irq_flags = -1;
 #endif

        /* Please check your schematic and fill right GPIO number which connected to
@@ -289,11 +290,14 @@
        printf("%s: host_oob_irq: %d\n", __FUNCTION__, host_oob_irq);

 #ifdef HW_OOB
-#ifdef HW_OOB_LOW_LEVEL
-       host_oob_irq_flags = IORESOURCE_IRQ | IORESOURCE_IRQ_LOWLEVEL | IORESOURCE_IRQ_SHAREABLE;
-#else
-       host_oob_irq_flags = IORESOURCE_IRQ | IORESOURCE_IRQ_HIGHLEVEL | IORESOURCE_IRQ_SHAREABLE;
-#endif
+       host_oob_irq_flags = IORESOURCE_IRQ | IORESOURCE_IRQ_SHAREABLE;
+       irq_flags = rockchip_wifi_get_oob_irq_flag();
+       if (irq_flags == 1)
+               host_oob_irq_flags |= IORESOURCE_IRQ_HIGHLEVEL;
+       else if (irq_flags == 0)
+               host_oob_irq_flags |= IORESOURCE_IRQ_LOWLEVEL;
+       else
+               pr_warn("%s: unknown oob irqflags !\n", __func__);
 #else
        host_oob_irq_flags = IORESOURCE_IRQ | IORESOURCE_IRQ_HIGHEDGE | IORESOURCE_IRQ_SHAREABLE;
 #endif
diff -ENwbur /home/carlosedp/ARM/friendlyarm-kernel/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_linux.c /home/carlosedp/ARM/rockchip-kernel/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_linux.c
--- /home/carlosedp/ARM/friendlyarm-kernel/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_linux.c 2018-07-17 11:30:14.540994507 -0500
+++ /home/carlosedp/ARM/rockchip-kernel/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_linux.c    2018-07-21 15:39:32.938030000 -0500
@@ -5029,6 +5029,7 @@
 #endif /* DHD_PCIE_RUNTIMEPM */

        DHD_GENERAL_LOCK(&dhd->pub, flags);
+#ifdef BCMPCIE
        if (DHD_BUS_CHECK_SUSPEND_OR_SUSPEND_IN_PROGRESS(&dhd->pub)) {
                DHD_ERROR(("%s: bus is in suspend(%d) or suspending(0x%x) state!!\n",
                        __FUNCTION__, dhd->pub.busstate, dhd->pub.dhd_bus_busy_state));
@@ -5047,7 +5048,12 @@
                return NETDEV_TX_BUSY;
 #endif
        }
-
+#else
+       if (DHD_BUS_CHECK_SUSPEND_OR_SUSPEND_IN_PROGRESS(&dhd->pub)) {
+               DHD_ERROR(("%s: bus is in suspend(%d) or suspending(0x%x) state!!\n",
+                       __FUNCTION__, dhd->pub.busstate, dhd->pub.dhd_bus_busy_state));
+       }
+#endif
        DHD_OS_WAKE_LOCK(&dhd->pub);
        DHD_PERIM_LOCK_TRY(DHD_FWDER_UNIT(dhd), lock_taken);

```

Honestly I have no idea on what might be wrong. You never saw this behavior before?

carlosedp, 2018-07-24:
Also there are other driver for it (I don't know which is used) on https://github.com/rockchip-linux/kernel/tree/release-4.4/drivers/net/wireless/rockchip_wlan/cywdhd/bcmdhd.

This one doesn't exist on FriendlyARM's tree but is disabled on my build and also disabled on all trees default config for the platform.

```
Symbol: CYW_BCMDHD [=n]                                                                                              x
  x Type  : tristate                                                                                                     x
  x Prompt: Cypress wireless sdio cards support                                                                          x
  x   Location:                                                                                                          x
  x     -> Device Drivers                                                                                                x
  x       -> Network device support (NETDEVICES [=y])                                                                    x
  x         -> Wireless LAN (WLAN [=y])                                                                                  x
  x           -> Rockchip Wireless LAN support (WL_ROCKCHIP [=y])                                                        x
  x (1)         -> ap6xxx wireless sdio cards support (AP6XXX [=y])                                                      x
  x   Defined at drivers/net/wireless/rockchip_wlan/cywdhd/Kconfig:2                                                     x
  x   Depends on: NETDEVICES [=y] && WLAN [=y] && WL_ROCKCHIP [=y] && !AP6XXX [=y]                                       x
  x   Selects: WIRELESS_EXT [=y] && WEXT_PRIV [=y] && CFG80211_WEXT [=y]        
```

The other one is:

```
  x Symbol: AP6XXX [=y]                                                                                                  x
  x Type  : tristate                                                                                                     x
  x Prompt: ap6xxx wireless sdio cards support                                                                           x
  x   Location:                                                                                                          x
  x     -> Device Drivers                                                                                                x
  x       -> Network device support (NETDEVICES [=y])                                                                    x
  x         -> Wireless LAN (WLAN [=y])                                                                                  x
  x (1)       -> Rockchip Wireless LAN support (WL_ROCKCHIP [=y])                                                        x
  x   Defined at drivers/net/wireless/rockchip_wlan/rkwifi/Kconfig:2                                                     x
  x   Depends on: NETDEVICES [=y] && WLAN [=y] && WL_ROCKCHIP [=y]                                                       x
  x   Selects: CFG80211 [=y] && MAC80211 [=y]               
```

So it doesn't look like something in the driver. I might try to replace this `rkwifi` driver from by the one from FriendlARM to test. If I build it would you test?

Fourdee, 2018-07-25:
> I might try to replace this rkwifi driver from by the one from FriendlARM to test. If I build it would you test?

Yep, please, would be great, I can test 👍 

> Honestly I have no idea on what might be wrong. You never saw this behavior before?

Not that I can remember. If we have issues with onboard WiFi during image creation/testing, we always make a note of it in the ticket. 
Either way, if I get a chance/time, i'll re-test and verify the FriendlyARM image.

carlosedp, 2018-07-25:
I've built the kernel with wlan driver from FriendlyARM tree: https://we.tl/4Ku7XBOpRD

carlosedp, 2018-07-27:
@Fourdee any news? I believe the download expires in 1 week.

Fourdee, 2018-07-27:
@carlosedp 

Checking it out now 👍

Fourdee, 2018-07-27:
@carlosedp 

I believe we should use your updated kernel/modules:
- Make users aware of the intermittent WiFi scan issue
- CIFS is more critical in my eyes, than WiFi.

Thoughts?

carlosedp, 2018-07-27:
I could check the CIFS problem but since WiFi is having the same behaviour, I'd keep the original driver from the Kernel instead of the driver from the other tree. The previous files I've sent.

Good to go but keep both images available, the one you already have and this one with newer Kernel and the modules that have been enabled.

GTG for me since it's running well on my board that is a Kubernetes node:

```
$ uname -a
Linux NanoT4-1 4.4.138 #3 SMP Sat Jul 21 18:29:19 EST 2018 aarch64 GNU/Linux

$ uptime
 18:41:43 up 4 days,  6:08,  1 user,  load average: 2.72, 2.67, 2.72

 $ lsmod
Module                  Size  Used by
configs                45056  0
xt_NFLOG               16384  1
veth                   16384  0
dummy                  16384  0
vport_vxlan            16384  1
vxlan                  45056  1 vport_vxlan
ip6_udp_tunnel         16384  1 vxlan
udp_tunnel             16384  1 vxlan
openvswitch            94208  3 vport_vxlan
nf_defrag_ipv6         53248  1 openvswitch
nfnetlink_log          20480  2 xt_NFLOG
xt_statistic           16384  10
xt_nat                 16384  58
xt_recent              20480  2
ipt_REJECT             16384  0
nf_reject_ipv4         16384  1 ipt_REJECT
xt_tcpudp              16384  166
ip_vs                 184320  0
ip_set_hash_ip         32768  19
xt_set                 20480  10
ip_set                 40960  2 ip_set_hash_ip,xt_set
xt_comment             16384  327
xt_mark                16384  5
ipt_MASQUERADE         16384  4
nf_nat_masquerade_ipv4    16384  1 ipt_MASQUERADE
nf_conntrack_netlink    40960  0
nfnetlink              16384  5 nfnetlink_log,ip_set,nf_conntrack_netlink
xfrm_user              32768  1
xfrm_algo              16384  1 xfrm_user
iptable_nat            16384  1
nf_conntrack_ipv4      24576  7
nf_defrag_ipv4         16384  1 nf_conntrack_ipv4
nf_nat_ipv4            16384  1 iptable_nat
xt_addrtype            16384  3
iptable_filter         16384  6
xt_conntrack           16384  6
nf_nat                 24576  3 nf_nat_ipv4,xt_nat,nf_nat_masquerade_ipv4
nf_conntrack          126976  8 ip_vs,openvswitch,nf_nat,nf_nat_ipv4,xt_conntrack,nf_nat_masquerade_ipv4,nf_conntrack_netlink,nf_conntrack_ipv4
br_netfilter           24576  0
bridge                126976  1 br_netfilter
stp                    16384  1 bridge
llc                    16384  2 stp,bridge
ipv6                  405504  278 ip_vs,bridge,nf_defrag_ipv6
overlay                45056  23
ip_tables              24576  2 iptable_filter,iptable_nat
x_tables               32768  14 xt_statistic,xt_mark,xt_comment,xt_recent,ip_tables,xt_tcpudp,ipt_MASQUERADE,xt_NFLOG,xt_conntrack,xt_nat,xt_set,iptable_filter,ipt_REJECT,xt_addrtype

```

Fourdee, 2018-07-28:
@carlosedp 

I'll do some more tests on the WiFi pre-friendlyARM WiFi driver:
Aside from the intermittent scan failures, if the Wifi is fully functional once connected + reboots, i'll mark this is a known issue on the device, and, set this new image live.

carlosedp, 2018-07-30:
Ah, one thing I noticed on the dietpi-config application is that if I set the CPU threshold, it's not saved and applied to the NanoPC-T4 board.

Fourdee, 2018-08-01:
Once connected, appears to run fine, will mark this as a known issue with WiFi and set new image live with @carlosedp kernel/modules.

@carlosedp 

Please email me your address, I'll send you some DietPi stickers for your troubles 👍 
daniel.knight@dietpi.com

Fourdee, 2018-08-01:
Completed.

Great work and thanks again for contributing this to DietPi @carlosedp 👍

closed 2018-08-01 by Fourdee

carlosedp, 2018-08-01:
Good to know @Fourdee , happy to contribute to the project!

I also created the package for the Firefly RK3399 and been using it for a while. Same Kernel, Firefly DTBs and DietPi rootfs. I can upload it in case you want to put on a different issue (like this) so people could test it.

Fourdee, 2018-08-02:
> I also created the package for the Firefly RK3399 and been using it for a while. Same Kernel, Firefly DTBs and DietPi rootfs. I can upload it in case you want to put on a different issue (like this) so people could test it.

@carlosedp 

By all means, please do. If you create a new ticket with the info/download links, we can flag is as Info/testing required.

I've sent an email to Firefly requesting a dev sample board.

Phil1988, 2018-08-14:
Can someone explain why this SOC has a lower CPU rating than the "disannounced" Odroid N1 as they both have the same CPU?

Fourdee, 2018-08-16:
@Phil1988 

> Can someone explain why this SOC has a lower CPU rating than the "disannounced" Odroid N1 as they both have the same CPU?

At a guess:
- Reduce heat for passive supplied sink
- Reduce power consumption to allow for 12V 2A rated PSU
- Onboard components not rated for the higher power usage of running at full rated clocks

I'd imagine its a heat related limitation with provided heatsink.

Fourdee, 2018-08-16:
@Phil1988 

Depends on the max CPU clock, N1 is faster in our tests:
https://github.com/Fourdee/DietPi/issues/1253#issuecomment-346881878

carlosedp, 2018-08-16:
Exactly, each board manufacturer(Odroid, Firefly, etc) configure it's clock speeds and voltages on the board DTSs that are compiled with the Kernel. You can in theory change this for the boards to "overclock". Check the kernel source files on `/arch/arm64/boot/dts/rockchip`.

sergiud, 2018-09-18:
Does the image support M.2 SSDs? DietPi does not seem detect my NVME SSD plugged into the NanoPC T4.

carlosedp, 2018-09-18:
In theory it should, at least the feature `CONFIG_BLK_DEV_NVME=y` to support it is enabled on the Kernel (I don't have a M.2 drive to test). Check on yours with:

```
sudo modprobe configs
zcat /proc/config.gz|grep NVME
```

The drive is supposed to be on `/dev/nvme*'. Does `lspci` shows it?

alex-savin, 2018-09-25:
@sergiud same here
SSD was not detected

Fourdee, 2018-09-28:
@alex-savin @sergiud @carlosedp 

Hi guys,

I've sent an email to FriendlyElec, with link to this ticket, hopefully they can provide some assistance.

Unfortunately, I also lack any .M2 drives, unable to test. However, its on my shopping list.

Fourdee, 2018-10-02:
Hi all,

Friendlyelec mentions that only NVME based SSD's are supported on the hardware.

This is confirmed (@MichaIng finally treated myself to one!!!)
[image]

MichaIng, 2018-10-02:
@Fourdee 
Great, I just updated the fs expansion script as I found that indeed the partitions on NVMe drives follow naming scheme:
`/dev/nvme[0-9]n[0-9]p[0-9]`

First integer might be controller number, second drive number and only third is partition number.

Could you confirm that in your case there is no partition table on the drive? We would need to fix drive manager then to handle this correctly.

rolandoman, 2018-10-02:
Has anyone succeeded in getting the two MIPI-CSI2 cameras working with DietPi on the NanoPC-T4? I have an application that requires stereo vision, and I don't want to move to Android. If anyone can point me in the right direction it would be much appreciated. I have played around with the vendor build from friendlyarm but their debian and ubuntu releases do not support the cameras. Thanks in advance!!

LieDanG, 2018-10-12:
I just received my board today, for updating the kernel and such we need to use the rockchip tools? Or as usual using the dietpi software?

Fourdee, 2018-10-12:
@LieDanG 

Our current image uses a modified kernel of FriendlyARM, provided and maintained by @carlosedp. It includes additions such as CIFS and NFS support.

If @carlosedp can provide an updated kernel, we will update the image. Kernel updates will require a fresh image write.

LieDanG, 2018-10-12:
I'll wait for the updated image, I tried flashing the current one but I only get a localhost login and can't get past of it, the friendlydesktop image boots fine so I'm not sure what's happening

Fourdee, 2018-10-15:
Unable to dump rootfs from device after update. Start is ```0x000B2000```, fails everytime, even on para dump ```0 - 2```
http://chinagadgetsreviews.blogspot.com/2014/09/download-androidtool-v21.html

So, need to PREP, then:
- Backup rootfs
- Compress to tar.gz
- Save to rootfs.img.iso

---

- PREP'd
- ```apt-get update; apt-get install rsync -y```
- 
```
FP_FILTER_INCLUDE_EXCLUDE=/mnt/exclude
cat << _EOF_ > $FP_FILTER_INCLUDE_EXCLUDE
#Global - Folders
- $FP_TARGET_BACKUP/
+ $G_FP_DIETPI_USERDATA_ACTUAL/
+ $G_FP_DIETPI_USERDATA/
- /mnt/*
- /media/
- /dev/
- /lost+found/
- /proc/
- /sys/
- /tmp/
- /run/

#Apt package cache (.deb)
- /var/cache/apt/archives/

#Manpages / Docs
- /usr/share/man/
- /usr/share/doc/

#Global - Files
- /var/swap
- .swap*
- $FP_LOG
- $FP_DIETPIBACKUP_SETTINGS
- $BACKUP_STATS_FILENAME
- fake-hwclock.data
- *.tmp

#MS Windows specific
- Thumbs.db
- desktop.ini
_EOF_

RSYNC_RUN_OPTIONS_BACKUP="-aH --info=name0 --info=progress2 --delete --delete-excluded --exclude-from=$FP_FILTER_INCLUDE_EXCLUDE"

mkdir -p /mnt/dietpi-backup

rsync $RSYNC_RUN_OPTIONS_BACKUP $RSYNC_LOGGING_OPTIONS / /mnt/dietpi-backup/
```

Fourdee, 2018-10-15:
@carlosedp 

Legend, thanks for updating the kernel for NanoPC-T4. I've updated the image and works a charm:
https://dietpi.com/downloads/images/DietPi_NanoPCT4-ARMv8-Stretch.7z

IvanCaregnato, 2018-10-15:
Hi guys, any chance to get Kodi available on it?

pshempel, 2018-11-26:
This does not provide a serial console to login, it does give output, but no input. It just does 
                                                                                                                                               } 
                                                                                                                                                         ] 
as an output once the login screen comes up.
Logging into the console, you get kicked out when logging in with root and dietpi
Also after looking at the source of this, who thought that putting into shadow the password for dietpi as plain text dietpi would work?
The issue with the login is not only on this hardware I have put this onto another odroid as well and serial console does the same thing. Not sure if anyone even tests that from what I can see.
So this is not a usable system at all as 
Not networking - this system does not come up with any network as it cannot find the network card.

Fourdee, 2018-12-02:
@pshempel 

Hi,

Serial console is disabled after initial boot up. You can override this by setting ```CONFIG_SERIAL_CONSOLE_ENABLE=1``` in ```/boot/dietpi.txt``` before 1st run of the device.

> Logging into the console, you get kicked out when logging in with root and dietpi

This does not occur in our local testing, possible a bad image write, insufficient PSU?

> Also after looking at the source of this, who thought that putting into shadow the password for dietpi as plain text dietpi would work?

```AUTO_SETUP_GLOBAL_PASSWORD=``` in ```dietpi.txt``` is an automation option. After 1st run setup, the plain password is removed, then stored with SHA encryption on rootfs.

> No networking - this system comes up with no network. It errors out during boot

I suspect a bad image write or failing PSU/SD card. 

For us to assist you further, please state:
- PSU (make,model,voltage,amp)
- SD card used

ramgopalkota, 2019-06-26:
In NanoPC-T4, I have connected XPG SX6000 Lite SSD (PCIe Based) SSD but lspci is empty and also I did not find any driver /dev/nvme. How to debug the NVMe based SSDs being detected or not ?
Is Samsung 970 EVO is only supported ?

Datasheet of XPG SX6000 Lite.. http://www.xpg.com/us/feature/597

MichaIng, 2019-06-26:
@ramgopalkota 
Hmm, not sure if M.2 Gen3x4 requires some special support, but I suggest this is all backwards compatible. I don't have much experience with neither M.2 nor NVMe in general.

However as a general debug I suggest you check:
```
dmesg
journalctl
```
- The first for kernel errors like power/voltage or anything else PCIe/NVMe related.
- The second for general boot log, e.g. detection of PCIe and block devices.

Ah and there seems to be a kernel module, perhaps on T4 this is not baked into the kernel but needs to be enabled:
```
modinfo nvme
lsmod
```

ramgopalkota, 2019-06-27:
Thanks for prompt reply.
I do see some voltage/power failed logs in boot log.
Is there anything I need to do in dts file ?? Or is it related to my power supply ? I am using 12V/4A supply.

root@DietPi:~# journalctl | grep -i pci
                                   PCI I/O : 0xffffffbffee00000 - 0xffffffbfffe00000   (    16 MB)
Jun 26 19:43:28 DietPi kernel: PCI/MSI: /interrupt-controller@fee00000/interrupt-controller@fee20000 domain created
Jun 26 19:43:28 DietPi kernel: PCI: CLS 0 bytes, default 64
Jun 26 19:43:28 DietPi kernel: phy phy-pcie-phy.5: Looking up phy-supply from device tree
Jun 26 19:43:28 DietPi kernel: phy phy-pcie-phy.5: Looking up phy-supply property in node /pcie-phy failed
Jun 26 19:43:28 DietPi kernel: rockchip-pcie f8000000.pcie: GPIO lookup for consumer ep
Jun 26 19:43:28 DietPi kernel: rockchip-pcie f8000000.pcie: using device tree for GPIO lookup
Jun 26 19:43:28 DietPi kernel: of_get_named_gpiod_flags: parsed 'ep-gpios' property of node '/pcie@f8000000[0]' - status (0)
Jun 26 19:43:28 DietPi kernel: rockchip-pcie f8000000.pcie: Looking up vpcie3v3-supply from device tree
Jun 26 19:43:28 DietPi kernel: rockchip-pcie f8000000.pcie: Looking up vpcie3v3-supply property in node /pcie@f8000000 failed
Jun 26 19:43:28 DietPi kernel: rockchip-pcie f8000000.pcie: no vpcie3v3 regulator found
Jun 26 19:43:28 DietPi kernel: rockchip-pcie f8000000.pcie: Looking up vpcie1v8-supply from device tree
Jun 26 19:43:28 DietPi kernel: rockchip-pcie f8000000.pcie: Looking up vpcie1v8-supply property in node /pcie@f8000000 failed
Jun 26 19:43:28 DietPi kernel: rockchip-pcie f8000000.pcie: no vpcie1v8 regulator found
Jun 26 19:43:28 DietPi kernel: rockchip-pcie f8000000.pcie: Looking up vpcie0v9-supply from device tree
Jun 26 19:43:28 DietPi kernel: rockchip-pcie f8000000.pcie: Looking up vpcie0v9-supply property in node /pcie@f8000000 failed
Jun 26 19:43:28 DietPi kernel: rockchip-pcie f8000000.pcie: no vpcie0v9 regulator found
Jun 26 19:43:28 DietPi kernel: rockchip-pcie f8000000.pcie: PCIe link training gen1 timeout!
Jun 26 19:43:28 DietPi kernel: rockchip-pcie: probe of f8000000.pcie failed with error -110
Jun 26 19:43:28 DietPi kernel: ehci-pci: EHCI PCI platform driver

"find /lib/modules -name *nvme*" returns nothing.
root@DietPi:~# zcat /proc/config.gz | grep NVME
CONFIG_BLK_DEV_NVME=y
CONFIG_NVMEM=y
root@DietPi:~#

ramgopalkota, 2019-06-27:
I dumped the DTS by using dts compiler (dts command)
dtc -I fs -O dts /sys/firmware/devicetree/base  and I am attaching the output.
I did not find vcc3v3-pcie-regulator ... Is there a new dts file which I need to use.

[dts-file.txt](https://github.com/MichaIng/DietPi/files/3332606/dts-file.txt)

[8 more replies]