MichaIng/DietPi #1832: Clementine crashes under V6.9 after upgrade from V6.8

lateboomer, 2018-06-09:
## Creating a bug report/issue:
I just update Dietpi at Allo Usbridge to V6.9 from V6.8. The Clementine player was previously stable under V6.8. After update, Clementine started to hang if click Pause or drag the track position on the slider while playing.

#### Required Information:
- DietPi version	| V6.9
- Distro version	| stretch and 9.4 
- Kernel version	| 3.10.38 
- SBC device		| Sparky SBC (armv7l)
- Power supply used	| 5 V 2 A for sparky, and 5V 1A for Usbridge board
- SDcard used		| built in eMMC

#### Additional Information (if applicable):
- Software title	| Clementine 1.3.1
- software installed in version V6.6
- Can this issue be replicated on a fresh installation of DietPi?
Yes. This issue can be replicated. I have installed lxde desktop on Usbridge, and use "startx /usr/bin/clementine" in systemd during startup to run Clementine.
- `dietpi-bugreport` ID
I failed to login to server to create ID.

#### Steps to reproduce:

1. Install latest V6.9 Dietpi on Allo Usbridge.
2. Install  small desktop lxde on Allo. Install ALSA and Xserver and Gmstreamer1.0 to Allo
3. Run 'startx /usr/bin/clementine'
4. Add an album to current playlist and start to play one track.
5. While playing click Pause or click forward position on track slider. Clementine will hang and cannot respond. I have installed Dietpi and Clementine on rpi model 2 B to try but these issue never happened.

#### Expected behaviour:

1. Clementine should be able to operate fully without hang.

#### Extra details:
- On previous version may be V6.6, same issue also occurred. Then I discovered that if I mount the disk with explicit vfat or exfat instead of default auto, the issue was solved. The current issue I play the music library from another server shared through samba. However, on V6.8, it was stable and no samba issue.

Fourdee, 2018-06-09:
@lateboomer 

Hi,

Thanks for the report 👍 

Unfortunately, clementine is not in our DietPi-Software library. 
As such, to assist you, we need to know how you installed clementine?

Was it via APT? If so, we run APT upgrade prior to DietPi-Update, the latest version of clementine may be unstable?

lateboomer, 2018-06-10:
I installed it at lower version using 'apt-get install' maybe V6.6 and used it continuously after each update. This is the sequence of steps I installed Clementine:
1) install ALSA from dietpi-software (for rpi, for Usbridge it is already preinstalled)
2) install Xserver from dietpi-software (for rpi, for Usbridge it is already preinstalled)
3) install LXDE desktop from from dietpi-software
4) install install gstreamer1.0-alsa and others (for rpi, for Usbridge it is already preinstalled)
5) install Clementine with 'apt-get install clementine'
6) For Usbridge, put 'startx /usr/bin/clementine' at custom.sh at systemd startup scripts.

lateboomer, 2018-06-10:
A very good and happy news. I took the heed from you . I went and ran 'apt-get update' and 'apt-get upgrade'. I rebooted Allo and then ran 'apt-get install --reinstall clementine'.

I am surprise the issue is solved. Now Clementine back to stable self. I can play, drag, pause switch playlist and it can respond without hang. I think the issue is solved and I am happy. Thank you very much for the advice.

closed 2018-06-13 by Fourdee