MichaIng/DietPi #1835: Software Request

symbios24, 2018-06-09:
Any change about putting this software to DIetPi? 

i did a request months ago but nothing happened.

I can help with instructions how to install all this all you have to do is put them in scripts in the software section,i will appreciate if you can add them

Openarena / The battle Wesnoth / AssaultCube Server / Minetest Server/ Simutrans / widelands / Boswars / Firefox Browser that can start at Boot like Chromium Browser /
Minecraft server (a simple server without web interface)
everything is tested and working.

Thanks

MichaIng, 2018-06-11:
@symbios24 
Thanks again for your request. As mentioned in the other issue, currently we need to concentrate on maintaining our existing software offer and do rarely add new titles. Man power and spare time is always a limiting factor and recently we had many deep core system changes that need further attention  and bug fixing.

I will keep this issue opened and here your old issue as reference: https://github.com/Fourdee/DietPi/issues/1469

symbios24, 2018-07-01:
if i knew how to i would send you everything will pull requests,

At least  you can put the easy one`s

Short instructions

Wesnoth (apt-get install Wesnoth)

Boswars (apt-get install boswars)

Widelands (apt-get install widelands)

simutrans (apt-get install simutrans) 

Openarena (apt-get install openarena libjpeg8)

wget http://www.berryterminal.com/dl/ioquake3_99.1.36-rpi01_armhf.deb
sudo dpkg -i ./ioquake3_99.1.36-rpi01_armhf.deb
(rpi-update) required for openarena
----------------------------------------

Thanks

symbios24, 2018-07-01:
if i knew how to i would send you everything will pull requests,

At least you can put the easy one`s

Short instructions

Wesnoth (apt-get install Wesnoth)

Boswars (apt-get install boswars)

Widelands (apt-get install widelands)

simutrans (apt-get install simutrans)

Openarena (apt-get install openarena libjpeg8)

wget http://www.berryterminal.com/dl/ioquake3_99.1.36-rpi01_armhf.deb 
sudo dpkg -i ./ioquake3_99.1.36-rpi01_armhf.deb 
(rpi-update) required for openarena

Thanks

closed 2018-07-09 by Fourdee