MichaIng/DietPi #1838: DietPi-Software | X2Go

DJCrashdummy, 2018-06-11:
from time to time i use [X2Go](http://x2go.org) instead of XoverSSH or VNC... and today i was surprised that X2Go isn't provided by DietPi-Software.

yes, it is not a really big issue because [`x2goclient` is in the debian-repos](https://packages.debian.org/stretch/x2goclient) and [the server](https://wiki.x2go.org/doku.php/doc:installation:x2goserver) can be installed via their repositories ([Debian](https://wiki.x2go.org/doku.php/wiki:repositories:debian)|[Raspbian](https://wiki.x2go.org/doku.php/wiki:repositories:raspbian))... but anyways IMHO it would ease the installation process (especially when you take care of its desktop-bindings and other additional packages).
i just wanted to mention this topic and bring some attention to it... no idea if its worth including it.

(btw: also [`x2goserver` finally made it at least into the testing-repo](https://packages.debian.org/buster/x2goserver). :tada:)

MichaIng, 2018-06-11:
@DJCrashdummy 
Thanks for your request. At the moment we are concentrating on stability, reliability and in case update of our existing software offers, thus it could take some time until we add it to the list. The more e.g. installation steps/bash code you can provide, the sooner we will be able integrate, just as a hint 😉.

I will keep it opened, let's see if there is more interest on X2Go.

Fourdee, 2018-12-26:
@MichaIng 

> Added to FeatHub, feel free to vote for it: https://feathub.com/Fourdee/DietPi/+7

Great idea, nice one 👍

MichaIng, 2026-10-26:
The X2Go server is in the Debian repo, too:
```sh
apt install x2goserver
```
Is there any relevant setup `dietpi-software` can do aside of installing the APT package, so that it is ready-to-run?