MichaIng/DietPi #1844: Pine64 | Wifi module not recognized

midwan, 2018-06-12:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `6.8`
- Distro version	| `stretch`
- Kernel version	| `3.10.105-bsp-1.2-ayufan-118`
- SBC device		| `Pine A64 (aarch64)`
- Power supply used	| Official RPI PSU 2.5A
- SDcard used		| Sandisk Ultra

#### Additional Information (if applicable):
- Software title	| ---
- Was the software title installed freshly or updated/migrated? ---
- Can this issue be replicated on a fresh installation of DietPi? `Yes`
- `dietpi-bugreport` ID - Not applicable

#### Steps to reproduce:

1. Install DietPi on a Pine64 with the official Wifi/Bluetooth module installed
2. Try to enable the Wifi module
3. The Wifi module is not found from `dietpi-config`

#### Expected behaviour:

- The Wifi module should be recognized and used as a network adapter

#### Actual behaviour:

- The wifi module is not found, so only Ethernet can be used for network connections

#### Extra details:

- ...

MichaIng, 2018-06-13:
@midwan 
Thanks for your report.

Could you paste outputs of:
```
ip l
ip r
ip a
ls -al /sys/class/net/
```

Possible that the WiFi device shows up under a not handled device/interface name (not `wlan[0-9]`). But if the above commands do not show it, I would assume a missing driver or something.

midwan, 2018-06-14:
`ip l`
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: dummy0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN mode DEFAULT group default
    link/ether 2e:62:1c:2b:4e:8d brd ff:ff:ff:ff:ff:ff
3: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether 0e:c5:e0:37:82:2e brd ff:ff:ff:ff:ff:ff
4: sit0: <NOARP> mtu 1480 qdisc noop state DOWN mode DEFAULT group default
    link/sit 0.0.0.0 brd 0.0.0.0
```

`ip r`
```
default via 192.168.1.1 dev eth0
192.168.1.0/24 dev eth0 proto kernel scope link src 192.168.1.238
```
`ip a`
```
default via 192.168.1.1 dev eth0
192.168.1.0/24 dev eth0 proto kernel scope link src 192.168.1.238
root@DietPi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: dummy0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default
    link/ether 2e:62:1c:2b:4e:8d brd ff:ff:ff:ff:ff:ff
3: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 0e:c5:e0:37:82:2e brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.238/24 brd 192.168.1.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::cc5:e0ff:fe37:822e/64 scope link
       valid_lft forever preferred_lft forever
4: sit0: <NOARP> mtu 1480 qdisc noop state DOWN group default
    link/sit 0.0.0.0 brd 0.0.0.0
```

`ls -al /sys/class/net/`
```
root@DietPi:~# ls -al /sys/class/net
total 0
drwxr-xr-x  2 root root 0 Jun 14 22:28 .
drwxr-xr-x 47 root root 0 Jun 14 22:28 ..
lrwxrwxrwx  1 root root 0 Jun 14 22:28 dummy0 -> ../../devices/virtual/net/dummy0
lrwxrwxrwx  1 root root 0 Jun 14 22:28 eth0 -> ../../devices/soc.0/1c30000.eth/net/eth0
lrwxrwxrwx  1 root root 0 Jun 14 22:28 lo -> ../../devices/virtual/net/lo
lrwxrwxrwx  1 root root 0 Jun 14 22:28 sit0 -> ../../devices/virtual/net/sit0
```

midwan, 2018-06-14:
Additionally, when trying to enable Wifi in Dietpi-config:
```
nl80211 not found.
[ SUB1 ] DietPi-Set_Hardware > wificreds (set)
[  OK  ] wificreds set | Completed
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[ SUB1 ] DietPi-Set_Hardware > wifimodules (enable)
[  OK  ] DietPi-Set_Hardware | Checking for pre-req APT packages: crda firmware-atheros firmware-brcm80211 iw rfkill wireless-tools wpasupplicant firmware-misc-nonfree
[ INFO ] DietPi-Set_Hardware | Pre-req APT packages are installed
[ INFO ] DietPi-Set_Hardware | Please wait, enabling WiFi Modules...
Created symlink /etc/systemd/system/multi-user.target.wants/wifi_disable_powersave.service → /etc/systemd/system/wifi_disable_powersave.service.
[  OK  ] wifimodules enable | Completed
[ INFO ] DietPi-Config | Dropping connections, please wait...
```

midwan, 2018-06-14:
Sorry for the barrage of messages:
After enabling Wifi in Dietpi-Config, the SSID is found and it seems to connect to it after manually entering the passphrase.

However, my problem was that after rebooting, the Wifi adapter was "lost" again.

I've just upgraded to DietPi v6.9 on the Pine64 today, but the problem remains I'm afraid.

Fourdee, 2018-06-16:
@midwan 

> After enabling Wifi in Dietpi-Config, the SSID is found and it seems to connect to it after manually entering the passphrase.
However, my problem was that after rebooting, the Wifi adapter was "lost" again.

Apologies for the delay, I don't have a PineA64 test system running yet. However, I believe the issue to be the module is missing from /etc/modules or /etc/modules.d.

When the WiFi is working, please paste results:
```
lsmod
```

Fourdee, 2018-08-08:
Should be resolved in v6.13. Assuming the WiFi addon cards always use ```8723bs``` chipset.

https://github.com/Fourdee/DietPi/issues/1995#issuecomment-410931618

If this still occurs after v6.13:
- Possible hardware issue with this WiFi addon card
- The chipset has changed from ```8723bs```, we need to see image of chip contained on the card.

closed 2018-08-08 by Fourdee

vcliment89, 2018-09-22:
I was having the same issue in 6.16. I resolved by removing /etc/modprobe.d/disable_wifi.conf. I was able to then see the card listed. I'm novice at linux but I think there's something wrong here [https://github.com/Fourdee/DietPi/blob/master/dietpi/func/dietpi-set_hardware#L1180](url) (this is how I was able to find this file

Fourdee, 2018-09-27:
@vcliment89 

> I resolved by removing /etc/modprobe.d/disable_wifi.conf. I was able to then see the card listed

Thanks for the info 👍 

Apologies for the delay.

The file ``` /etc/modprobe.d/disable_wifi.conf``` should be removed when you enable WiFi in ```dietpi-config```. Please can confirm that you did indeed enable WiFi via ```dietpi-config```, and, the file still remained?

midwan, 2018-12-31:
@Fourdee 
Sorry to resurect this one, but I just run across it again. My SDcard got corrupt and I had to re-install my Pine64 from scratch with the latest DietPi image.

Here are the symptoms I noticed:
- After the initial startup (and after the self-update function), when going into Dietpi-config, in the Adapter options the Wifi module is shown as "Enabled" but "Not found". Any attempts to use it fail.
- I tried to change only a few things in the Regional settings options, specifically: Timezone to Europe/Stockholm, Locale to EN_US (UTF-8) and Keyboard to "us".
- With no other changes made, I tried to apply my preferences, but DietPi-Config kept failing to do so. There was an error regarding a missing "header" in the translations (sorry, I didn't keep note of the exact message at the time).
- I tried re-entering Dietpi-Config at that point, and changed the Locale and Keyboard options back to their UK defaults. I only left the Timezone to Europe/Stockholm. Unfortunately, it still failed with the same error.
- However, I noticed that the Wifi module was now found, so I added my SSID details and set it to DHCP. Even though the settings were correct, it failed to connect.
- I decided to reboot the device at that point, to see what happens.
- After rebooting, we get back into Dietpi-config and I noticed that the Wifi module is now Enabled and Connected. Strange, isn't it?

midwan, 2018-12-31:
Update: the same error comes up if I manually try to run `apt update` on the console.