MichaIng/DietPi #1845: DietPi-Software | Ubooquity

midwan, 2018-06-12:
## Creating a feature request:

#### Is your feature request related to a problem? Please describe:

- It would be great to have Ubooquity as a software installation option in DietPi, to avoid having to set it up manually

#### Describe the solution you'd like:

- Add Ubooquity as an option in DietPi-Software. It will require Java to be installed as well.
- Documentation is here: http://vaemendis.net/ubooquity/

#### Describe alternatives you've considered:

- Installing it manually is possible as well, but it's a bit of a hassle to set it up to run automatically as a service. Advanced users might be ok, but not everyone is happy editing service files to make it  happen ;)

#### Additional context:

- ...

Fourdee, 2018-06-13:
@midwan 

Install Java:
```
dietpi-software install 8
```

Download Ubo....
```
mkdir -p $G_FP_DIETPI_USERDATA/ubooquity
wget https://dietpi.com/downloads/binaries/all/Ubooquity.jar -O $G_FP_DIETPI_USERDATA/ubooquity/Ubooquity.jar
chmod +x $G_FP_DIETPI_USERDATA/ubooquity/Ubooquity.jar
```

Run
```
$(which java) -jar $G_FP_DIETPI_USERDATA/ubooquity/Ubooquity.jar --headless --remoteadmin --adminport 2038 --libraryport 2039
```

Service
```
cat << _EOF_ > /etc/systemd/system/ubooquity.service
[Unit]
Description=Ubooquity DietPi
After=dietpi-boot.service network.target

[Service]
Type=simple
ExecStart=$(which java) -jar $G_FP_DIETPI_USERDATA/ubooquity/Ubooquity.jar --headless --remoteadmin --adminport 2038 --libraryport 2039
User=root

[Install]
WantedBy=multi-user.target
_EOF_
systemctl daemon-reload
systemctl enable ubooquity
systemctl start ubooquity
```

midwan, 2018-06-14:
@Fourdee 
Thanks, the steps above work flawlessly.
It would be awesome if we could have it as a menu option as well.

Fourdee, 2018-06-15:
Notes:
- index = 80
- online doc = https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=12969#p12969
- mkdir -p /mnt/dietpi_userdata/ebooks /mnt/dietpi_userdata/comics

Fourdee, 2018-06-15:
Completed.

closed 2018-06-15 by Fourdee