MichaIng/DietPi #1846: DietPi-Drive_Manager | Disallow moving dietpi_userdata to filesystem without permissions support

MichaIng, 2018-06-12:
## Creating a feature request:

#### Is your feature request related to a problem? Please describe:

- When moving `dietpi_userdata` to another drive via `dietpi-drive_manager`, it is possible to move it to a file system without user permissions support, which breaks several of our software installations.
- `dietpi-drive_manager` checks for `vfat`, which does not support permissions, throws a hint in case but allows to go on: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-drive_manager#L1118-L1134
- `dietpi-set_userdata` tests (again) permission support directly, but just preserves them on copy process in case, otherwise goes on without preserving permissions: https://github.com/Fourdee/DietPi/blob/master/dietpi/func/dietpi-set_userdata#L72-L79

#### Describe the solution you'd like:

- On the one hand, as `dietpi-set_userdata` is a standalone script, it should do all these checks by itself, thus those can be skipped within `dietpi-drive_manager`. Only if we wan't to have a cleaned menu there, we would need to check first, based on all file systems we support and then tell set_userdata to skip additional tests, as this was already done.
- On the other hand, we should definitely not allow to move userdata, if the check fails, as described above this breaks expected DietPi usage and leads to possible data loss when user is doing wrong steps as reaction.

#### Describe alternatives you've considered:

- It would be possible as well, if we wan't to allow user full flexibility, to give a clear warning, if the check fails, but then on all known file systems that do not support permissions, or after direct write/read test.

#### Additional context:

- This issue was already found before, but forgotten again. Now recovered it due to user report, although it may not be related: https://dietpi.com/phpbb/viewtopic.php?f=11&t=3888&p=12951#p12951

Fourdee, 2018-06-13:
[image]

Fourdee, 2018-06-13:
[image]

Fourdee, 2018-06-13:
Completed.

closed 2018-06-13 by Fourdee

MichaIng, 2018-06-13:
Wow you were active at night :smiley:.

Btw.: Did you see the GPLv2 hint on repo main page? :)

Fourdee, 2018-06-14:
@MichaIng 

> Btw.: Did you see the GPL-2.0 hint on repo main page? :)

Ah yeah, at the title bar 😄 
https://github.com/Fourdee/DietPi/blob/master/LICENSE Looking good :) nice one 👍