MichaIng/DietPi #1850: mariadb

magelan80, 2018-06-14:
#### Details:
- Date           | Thu 14 Jun 16:23:41 BST 2018
- BR sent?       | 0 : b0ed4076-246c-46fa-89d5-31d9231475ad
- DietPi version | v6.9
- Img creator    | 0
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Distro         | stretch (index=4)
- Command        | G_AGI: mariadb-server
- Exit code      | 100
- Software title | DietPi-Software

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional Information:

#### Additional logs:
```
Log file contents:
Preconfiguring packages ...
Selecting previously unselected package mariadb-server-10.1.
(Reading database ... 23789 files and directories currently installed.)
Preparing to unpack .../mariadb-server-10.1_10.1.23-9+deb9u1_armhf.deb ...
/var/lib/mysql: found previous version 10.1
Unpacking mariadb-server-10.1 (10.1.23-9+deb9u1) ...
Selecting previously unselected package mariadb-server.
Preparing to unpack .../mariadb-server_10.1.23-9+deb9u1_all.deb ...
Unpacking mariadb-server (10.1.23-9+deb9u1) ...
Processing triggers for systemd (232-25+deb9u2) ...
Setting up mariadb-server-10.1 (10.1.23-9+deb9u1) ...
dpkg: error processing package mariadb-server-10.1 (--configure):
 subprocess installed post-installation script returned error exit status 1
dpkg: dependency problems prevent configuration of mariadb-server:
 mariadb-server depends on mariadb-server-10.1 (>= 10.1.23-9+deb9u1); however:
  Package mariadb-server-10.1 is not configured yet.

dpkg: error processing package mariadb-server (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 mariadb-server-10.1
 mariadb-server
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2018-06-16:
@magelan80 
Thank you very much for your report.

Analysing bug report log:
```
Jun 14 21:43:49 DietPi mysqld[13213]: 2018-06-14 21:43:49 1988263936 [Note] /usr/sbin/mysqld (mysqld 10.1.23-MariaDB-9+deb9u1) starting as process 13213 ...
Jun 14 21:43:49 DietPi mysqld[13213]: 2018-06-14 21:43:49 1988263936 [Warning] Can't create test file /var/lib/mysql/DietPi.lower-test
Jun 14 21:43:49 DietPi mysqld[13213]: #007/usr/sbin/mysqld: Can't change dir to '/var/lib/mysql/' (Errcode: 13 "Permission denied")
Jun 14 21:43:49 DietPi mysqld[13213]: 2018-06-14 21:43:49 1988263936 [ERROR] Aborting
```
- DietPi-Software creates symlink from `/var/lib/mysql` to `/mnt/dietpi_userdata/mysql`, but changes owner of target to mysql user: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L3587
- @magelan80 could you paste output of:
```
ls -al /var/lib/mysql
ls -al /var/lib/mysql/
```
- Replicate on RPi needed, as I can't replicate on VM.

Most properly unrelated, but spamming log (Linux Infrared Remote Control):
```
Jun 14 21:21:44 DietPi lircd[490]: lircd-0.9.4c[490]: Error: Cannot glob /sys/class/rc/rc0/input[0-9]*/event[0-9]*
Jun 14 21:21:44 DietPi lircd-0.9.4c[490]: Error: Cannot glob /sys/class/rc/rc0/input[0-9]*/event[0-9]*
```

Another one unrelated:
`systemd-fstab-generator[2166]: Failed to create mount unit file /run/systemd/generator/var-log.mount, as it already exists. Duplicate entry in /etc/fstab?`

mismamente, 2018-06-17:
I don't know if related, but yesterday I upgraded from 6.8 to 6.9 and I've realized that mysql doesn't start, it complains about permission denied in /var/lib/mysql
After searching problems, logs etc, I've fixed it adding mysql user to group dietpi (in /etc/group), because that folder has writting permissions only for dietpi (both user and group)

MichaIng, 2018-06-17:
@mismamente 
From reviewing the installation and update code, I cannot find a reason, why permissions about database folder should have changed.
`/var/lib/mysql` should be a symlink to `/mnt/dietpi_userdata/mysql`, which should in any case be owned by mysql user.

Also in your case, could you paste the result of:
```
ls -al /var/lib/mysql
ls -al /var/lib/mysql/
```

mismamente, 2018-06-17:
Yes, it's a RPi.
The result of ls is:

root@dietpi:~# ls -la /var/lib/mysql
lrwxrwxrwx 1 dietpi dietpi 26 mar  4 11:50 /var/lib/mysql -> /mnt/dietpi_userdata/mysql
root@dietpi:~# ls -la /var/lib/mysql/
total 176200
drwxrwx---  5 mysql  mysql      4096 jun 17 22:15 .
drwxrwxr-x 12 dietpi dietpi     4096 jun 17 22:15 ..
-rwxrwx---  1 mysql  mysql     16384 jun 17 20:30 aria_log.00000002
-rwxrwx---  1 mysql  mysql        52 jun 17 20:30 aria_log_control
-rwxrwx---  1 mysql  mysql         0 mar  2 22:27 debian-10.1.flag
-rwxrwx---  1 mysql  mysql  79691776 jun 17 22:15 ibdata1
-rwxrwx---  1 mysql  mysql  50331648 jun 17 22:15 ib_logfile0
-rwxrwx---  1 mysql  mysql  50331648 may 22 20:30 ib_logfile1
-rwxrwx---  1 mysql  mysql         0 mar  2 22:28 multi-master.info
drwxrwx---  2 mysql  mysql      4096 mar  2 22:28 mysql
-rwxrwx---  1 mysql  mysql       280 jun 16 09:45 mysqld_multi.log
-rwxrwx---  1 mysql  mysql        15 mar  2 22:28 mysql_upgrade_info
drwxrwx---  2 mysql  mysql      4096 mar  4 21:09 owncloud
drwxrwx---  2 mysql  mysql      4096 mar  2 22:28 performance_schema
-rw-rw----  1 mysql  mysql     24576 jun 17 22:15 tc.log

Fourdee, 2018-06-18:
@magelan80 

- Power supply used on the system?
- Have you installed or configured any linux options/software, outside of DietPi programs?

Are these drives USB host powered, or, 3.5inch and contain their own PSU for each device?
```
/dev/sda1: LABEL="rootfs" UUID="f816b411-0a86-45d3-9b08-400ff5c316e0" TYPE="ext4" PARTUUID="132dd420-01"
/dev/sdb1: LABEL="Seagete" UUID="bed0ff5c-e1e2-49ae-a6ad-78bf3880bfba" TYPE="ext4" PARTUUID="3787d420-01"
```

MichaIng, 2018-06-28:
@Fourdee 
Looks very similar: https://github.com/Fourdee/DietPi/issues/1860
And there is another bug report with MariaDB installation issue.

If it really turns out that all of this is due to insufficient power to keep up the data drive on installations, or manually/accidentally unmounted drives, perhaps we could include some userdata drive availability check into `dietpi-software`?

Fourdee, 2018-06-29:
@MichaIng 

> Looks very similar: #1860
And there is another bug report with MariaDB installation issue.

If it really turns out that all of this is due to insufficient power to keep up the data drive on installations, or manually/accidentally unmounted drives, 

Yep, appears power/unplugged related. And/or, userdata not moved prior to install with unplugging drive.

> perhaps we could include some userdata drive availability check into dietpi-software?

```G_CHECK_USERDATA```? Checks ```/mnt/dietpi_userdata``` directory/symlink against where it should be (eg: mounted drives)?

Fourdee, 2018-06-29:
```
root@DietPi:~# G_CHECK_USERDATA
[  OK  ] DietPi-Userdata validation: /mnt/c976771f-1ec9-445b-949c-d6b2883e7860/dietpi_userdata
root@DietPi:~# umount /mnt/c976771f-1ec9-445b-949c-d6b2883e7860

root@DietPi:~# G_CHECK_USERDATA
[FAILED] DietPi-Userdata validation: /mnt/c976771f-1ec9-445b-949c-d6b2883e7860/dietpi_userdata. Drive is not mounted/available, unable to continue.
```

Fourdee, 2018-06-29:
@MichaIng 

G_CHECK_USERDATA: https://github.com/Fourdee/DietPi/commit/e70e3696ec6b2df15027320badad1c2769f0ff4c

Added, only checks symlink + external drive it mounted, we can always add to it if needed (eg: verify file can be written etc)?

Fourdee, 2018-06-29:
Marking as closed in favor of v6.10 added prevention: https://github.com/Fourdee/DietPi/issues/1850#issuecomment-401426132

MichaIng, 2018-07-03:
@Fourdee 
Next report that looks similar: https://dietpi.com/phpbb/viewtopic.php?f=11&t=4163

@magelan80 @tjblair
Might it be the case that you use external USB drives without own power supply and did not enable "Max USB current within "dietpi-config > Advanced Options"?

Fourdee, 2018-07-03:
@MichaIng 

> did not enable "Max USB current within "dietpi-config > Advanced Options"?

This is enabled by default, 1.2A max I believe on all devices:
https://github.com/Fourdee/DietPi/blob/master/config.txt#L54

So 2x 700mAh drives will exceed that value.

Fourdee, 2018-07-05:
@MichaIng 

> As we count not replicate on several attempts, I am out of ideas what could be an issue from DietPi code side and also the APT packages generally seem to work well.

Yep, I agree. I believe this is a hardware related issue, outside of our support.

As no response from user, and due to the above, I'll mark this as closed.

closed 2018-07-05 by Fourdee