MichaIng/DietPi #1852: Can't start Octoprint

SimoneBnc, 2018-06-15:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `6.9`
- Distro version	| `stretch`
- Kernel version	| `4.14.34-v7+`
- SBC device		| RPi3
- Power supply used	|  5V 2A 
- SDcard used		| (Kingston 16GB class10)

#### Additional Information (if applicable):
- Software title	octoprint
- Was the software title installed freshly or updated/migrated?

fresh install

- Can this issue be replicated on a fresh installation of DietPi?

Installed latest version and added Octoprint from the available software

- `dietpi-bugreport` ID bb44cf6e-b5eb-48d5-a8dc-2418890e0e65

#### Steps to reproduce:
1. install octoprint from fresh install
2. octoprint is not started on boot
3. tried to run manually with root@DietPi:/mnt/dietpi_userdata/octoprint# octoprint serve

#### Expected behaviour:

- the octoprint service should be running on port 5000 but service is not started

#### Actual behaviour:
- this is the log from the CLI

Traceback (most recent call last):
  File "/usr/local/bin/octoprint", line 11, in <module>
    load_entry_point('OctoPrint==1.3.8', 'console_scripts', 'octoprint')()
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.8-py2.7.egg/octoprint/__init__.py", line 565, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "/usr/local/lib/python2.7/dist-packages/click-6.2-py2.7.egg/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click-6.2-py2.7.egg/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click-6.2-py2.7.egg/click/core.py", line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click-6.2-py2.7.egg/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click-6.2-py2.7.egg/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click-6.2-py2.7.egg/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.8-py2.7.egg/octoprint/cli/server.py", line 183, in serve_command
    ignore_blacklist)
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.8-py2.7.egg/octoprint/cli/server.py", line 84, in run_server
    after_environment_detector=log_register_rollover)
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.8-py2.7.egg/octoprint/__init__.py", line 66, in init_platform
    logger, recorder = preinit_logging(debug, verbosity, uncaught_logger, uncaught_handler)
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.8-py2.7.egg/octoprint/__init__.py", line 171, in preinit_logging
    logger = set_logging_config(config, debug, verbosity, uncaught_logger, uncaught_handler)
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.8-py2.7.egg/octoprint/__init__.py", line 275, in set_logging_config
    import logging.config as logconfig
ValueError: bad marshal data (unknown type code)

#### Extra details:

- ...

Fourdee, 2018-06-15:
@SimoneBnc 

Thanks for the report, 👍 

Please try:
```
G_AGI python3
dietpi-services restart
dietpi-services status
```

Fourdee, 2018-06-15:
Notes:
Unable to replicate on fresh install on RPi 3 B+, ```python3``` is installed.

Fourdee, 2018-06-15:
@SimoneBnc 

Strange, can't see to find any differences in our installs, yet our local testing version works fine. 

Lets try reinstalling, see if that resolves.

Please run the following commands.
```
rm -R /usr/local/lib/python2.7/dist-packages/OctoPrint
dietpi-software uninstall 153
dietpi-software install 153

SimoneBnc, 2018-06-15:
I think I solved the errors by running 

find /usr -name '*.pyc' -delete

as suggested here -> https://stackoverflow.com/questions/30861493/how-to-fix-python-valueerrorbad-marshal-data  but I don't think it is the correct solution to use given the side effects

SimoneBnc, 2018-06-15:
I tried a fresh install and now it works... maybe the order of installation of 3-4 other sw broke it... If I have time I'll try some more times to see if the problem occurs again

Fourdee, 2018-06-18:
@SimoneBnc 

Excellent, great to see this issue is now resolved. As I am unable to replicate this, I'll mark as closed. However, please reopen if required.

closed 2018-06-18 by Fourdee