MichaIng/DietPi #1854: DietPi NativePC BIOS 32 Bit Version

symbios24, 2018-06-16:
Hi, any change to make a Native PC BIOS x86 32 bit DietPi version for old computers?

Thanks

symbios24, 2018-06-16:
Thanks for the response.
But if already you tested the Native PC BIOS x64 that could mean that the software that includes  will work theoretically only the os need`s 32 bit architecture/conversion etc.

symbios24, 2018-06-16:
Thanks for the response.
But if already you tested the Native PC BIOS x64 that could mean that the software that includes will work theoretically only the os need`s 32 bit architecture/conversion etc.

i can help by testing it if you want

symbios24, 2018-06-16:
Thanks for the response. i tested the image today and thought it was both 32/64 bit because of the name etc x88 x64
But if already you tested the Native PC BIOS x64 that could mean that the software that includes will work theoretically only the os need`s 32 bit architecture/conversion etc.

i can help with the testing it if you want

symbios24, 2018-06-16:
Thanks for the response. i tested the image today and thought it was both 32/64 bit because of the name etc x86 x64
But if already you tested the Native PC BIOS x64 that could mean that the software that includes will work theoretically only the os need`s 32 bit architecture/conversion etc.

i can help with the testing it if you want

symbios24, 2018-06-16:
Thanks for the response. i tested the image today and thought it was both 32/64 bit because of the name etc x86 x64
But if already you tested the Native PC BIOS x64 that could mean that the software that includes will work theoretically only the os need`s 32 bit architecture/conversion etc.
i can help with the testing it if you want

For example you can start by putting the 10 most popular software eg:
Plex Server / Transmission / deluge / hotspot / Tor / amiberry / kodi etc
and in time putting more if tested and working you dont have to put everything at once
just a suggestion

Fourdee, 2018-06-29:
#### I'll mark this as closed for the following reasons:
- 32bit x86 is being phased out, most PC's in the last 8+ years will have 64bit support.
- The sheer workload required to add support for 32bit x86, would take weeks/months of coding and testing. Time we could spend on better improving DietPi in more critical areas.

#### In short: We are not planning to provide a 32bit x86 Native PC image, or support at this time.

closed 2018-06-29 by Fourdee