MichaIng/DietPi #1860: #### Details: - Date           | Wed 20 Jun 20:16:40 CDT 2018 - BR sent?       | 0 : 621456d1-4a18-44eb-bcc6-6ab35ba0586d - DietPi version | v6.9 - Img creator    | 0 - Pre-image      | Raspbian Lite - SBC device     | RPi 3 Model B+ (armv7l) (index=3) - Distro         | stretch (index=4) - Command        | mkdir -p /mnt/dietpi_userdata/mysql - Exit code      | 1 - Software title | DietPi-Software  #### Expected behaviour: <!-- What SHOULD be happening? -->  #### Actual behaviour: <!-- What IS happening? -->  #### Steps to reproduce: <!-- Explain how to reproduce the issue -->  #### Additional Information: <!-- if applicable -->  #### Additional logs: ``` Log file contents: mkdir: cannot create directory ‘/mnt/dietpi_userdata’: File exists

tjblair, 2018-06-21:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `paste -sd '.' /DietPi/dietpi/.version`
- Distro version	| `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version	| `uname -r`
- SBC device		| `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used	| (EG: 5V 1A RAVpower)
- SDcard used		| (EG: SanDisk ultra)

#### Additional Information (if applicable):
- Software title	| (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
- `dietpi-bugreport` ID

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

MichaIng, 2018-06-22:
Possibly related? https://github.com/Fourdee/DietPi/issues/1850

Note to self:
- Test MariaDB installation with moved dietpi_userdata, thus doubled symlink redirection.

MichaIng, 2018-06-28:
@tjblair 
I can't replicate the issue or any explanation within the code.

It seems you moved your `dietpi_userdata`. Could you the target drive is/was successfully mounted? 

- `mkdir: cannot create directory ‘/mnt/dietpi_userdata’: File exists` means that `/mnt/dietpi_userdata` should be a symlink, which is the case, if userdata is moved to another location/drive.
- `mkdir -p /mnt/dietpi_userdata/mysql` creates `/mnt/dietpi_userdata` if not existent, but in case the symlink is active, it should not try this, at least it never does on my tests.
- Thus only explanation I can think of, is that the dietpi_userdata drive is unavailable.

Possible explanation could be insufficient power during heavy device usage (which is usually the case on dietpi-software installations) to keep up USB drives without dedicated power supply.
- Does your external drive have an own power supply or has it just USB connector to RPi (USB Sticks, 2,5" factor drives usually)?
- Which power supply do you use for the RPi?

Fourdee, 2018-06-29:
Marking as closed in favor of: https://github.com/Fourdee/DietPi/issues/1850#issuecomment-401426132

closed 2018-06-29 by Fourdee