MichaIng/DietPi #1861: Node Red installs but does not run

9H5G, 2018-06-21:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| 6.9
- Distro version	|Stretch
- Kernel version	| 4.14.34-v7+
- SBC device		| RPI 3B
- Power supply used	| 3A
- SDcard used		| SanDisk ultra

#### Additional Information (if applicable):
- Software title	| Node Red
- Was the software title installed freshly or updated/migrated? Fresh
- Can this issue be replicated on a fresh installation of DietPi? YES
- `dietpi-bugreport` ID

#### Steps to reproduce:

1. Install DietPI on fresh card.
2. Allow to boot and update.  Change to static IP, change hostname, Dietpi Userdata location, passwords, reboot
3. Install Mosquitto, NR, I2C support using installer
4. reboot
5. Node Red does not run

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[  OK  ] DietPi-Services | mosquitto	active (running) since Thu 2018-06-21 07:56:05 BST; 12min ago
[FAILED] DietPi-Services | node-red	failed (Result: exit-code) since Thu 2018-06-21 07:56:08 BST; 12min ago
● node-red.service - Node-Red
   Loaded: loaded (/etc/systemd/system/node-red.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2018-06-21 07:56:08 BST; 12min ago
  Process: 1003 ExecStart=/usr/local/bin/node-red -u /mnt/dietpi_userdata/node-red (code=exited, status=1/FAILURE)
 Main PID: 1003 (code=exited, status=1/FAILURE)

Jun 21 07:56:05 gozo systemd[1]: Started Node-Red.
Jun 21 07:56:08 gozo node-red[1003]: /usr/local/lib/node_modules/node-red/node_modules/fs-extra/lib/mkdirs/mkdirs-sync.js:49
Jun 21 07:56:08 gozo node-red[1003]:           throw err0
Jun 21 07:56:08 gozo node-red[1003]:           ^
Jun 21 07:56:08 gozo node-red[1003]: Error: EACCES: permission denied, mkdir '/mnt/dietpi_userdata/node-red'
Jun 21 07:56:08 gozo node-red[1003]:     at Object.mkdirSync (fs.js:738:3)
Jun 21 07:56:08 gozo node-red[1003]:     at mkdirsSync (/usr/local/lib/node_modules/node-red/node_modules/fs-extra/lib/mkdirs/mkdirs-sync.js:31:9)
Jun 21 07:56:08 gozo systemd[1]: node-red.service: Main process exited, code=exited, status=1/FAILURE
Jun 21 07:56:08 gozo systemd[1]: node-red.service: Unit entered failed state.
Jun 21 07:56:08 gozo systemd[1]: node-red.service: Failed with result 'exit-code'.
[  OK  ] DietPi-Services | cron	active (running) since Thu 2018-06-21 07:56:05 BST; 12min ago
[  OK  ] DietPi-Services | ssh	active (running) since Thu 2018-06-21 07:55:52 BST; 12min ago
[  OK  ] DietPi-Services | dietpi-boot	active (exited) since Thu 2018-06-21 07:56:04 BST; 12min ago
[  OK  ] DietPi-Services | dietpi-postboot	active (exited) since Thu 2018-06-21 07:56:04 BST; 12min ago

#### Expected behaviour:

- NR should be running

#### Actual behaviour:

- See log above

#### Extra details:
I have attempted this 3 times with the same result.

Fourdee, 2018-06-21:
Please try:
```
mkdir -p  $G_FP_DIETPI_USERDATA/node-red
chown -R nodered:nodered $G_FP_DIETPI_USERDATA/node-red
dietpi-services restart
```

9H5G, 2018-06-21:
Thank you.

Completed.  Owner now 109 Group 103.  However, NR still does not run under
service. Somehow mosquitto takes 2 restarts to start (consistent).

Kind Regards
John

Fourdee, 2018-06-22:
Unable to replicate on a fresh install:
```
[  OK  ] DietPi-Services | mosquitto    active (running) since Fri 2018-06-22 17:39:16 BST; 1min 33s ago
[  OK  ] DietPi-Services | node-red     active (running) since Fri 2018-06-22 17:39:16 BST; 1min 33s ago
[  OK  ] DietPi-Services | cron active (running) since Fri 2018-06-22 17:39:16 BST; 1min 33s ago
[  OK  ] DietPi-Services | ssh  active (running) since Thu 2018-06-21 08:17:04 BST; 1 day 9h ago
[  OK  ] DietPi-Services | dietpi-boot  active (exited) since Thu 2018-06-21 08:17:08 BST; 1 day 9h ago
[  OK  ] DietPi-Services | dietpi-postboot      active (exited) since Thu 2018-06-21 08:17:08 BST; 1 day 9h ago
```

Fourdee, 2018-06-22:
>Allow to boot and update. Change to static IP, change hostname, Dietpi Userdata location, passwords, reboot
Install Mosquitto, NR, I2C support using installer

Full test as per above:
TBA

Fourdee, 2018-06-22:
@9H5G 

> Dietpi Userdata location

Lets check this, please send us a bug report
```
dietpi-bugreport
```

Fourdee, 2018-06-22:
Unable to replicate on fresh install as per user installation instructions, ext4 userdata folder:
```
root@DietPi:~# dietpi-services status
[  OK  ] Root access verified.

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[  OK  ] DietPi-Services | mosquitto    active (running) since Fri 2018-06-22 18:01:17 BST; 4min 39s ago
[  OK  ] DietPi-Services | node-red     active (running) since Fri 2018-06-22 18:01:17 BST; 4min 39s ago
[  OK  ] DietPi-Services | cron active (running) since Fri 2018-06-22 18:01:17 BST; 4min 39s ago
[  OK  ] DietPi-Services | dropbear     active (running) since Fri 2018-06-22 18:01:01 BST; 4min 54s ago
[  OK  ] DietPi-Services | dietpi-boot  active (exited) since Fri 2018-06-22 18:01:08 BST; 4min 48s ago
[  OK  ] DietPi-Services | dietpi-postboot      active (exited) since Fri 2018-06-22 18:01:08 BST; 4min 48s ago
```

I suspect fat32 userdata, which is prevented in https://github.com/Fourdee/DietPi/issues/1846, need to verify ```dietpi-software```. Waiting for user bug report.

9H5G, 2018-06-22:
Sent - 8967f513-594a-4150-b577-6187cb8eb6db

Kind Regards
John

9H5G, 2018-06-22:
Weird.  Did you have a USB drive as your Dietpi_userdata?

Kind Regards
John

Fourdee, 2018-06-22:
@9H5G 

Thanks for the bug report 👍 

Confirmed ext4 FS.

Please can you paste results, lets check the permissions we set got applied (https://github.com/Fourdee/DietPi/issues/1861#issuecomment-399201081) :
```
ls -lha /mnt/dietpi_userdata/node-red
```

---

>   Did you have a USB drive as your Dietpi_userdata?

Yep:
```
root@DietPi:~# ls -lha /mnt/dietpi_userdata/node-red/
total 52K
drwxrwxr-x 3 nodered nodered 4.0K Jun 23 00:06 .
drwxrwxr-x 7 root    root    4.0K Jun 23 00:06 ..
-rw-r--r-- 1 nodered nodered  11K Jun 23 00:06 .config.json
-rw-r--r-- 1 nodered nodered  11K Jun 23 00:06 .config.json.backup
drwxr-xr-x 3 nodered nodered 4.0K Jun 23 00:06 lib
-rw-r--r-- 1 nodered nodered   99 Jun 23 00:06 package.json
-rw-r--r-- 1 nodered nodered  11K Jun 23 00:06 settings.js
root@DietPi:~# ls -lha /mnt/dietpi_userdata
lrwxrwxrwx 1 dietpi dietpi 57 Jun 23 00:03 /mnt/dietpi_userdata -> /mnt/195167f6-8ce3-476c-b334-35479ab592c2/dietpi_userdata
root@DietPi:~# blkid
/dev/mmcblk0p1: LABEL="boot" UUID="5DB0-971B" TYPE="vfat" PARTUUID="27504eef-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="060b57a8-62bd-4d48-a471-0d28466d1fbb" TYPE="ext4" PARTUUID="27504eef-02"
/dev/sda1: UUID="195167f6-8ce3-476c-b334-35479ab592c2" TYPE="ext4" PARTLABEL="primary" PARTUUID="a5e58917-03f8-45c6-b415-f938244c3859"
/dev/mmcblk0: PTUUID="27504eef" PTTYPE="dos"
```

9H5G, 2018-06-23:
> applied (#1861 (comment)

Here you go!

root@gozo:~# ls -lha /mnt/dietpi_userdata/node-red
total 128K
drwxrwxr-x  4 nodered nodered 4.0K Jun 22 21:06 .
drwxrwxr-x  8 root    root    4.0K Jun 21 20:47 ..
-rw-r--r--  1 nodered nodered  17K Jun 21 18:29 .config.json
-rw-r--r--  1 nodered nodered  17K Jun 21 17:56 .config.json.backup
drwxr-xr-x  3 nodered nodered 4.0K Jun 21 17:52 lib
drwxr-xr-x 68 nodered nodered 4.0K Jun 21 17:56 node_modules
-rw-r--r--  1 nodered nodered  265 Jun 21 17:56 package.json
-rw-r--r--  1 nodered nodered  24K Jun 21 17:56 package-lock.json
-rw-r--r--  1 nodered nodered  11K Jun 21 17:52 settings.js
-rw-r--r--  1 root    root    9.6K Jun 22 21:06 start
-rw-r--r--  1 root    root    9.6K Jun 21 20:57 .start.backup
-rw-r--r--  1 nodered nodered  120 Jun 21 18:05 start_cred
-rw-r--r--  1 nodered nodered   44 Jun 21 17:53 .start_cred.backup

Kind Regards
John

Fourdee, 2018-06-23:
@9H5G 

Thanks 👍 

```
-rw-r--r--  1 root    root    9.6K Jun 22 21:06 start
-rw-r--r--  1 root    root    9.6K Jun 21 20:57 .start.backup
```
Interesting

Any chance you have been using the folder to transfer data ```~/.node-red``` logged in as root?

The only thing I can suggest is trying to reinstall nodered on the system:
```
dietpi-software uninstall 122
dietpi-software install 122
```
Failing the above, we would need SSH access to debug/resolve, as, i'am unable to replicate the issue locally.

9H5G, 2018-06-23:
The dietpi_userdata folder on the USB drive was created under a previous
version (prior to the mandatory reinstall). Perhaps that is the problem.
BTW, the install overwrote the files in the NR directory on installation.
In previous versions it would just pick up the flows for that were there. I
can arrange SSH access - please contact me privately by email.

Kind Regards
John

Fourdee, 2018-06-24:
@9H5G 

> The dietpi_userdata folder on the USB drive was created under a previous
version (prior to the mandatory reinstall). Perhaps that is the problem.

Yep, pretty sure that is the cause.

When creating a new installation, you should always start with a fresh userdata folder (ideally, format the disk before use in ```dietpi-drive_manager```).

Fourdee, 2018-06-29:
Marking as closed due to: https://github.com/Fourdee/DietPi/issues/1861#issuecomment-399789460, please reopen if required.

closed 2018-06-29 by Fourdee