MichaIng/DietPi #1862: Dietpi Services does not produce required result

9H5G, 2018-06-21:
Required Information:
DietPi version	| 6.9
Distro version	|Stretch
Kernel version	| 4.14.34-v7+
SBC device	| RPI 3B
Power supply used	| 3A
SDcard used	| SanDisk ultra
Additional Information (if applicable):
Software title	| Node Red
Was the software title installed freshly or updated/migrated? Fresh
Can this issue be replicated on a fresh installation of DietPi? YES
dietpi-bugreport ID

Steps to reproduce:
Install DietPI on fresh card.
Allow to boot and update. Change to static IP, change hostname, Dietpi Userdata location, passwords, reboot
Install Mosquitto, NR, I2C support using installer
reboot
open HTOP - note mosquitto and cron are running (NR has failed)
use dietpi-services restart from another terminal

#### Expected behaviour:
mosquitto and cron stop and should then restart
- ...

#### Actual behaviour:
mosquitto and cron stop but do not restart
- ...

#### Extra details:
Discovered when trying to restart the (failed) NR service and determine its status, This behaviour may be because NR is failing but the mosquitto service starts before the NR and so this is unlikely.

Fourdee, 2018-06-21:
@9H5G 

Thanks for the report 👍 

Please paste results:
```
dietpi-services status
```

closed 2018-06-21 by Fourdee

9H5G, 2018-06-21:
root@gozo:~# dietpi-services status
[  OK  ] Root access verified.

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[  OK  ] DietPi-Services | avahi-daemon active (running) since Thu
2018-06-21 20:49:26 CEST; 3min 17s ago
[  OK  ] DietPi-Services | mosquitto active (running) since Thu 2018-06-21
20:49:26 CEST; 3min 17s ago
[FAILED] DietPi-Services | node-red failed (Result: exit-code) since Thu
2018-06-21 20:49:28 CEST; 3min 15s ago
● node-red.service - Node-Red
   Loaded: loaded (/etc/systemd/system/node-red.service; enabled; vendor
preset: enabled)
   Active: failed (Result: exit-code) since Thu 2018-06-21 20:49:28 CEST;
3min 16s ago
  Process: 1142 ExecStart=/usr/local/bin/node-red -u
/mnt/dietpi_userdata/node-red (code=exited, status=1/FAILURE)
 Main PID: 1142 (code=exited, status=1/FAILURE)

Jun 21 20:49:26 gozo systemd[1]: Started Node-Red.
Jun 21 20:49:28 gozo node-red[1142]:
/usr/local/lib/node_modules/node-red/node_modules/fs-extra/lib/mkdirs/mkdirs-sync.js:49
Jun 21 20:49:28 gozo systemd[1]: node-red.service: Main process exited,
code=exited, status=1/FAILURE
Jun 21 20:49:28 gozo systemd[1]: node-red.service: Unit entered failed
state.
Jun 21 20:49:28 gozo systemd[1]: node-red.service: Failed with result
'exit-code'.
[  OK  ] DietPi-Services | cron active (running) since Thu 2018-06-21
20:49:26 CEST; 3min 17s ago
[  OK  ] DietPi-Services | ssh active (running) since Thu 2018-06-21
20:49:14 CEST; 3min 29s ago
[  OK  ] DietPi-Services | dietpi-boot active (exited) since Thu 2018-06-21
20:49:23 CEST; 3min 21s ago
[  OK  ] DietPi-Services | dietpi-postboot active (exited) since Thu
2018-06-21 20:49:23 CEST; 3min 21s ago

Kind Regards
John