MichaIng/DietPi #1864: GitHub | Implement automated code check?

MichaIng, 2018-06-23:
As the "Checks" tab on PRs always pop into my eyes, I had a look into this code checker marketplace. For open source projects, everything I had a look at is for free.
E.g. `Codacy` supports shell and does code quality checks: https://github.com/marketplace/codacy

Of course as we are legend, we don't need it 😀, but could be still interesting what they find?

Fourdee, 2018-06-29:
@MichaIng 

I tried codacy for a while, but didn't like it if i'am honest. Great for error detection, but its mostly "minor recommends" which do not suit our project coding style. eg:

```
if (( $var )); then
= $ not required. Yes I know, but I prefer it to match [[ $var ]] style code.
```

Couldn't find a way to change the detection options either, so removed it.

MichaIng, 2018-06-29:
@Fourdee 
Looks like now this is possible:
- https://support.codacy.com/hc/en-us/articles/207994335-Code-Patterns
- https://support.codacy.com/hc/en-us/articles/208241349-Default-Patterns

MichaIng, 2019-05-06:
I mark this as closed for now. Will recheck probably sometime in the future.

closed 2019-05-06 by MichaIng