MichaIng/DietPi #1866: DietPi-Software | Roon Extension Manager

Fourdee, 2018-06-23:
Testing https://github.com/Fourdee/DietPi/pull/1865

Fourdee, 2018-06-29:
Unable to test, dev membership expired ☹️ 
[image]

Fourdee, 2018-06-29:
🈯️ Works a charm

@JanKoudijs Thank for adding this software into DietPi, we really appreciate your contribution!

As a small thank you, please email me daniel.knight@dietpi.com with your address, i'll send you some DietPi stickers.

Online doc: https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=13160#p13160

[image]

Fourdee, 2018-06-29:
Completed, available in v6.10 update.

closed 2018-06-29 by Fourdee

JanKoudijs, 2018-06-29:
Thanks @Fourdee Dan!

Now that went quick, just a week ago I downloaded the virtualbox image of DietPi to find out how a software package can be added, and now the Roon Extension Manager is ready for inclusion in v6.10 :+1: 

Proofs to me how well DietPi is set up!

DietPi is the first distribution / device that has the Extension Manager as an integrated part. Roon users that already use it for running the Roon Bridge now have the option to run extensions as well! I like it!

Thanks again.