MichaIng/DietPi #1868: Mosquitto installation problem - can't find executable

lupa18, 2018-06-25:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| 6.9
- Distro version	| stretch
- Kernel version	| 4.4.77-rockchip-ayufan-136
- SBC device		| Rock64 (aarch64)
- Power supply used	| 5V 3A
- SDcard used		| Class 10 micro SD

#### Additional Information (if applicable):
- Software title	| Mosquitto
- Was the software title installed freshly or updated/migrated? Installed Freshly
- Can this issue be replicated on a fresh installation of DietPi? Yes
- `dietpi-bugreport` f08067cc-5550-4254-968d-09db2bbf6111

#### Steps to reproduce:

1. Go to dietpi-software
2. Then choose Software Optimized
3. Mark Mosquitto (123)
4. Choose Install

#### Expected behavior:

Mosquito has to be installed, its executable must be in some place. Based on init.d script, must be at: /usr/sbin/mosquitto

#### Actual behaviour:

Executable can't be founded anywhere

MichaIng, 2018-06-25:
@lupa18 
Thank you very much for your report. I will review your bug report and try to replicate on x86_64 VM.

Fourdee, 2018-06-25:
Confirmed ARMv8 issue:
```
Errors were encountered while processing:
 mosquitto:armhf
[  OK  ] DietPi-Software | APT fix, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Correcting dependencies... Done
The following packages will be REMOVED:
  mosquitto:armhf
0 upgraded, 0 newly installed,
```

MichaIng, 2018-06-25:
@Fourdee 
Okay, hmm maybe we can find out which package is missing on ARMv8. Maybe some update of our deb package helps. v1.5 is available.

Fourdee, 2018-06-25:
Notes:
- Default compile missing services.
- init.d used previously, lets make a systemD version ```/usr/sbin/mosquitto -c /etc/mosquitto/mosquitto.conf```

MichaIng, 2018-06-25:
@Fourdee 
> Custom debian repo lacks ARMv8

Might be the reason why we added armhf arch to arm64 devices within current install script?

Fourdee, 2018-06-25:
Ok:
- All devices will use official Debian repo install for all devices (1.4 currently)
- If we get complaints for this change (eg: I need 1.5 for features), we'll re-investigate ARMv7 and AMD64 using http://repo.mosquitto.org/debian/dists/stretch/main/ (1.5)

Fourdee, 2018-06-25:
@lupa18 

Workaround (nice and simple), to install mosq:
```
G_AGI mosquitto
dietpi-services dietpi_controlled
dietpi-services restart
```