MichaIng/DietPi #1869: Backup not possible with an rsync daemon running

sergiud, 2018-06-25:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| 6.9
- Distro version	| 9.4
- Kernel version	| 4.14.32+
- SBC device		| Oroid XU4Q with CloudShell2
- Power supply used	| 15V 4A
- SDcard used		| eMMC

#### Additional Information (if applicable):
- Running OpenMediaVault with rsync server

#### Steps to reproduce:

1. Start rsync daemon
2. Start DietPi backup

#### Expected behaviour:

- Backup start

#### Actual behaviour:

- Backup does not start because rsync is already running

#### Extra details:

- Rsync daemon must be stopped prior to starting a backup

MichaIng, 2018-06-25:
@sergiud
Thanks very much for your report.

I didn't ever work with rsync in daemon mode, so don't know what's possible. Would it be possible to use the rsync daemon for the backup, if it's already running? I am just thinking, that just killing the daemon could interrupt running remote file transfers, or the daemon denies to stop, e.g. if configured as automated cron jobs etc. Killing it forcefully on the other hand could lead to partly transferred/corrupted files?

sergiud, 2018-06-25:
The rsync daemon serves only specific paths, so it don't believe it can be used for backup without restarting it. Stopping the dameon, on the other hand, will not cause any major problems.

Fourdee, 2018-06-25:
@MichaIng 

If the rsync daemon has a fixed service we could add to ```dietpi-services```, else, ```killall -w rsync``` ? 
However, probably best to ```killall``` in case user manually ran via cmdline.

Fourdee, 2018-06-25:
Note, error message should already be provided
```
			#Error: Rsync is already running
			elif (( $(ps aux | grep -ci -m1 "[r]sync") )); then

				Error_Rsync_Already_Running

```

sergiud, 2018-06-25:
The rsync daemon can be controlled using the usual systemd mechanism. It would be nice if the daemon is restarted once the backup is complete.

Fourdee, 2018-06-25:
@sergiud 

> The rsync daemon can be controlled using the usual systemd mechanism. It would be nice if the daemon is restarted once the backup is complete.

Service name?

MichaIng, 2018-06-25:
@Fourdee @sergiud 
Would it be possible to add dietpi-backup/sync location to the list (if it's a list), possibly with limited access for local transfer or something?

Otherwise YESNO sounds good. However what happens to running rsync transfers? Do they go on if/since rsync daemon is just responsible to establish connection and initiate transfer to target location? But `killall` should kill all child processes as well... In case there is no way to preserve running transfer, default should be (if G_USER_INPUT=0, cron job etc.) to skip `dietpi-backup/sync`, to not risk any file corruption or unexpected cancellations.

Fourdee, 2018-06-25:
@MichaIng 

> Would it be possible to add dietpi-backup/sync location to the list (if it's a list), possibly with limited access for local transfer or something?

Risky for ```dietpi-backup``` as we need to stop services prior to backup, else, inuse files etc.

Lets try it with the service control 1st, stop and start, fine if not set in config:
```
root@DietPi:~# systemctl start rsync
root@DietPi:~# systemctl status rsync -l
● rsync.service - fast remote file copy program daemon
   Loaded: loaded (/lib/systemd/system/rsync.service; enabled; vendor preset: en
abled)
   Active: inactive (dead)
Condition: start condition failed at Mon 2018-06-25 13:00:33 BST; 4s
ago
           └─ ConditionPathExists=/etc/rsyncd.conf was not met
```

Fourdee, 2018-06-25:
@sergiud 

Please patch the following files and retry ```dietpi-backup```
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/df2ef464c8f510acad8b00af6e1b33ff6a82bcd4/dietpi/dietpi-backup -O /DietPi/dietpi/dietpi-backup
wget https://raw.githubusercontent.com/Fourdee/DietPi/df2ef464c8f510acad8b00af6e1b33ff6a82bcd4/dietpi/dietpi-services -O /DietPi/dietpi/dietpi-services
```

Fourdee, 2018-06-25:
@MichaIng 

> €: @Fourdee /lib/systemd/system/rsync.service runs, if config file exists rsync --daemon. So service and process names are both rsync. But as you said, if user runs rsync --daemon manually, systemd can't control it of course.

Yep, i'll add a ```killall -w``` just incase.

I'll leave ```G_TERMINATE_PROCESS ``` in the mind bank, this is the only thing I can think of that would use it, and, we should reserve ```G_``` for items that are commonly used.

Fourdee, 2018-06-25:
@sergiud 

Many thanks for verifying working 👍 

Completed. Please reopen if required.

closed 2018-06-25 by Fourdee

Fourdee, 2018-06-28:
@MichaIng 

Yep, agree, needs cleaning up 👍