MichaIng/DietPi #1871: DietPi-Software | New version of UrBackup server available (2.2.1.1)

LexiconCode, 2018-06-25:
There is a new version of UrBackup server available (2.2.1.1). Download it [here](http://www.urbackup.org/download.html).

- DietPi Version 6.9
- SBC Device (EG: C2)

Fourdee, 2018-06-26:
@LexiconCode 

Thanks, we'll update it for v6.10 👍

MichaIng, 2018-06-26:
@LexiconCode @Fourdee 
Hmm, we actually updated `dietpi-software` to install the new version with v6.9: 
- https://github.com/Fourdee/DietPi/pull/1774
- https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L4246-L4259

But we usually do not upgrade software titles on `dietpi-update`, because the installation scripts often overwrite config files, also for UrBackup: https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L4282-L4285
So you could do `dietpi-software reinstall 111` to update to current version, but be sure to backup config files and data before.

LexiconCode, 2018-06-26:
@MichaIng 
Whoops I miss that. I will back up the configuration.

This seems like a straightforward enhancement to dietpi. 
For the software that requires a reinstall to update could we backup then restore configurations after install?

Fourdee, 2018-06-27:
> For the software that requires a reinstall to update could dietpi backup then restore configurations for dietpi-software reinstall xxx?

Not a bad idea 👍 

I'll add a prompt to ```reinstall``` for a ```dietpi-backup 1```.

Fourdee, 2018-06-27:
```G_BACKUP```
[image]

Fourdee, 2018-06-27:
Completed.

Fourdee, 2018-06-27:
Actually, we need to exclude this from patching reinstalls, add the prompt prior to ```dietpi-update```

Fourdee, 2018-06-27:
Completed.

closed 2018-06-27 by Fourdee

LexiconCode, 2018-06-27:
Real quick is that a complete dietpi backup or just the configuration files specific to the software that are restored on top of the new reinstall?

Fourdee, 2018-06-27:
@LexiconCode 

> Real quick is that a complete dietpi backup or just the configuration files specific to the software that are restored on top of the new reinstall?

Complete backup. Its the only sure way we can ensure all configuration files/settings are backed up.