MichaIng/DietPi #1873: DietPi-Software: Connection test Exit code 4

Tealk, 2018-06-26:
Hi,

i have fresh install DietPi on my RespberryPi and since first run i get this message:
[image]

I think this is a persistant problem, i hope someone can help me out

MichaIng, 2018-06-26:
@Tealk 
Thanks for your report.

Hmm, here it works:
```
2018-06-26 19:48:38 root@micha:/var/log# wget --spider http://raspbian.raspberrypi.org/raspbian
Spider mode enabled. Check if remote file exists.
--2018-06-26 19:48:42--  http://raspbian.raspberrypi.org/raspbian
Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... 93.93.128.193, 2a00:1098:0:80:1000:75:0:3
Connecting to raspbian.raspberrypi.org (raspbian.raspberrypi.org)|93.93.128.193|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: http://raspbian.raspberrypi.org/raspbian/ [following]
Spider mode enabled. Check if remote file exists.
--2018-06-26 19:48:42--  http://raspbian.raspberrypi.org/raspbian/
Connecting to raspbian.raspberrypi.org (raspbian.raspberrypi.org)|93.93.128.193|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Remote file exists and could contain further links,
but recursion is disabled -- not retrieving.
```

Also your error massage indicates, that your name resolver/DNS entry seems to not work, thus you should be not able to connect to any web resource, besides entering the plain IP.
To verify, `wget --spider http://google.com` should throw the same error message to terminal.

How did you configure your network connection? Did you adjust something inside `dietpi.txt`?
If not, then DHCP is default and the DHCP server should as well serve a DNS server.
Check: `cat /etc/resolv.conf`

If there is no (valid) entry, then try to set a static IP with `dietpi-config` > `Network Options / Adapters` and set e.g. google DNS server as static DNS entry: `8.8.8.8 8.8.4.4`

Tealk, 2018-06-26:
@MichaIng 
Thank you for the answer.

I have configure nothing, only install the image and connect the raspberrypi connected to power and internet

Here the wget test:
```
root@DietPi:~# wget --spider http://raspbian.raspberrypi.org/raspbian
Spider mode enabled. Check if remote file exists.
--2018-06-26 19:08:33--  http://raspbian.raspberrypi.org/raspbian
Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... failed: Temporary failure in name resolution.
wget: unable to resolve host address ‘raspbian.raspberrypi.org’
```
```
root@DietPi:~# wget --spider http://google.com
Spider mode enabled. Check if remote file exists.
--2018-06-26 19:09:48--  http://google.com/
Resolving google.com (google.com)... 216.58.213.206, 2a00:1450:4005:803::200e
Connecting to google.com (google.com)|216.58.213.206|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: http://www.google.com/ [following]
Spider mode enabled. Check if remote file exists.
--2018-06-26 19:09:53--  http://www.google.com/
Resolving www.google.com (www.google.com)... 216.58.213.196, 2a00:1450:4005:803::2004
Connecting to www.google.com (www.google.com)|216.58.213.196|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Remote file exists and could contain further links,
but recursion is disabled -- not retrieving.
```

Sorry atm i have no card reader here, so i cant check the files, or is it possible to enable sftp?

Tealk, 2018-06-26:
i have now set up the dns server static on google and now it run fine :/

Tealk, 2018-06-26:
i don't understand why the pc could resolve the domain but RPi just couldn't

Fourdee, 2018-06-27:
@Tealk 

> I use DHCP my dns server i have set up in the fritzbox 213.73.91.35 and 89.233.43.71
My tower can resolve http://raspbian.raspberrypi.org/raspbian

Could be any of the following:
- ```http://raspbian.raspberrypi.org/raspbian``` temporarily offline (common)
- Configuration issue on the router
- incompatibility with RPi
- bug with router (check firmware updates)

> i have now set up the dns server static on google and now it run fine :/

As this issue is now resolved, and, we are unable to replicate it our end, i'll mark it as closed. However, please free to continue discussion as needed.

closed 2018-06-27 by Fourdee