MichaIng/DietPi #1875: Nukkit Site Offline/Installation Broken

symbios24, 2018-06-26:
Hello upon Installation of Nukkit - Server for Minecraft Pocket Edition on a Raspberry pi 3 i got an error saying: that remote file does not exist -- broken link, referring to the file: nukkit-1.0-SNAPSHOT.jar 

By going manual from a browser to this site that dietpi tries to connect to get the file i got this message in Chinese i have translate it into english: 

Site:  http://ci.mengcraft.com:8080/job/nukkit/  

Message: Reminder: The website cannot be accessed temporarily because it has not been filed or has access to illegal content. Please contact the access provider.

looks like the Nukkit webpage is down for good i have found an alternative link/page that is working if you want to put it to the nukkit installer script

https://ci.potestas.xyz/job/NukkitX/job/master/lastSuccessfulBuild/artifact/target/nukkit-1.0-SNAPSHOT.jar

how can i change the script for now to make nukkit install successfully for now?

Thank you

MichaIng, 2018-06-27:
@symbios24 
Thanks for your report. Good find and solution 😃!

I was also checking the Nukkit GitHub repo (https://github.com/Nukkit/Nukkit), which leads to this again different source: https://ci.nukkitx.com/job/NukkitX/job/master/lastSuccessfulBuild/artifact/target/nukkit-1.0-SNAPSHOT.jar
- Ah, your link actually redirects to this as well. So I guess we should use the above directly?

Fix for you:
`sed -i "\|ci.mengcraft.com|c\			INSTALL_URL_ADDRESS='https://ci.nukkitx.com/job/NukkitX/job/master/lastSuccessfulBuild/artifact/target/nukkit-1.0-SNAPSHOT.jar'" /DietPi/dietpi/dietpi-software`

I already added the fix for v6.10, many thanks again!: 
https://github.com/Fourdee/DietPi/commit/319c5b7f66648ebfe21ca66e3d75ea5d48ee5f5e
https://github.com/Fourdee/DietPi/commit/07e56c60e6117a0dae76566c0bac71da26764282

MichaIng, 2018-06-27:
@symbios24 
Okay, this is a separate issue, I just checked our script and official instructions: https://github.com/hexparrot/mineos-node
- Replication test needed. I can do that on x86_64 VM later.
Possible issue:
- Node version, which often is the issue. Official doc expacts Node 8

Further notes:
- We use supervisor for startup. Why not using systemd which is also offered: https://github.com/hexparrot/mineos-node/blob/master/init/systemd_conf
- Hehe here we already use `mineos` as unprivileged user 😄.

MichaIng, 2018-06-28:
@symbios24 
I opened a new issue about MineOS. Indeed the installation is broken as well: https://github.com/Fourdee/DietPi/issues/1880

I will close this issue, as your initial bug was fixed. Lets move to the other topic to get the 2nd fixes.

closed 2018-06-28 by MichaIng