MichaIng/DietPi #1877: Security risk. Applications run as root user.

userdeveloper98, 2018-06-27:
## Creating a bug report/issue:
Hi there !
I have noticed that most of the apps installed by dietpi scripts are ruining as root user which is not recommended of sure.
I am happy to support migration of all applications to run under non-root user.
I did it already for almost all my software pack but it will be nice if we have this out of the box because most of the users may not be aware of this.
I will start with Gogs as it was the easiest to migrate application. 

#### Required Information:
- DietPi version	| 6.9
- Distro version	| 9.4
- Kernel version	| 4.14.34-v7+
- SBC device		| RPi3
- Power supply used	| 5V 2.1A RAVpower
- SDcard used		| SanDisk

#### Additional Information (if applicable):
- Software title	| Gogs
- Was the software title installed freshly or updated/migrated? 
Yes
- Can this issue be replicated on a fresh installation of DietPi? 
Yes

#### Steps to reproduce:
1. Install Gogs from dietpi-software as usually. 

#### Expected behaviour:
Gogs should run under non root user.
ex: **gogs** user. 

1. create user gogs (with home folder, no password, no ssh allowed, with group "applications")
2. chmod -R gogs:applications /etc/gogs
3. chmod -R gogs:applications /home/gogs
4. chmod -R gogs:applications /var/log/gogs (create folder if not exist)
5. chmod -R gogs:applications /mnt/dietpi_userdata/gogs-repo
6. edit file /etc/gogs/start.sh (more elegant solution will be probably to edit systemd file)
```
sudo -u gogs /etc/gogs/gogs web &> /var/log/gogs/gogs_daemon.log &
exit
```

#### Actual behaviour:
Security issue.
Gogs run under root user.
Exploiting Gogs vulnerability allow full root access to server.

#### Extra details:
This is the simplest app to migrate I would like to be added as reviewer to PR. (this will facilitate learning the source code :) )
After this I believe I will be able to create my own PR-s and support migration of other apps.
Including but not limited to:
- docker
- proftpd
- lighttpd
- minidlna
- deluged + deluge-web
- subsonic
- Radarr
- Jackett 

Thanks !

MichaIng, 2018-06-27:
@userdeveloper98 
Thanks very much for your hint. Old topic, but not unimportant from my point of view, thus good that someone pushes it. Your help in this of course is highly appreciated.

The work should not be too much to directly apply this to more/all affected software titles. The higher effort will be to test those software titles, their internal e.g. update functions etc. Best is to check their official installation instructions and see if anyway an own user is recommended or there is some clear reason for using root.

- I would just skip all APT repo installed software, including Lighttpd. There are good reasons if/why Debian/maintainer chooses the binary to start as root. For webservers this is common practise and needed to allow the server bind to SSL port, access to SSL certificates etc., AFAIK. The webservers by default anyway spawn child processes running as www-data user, which are the ones, handling web access. Just starting Lighttpd itself as non-root user will most likely cause issues, break features, messes with what the package maintainer/package update script is expecting.
- I guess you suggest `applications` group to allow another non-root user access, by adding it as well to this group? If we want to do that, I would vote for using e.g. `gogs:gogs` respectively, to really limit access to the application itself and root/sudo. If for sure cross-access between applications is needed (`www-data` access `redis` server to use for own-/Nextcloud transactional file locking), than just add the accessing application to the servers group directly: `usermod -a -G redis www-data`. Or do you have another reason for `applications` group?
- I guess for most of those application users we could just skip home directory completely?
- An alternative would be to use the `dietpi` user for all applications. But this only makes sense if it is removed from sudoers, which then might break other tasks of `dietpi` I guess, as the plan is to use this for more and more DietPi scripts, instead of root.

___________________________
A bid off-topic, but as we already have this users topic:
- I am generally not too sure about the benefit of using non-root user for terminal access that then is added to `sudoers` without sudo password requirement.
- If someone get's access to this non-root user, he can simply use `sudo` to have full root access anyway.
- So why not just directly use root user (for terminal access), skip the need to deal with 2 passwords, better assure full security for root with strong PW, SSH key authentication etc.
- The only benefit I can think of, is to prevent the system from end user accidents. Without root access the user simply cannot really damage the system that easily. But this is more the thinking of Apple and MS and I am more on the side to giving users always full possibilities, including full responsibility.

userdeveloper98, 2018-06-27:
> I would just skip all APT repo installed software, including Lighttpd.

Totally agree.

> I guess you suggest applications group to allow another non-root user access, by adding it as well to this group?

Yes that was the idea. locally I made it as `gogs:gogs`. For wider changes I was thinking to group more strategically, but this need to be configured individually as you mentioned.

> I guess for most of those application users we could just skip home directory completely?

Sure, but with gogs I didn't managed to get rid of this even after extensive playing with gogs configurations :(

> An alternative would be to use the dietpi user for all applications.

I would agree to leave everything being installed as `root `or `dietpi ` user because installation time is limited, don't see any major security issues as long as you trust the script author :)

--------------------------------------------------------------------------------------------

Using non-root users is a good practice if an account is compromised or misused, the affect will be isolated.
Ideally application users (or 'system users') don't have login capabilities. 
Ex: `messagebus:x:105:109::/var/run/dbus:/bin/false`. 
after doing `su messagebus` nothing will happen.
Or: `sshd:x:107:65534::/var/run/sshd:/usr/sbin/nologin` 
try `su sshd`  you will get: `This account is currently not available.`

One of major concerns that I have related to application running as root user is that if application can be exploited via some security vulnerabilities it potentially can get control to user under it is running.
1. **Gogs run under root.** Googs get exploited and get root account which can do virtually anything on your system.
2. **Gogs run under limited gogs user**. Googs get exploited  but the impact will be minimal, it probably will get access to gogs files and repositories but it will not be able to do any major changes in system. Ex: read passwd file or remove all backups etc.. 
Taking into consideration that user don't even have access to shell, the chances that it will break something will be dramatically low.

Strong passwords will not secure you from buffer overflows in applications and remote code execution because it will bypass the authentication and may get direct access to user under it is running.

Also taking into consideration that we are installing software that are maintained sometimes by single person, it may be that they will don't care or don't have time to fix security issues, or security flaws will be misused for a long time before public announcements.

There are lot of others reasons, hope that's enough.

BTW. 
I am planing to add fail2ban rules (will share once I test it) for applications so it will be impossible to brute-force Gogs account for instance. 
But as I can see https://github.com/gogs/gogs/issues/2384 they don't have yet implemented proper logs for that, let's see if we have some workarounds.

Fourdee, 2018-06-27:
@MichaIng 

Agree 👍 

@userdeveloper98 

Commit for gogs to change to its own user, also cleaned up the service: https://github.com/Fourdee/DietPi/commit/e915354689d2c4953a459761d4e496b39d127dce
**EDIT**: Yes it needs a home dir, resolved with latest commit
```
root@DietPi:~# systemctl status gogs
● gogs.service - DietPi Gogs service
   Loaded: loaded (/etc/systemd/system/gogs.service; disabled; vendor preset: en
abled)
   Active: active (running) since Wed 2018-06-27 14:45:45 BST; 8s ago
 Main PID: 8224 (bash)
   CGroup: /system.slice/gogs.service
           ├─8224 /bin/bash -c /etc/gogs/gogs web &> /var/log/gogs_daemon.log
           └─8226 /etc/gogs/gogs web

Jun 27 14:45:45 DietPi systemd[1]: Started DietPi Gogs service.
root@DietPi:~# cat /var/log/gogs_daemon.log
2018/06/27 14:45:45 [ WARN] Custom config '/etc/gogs/custom/conf/app.ini' not found, ignore this if you're running first time
2018/06/27 14:45:45 [TRACE] Custom path: /etc/gogs/custom
2018/06/27 14:45:45 [TRACE] Log path: /etc/gogs/log
2018/06/27 14:45:45 [TRACE] Log Mode: Console (Trace)
2018/06/27 14:45:45 [ INFO] Gogs: Go Git Service 0.9.141.0211
2018/06/27 14:45:45 [ INFO] Cache Service Enabled
2018/06/27 14:45:45 [ INFO] Session Service Enabled
2018/06/27 14:45:45 [ INFO] SQLite3 Supported
2018/06/27 14:45:45 [ INFO] Run Mode: Development
2018/06/27 14:45:47 [ INFO] Listen: http://0.0.0.0:3000
```

> Gogs run under root. Googs get exploited and get root account which can do virtually anything on your system.
Gogs run under limited gogs user. Googs get exploited but the impact will be minimal, it probably will get access to gogs files and repositories but it will not be able to do any major changes in system. Ex: read passwd file or remove all backups etc..
Taking into consideration that user don't even have access to shell, the chances that it will break something will be dramatically low.

Some good points, although personally, I believe we shouldn't have to live in a world where we don't grant any trust in the program.
Either way, no harm in limiting permissions.

MichaIng, 2018-06-27:
@Fourdee 
Nice, that was quick 😃.

@userdeveloper98 
Just to clarify my argumentation:
- For applications it definitely makes sense to use limited user, as long as it is simply not possible without root permissions (SSL bind, ...). Besides the arguments you mentioned, also ext4 reserved blocks are reserved for root user only, if disk is filled, thus a non-essential application running under root user, can still fill up the left reserved blocks and lead to essential system tasks fail.
- For installing the software (running dietpi-software), yeah root permissions are mostly needed to put systemd units in place, move and chown/chmod the files accordingly etc.
- Using the `dietpi` user (then without sudo permissions) as shared application user would be an alternative, but would break some of the benefits of individual users, as an exploited application allows then at least to access all other installed software, which is bad enough, even without root permissions. If we do this, then really, from my point of view.
- My argumentation was against the `dietpi` user as it is currently used (not for applications, but as an alternative for terminal/SSH login and owner of some of our script/settings/data locations) and against how most distributions by default have such a non-root user (RPi => "pi") for login, which then has sudo permissions without password need. Of course this still allows proper function of ext4 reserved blocks etc. but all this is more towards protecting the user from itself. If someone hacks this user, the password-free sudo access renders all security efforts about root user obsolete. For this reason, if there is no other good argument, I don't see, I would remove `dietpi` and stay with root as only initial unix user. Our end users are as always still free in creating their own user with permissions as they wish.

Fourdee, 2018-07-08:
## Software titles that currently use ```root```:
- 128 MPD | target = ```dietpi``` (required for userdata access/media etc)
- 32 YMPD | target = ```dietpi``` (required for userdata access/media etc)

I'll update this list.

MichaIng, 2018-07-08:
@Fourdee 
What do you think about making `dietpi` a system user only, to run software that needs shared data access? Otherwise as in fact it has root permissions via sudo without support, there is no security benefit in comparison to root. And users, that want to access terminal/SSH not as root, will create their own user anyway, as they do already now.

Alternative:
- Let software with `dietpi_userdata` access need, still run as their own users, but `usermod -a -G dietpi mpd` to add user `mpd` to group `dietpi` to allow dietpi_userdata r/w access:
```
root@VM-Jessie:~# l /mnt/dietpi_userdata/
total 16
drwxrwxr-x 2 dietpi dietpi 4096 May 20 18:59 downloads
drwxrwxr-x 2 dietpi dietpi 4096 May 20 18:59 Music
drwxrwxr-x 2 dietpi dietpi 4096 May 20 18:59 Pictures
drwxrwxr-x 2 dietpi dietpi 4096 May 20 18:59 Video
```

Fourdee, 2018-07-08:
@MichaIng 

> Let software with dietpi_userdata access need, still run as their own users, but usermod -a -G dietpi mpd to add user mpd to group dietpi to allow dietpi_userdata r/w access:

Yep, lets try it 👍

Fourdee, 2018-07-08:
MPD/YMPD: https://github.com/Fourdee/DietPi/commit/89164b0696f4c9102192631607716ee93cb86309

userdeveloper98, 2018-07-09:
@MichaIng  @Fourdee 
Great work guys !! 👍 

I would prefer to be even more specific.
Ex: `minidlna` can have read-only access to media files, because it will stream files but will not change it.
`QbitTorrent ` should have read/write access to files, because it will modify/read files.
etc..

MichaIng, 2018-07-09:
@userdeveloper98 
Jep, good point. If we already have separate software users, then limit their permissions to what is really needed.

Hmm but via groups currently not sure how to achieve.

If `dietpi_userdata/Music` is dietpi:dietpi 640 and software user is in dietpi group, then software can just read
660: software can always write as well
664: everyone can read and just software within dietpi group can write as well, the only solution with above result and since it is "just" Music, public read permissions is properly okay?

So

Fourdee, 2018-07-09:
qBitTorrent requires a local user account with login creds, as it uses:
WebUI\LocalHostAuth=true

Fourdee, 2018-07-09:
Son/rad arr, require home user dir?
```
Jul 09 21:18:30 DietPi mono[30099]: [v2.0.0.5235] NzbDrone.Common.Exceptions.Son
arrStartupException: Sonarr failed to start: Cannot create AppFolder, Access to
the path /home/sonarr/.config/NzbDrone is denied
```

userdeveloper98, 2018-07-10:
@Fourdee 

> Son/rad arr, require home user dir?

We can use the same trick as I did for gogs and Jackett https://github.com/Fourdee/DietPi/pull/1895 

in Service file write:
`WorkingDirectory=/opt/jackett`
`Environment=USER=jackett HOME=/opt/jackett`

You will need to replace `/opt/jackett` with actual install path of the application
It will make them think that the home folder is the folder where the app is installed. Nice and compact.

Fourdee, 2018-07-10:
Sonarr/Radarr/Sickrage patch enable:
```
root@DietPi:~# ls -lha /opt/NzbDrone/
total 9.1M
drwxr-xr-x 3 root root 4.0K Jul 10 15:54 .
drwxr-xr-x 5 root root 4.0K Jul 10 15:56 ..
-rw-r--r-- 1 root root  23K Jul  7 10:14 antlr.runtime.dll
-rw-r--r-- 1 root root  13K Jul  7 10:14 antlr.runtime.dll.mdb
-rw-r--r-- 1 root root 120K Jul  7 10:14 CookComputing.XmlRpcV2.dll
-rw-r--r-- 1 root root  50K Jul  7 10:14 CurlSharp.dll
-rw-r--r-- 1 root root  338 Jul  7 10:14 CurlSharp.dll.config
-rw-r--r-- 1 root root  20K Jul  7 10:14 CurlSharp.dll.mdb
-rw-r--r-- 1 root root 106K Jul  7 10:14 FluentMigrator.dll
-rw-r--r-- 1 root root  62K Jul  7 10:14 FluentMigrator.dll.mdb
-rw-r--r-- 1 root root 252K Jul  7 10:14 FluentMigrator.Runner.dll
-rw-r--r-- 1 root root 119K Jul  7 10:14 FluentMigrator.Runner.dll.mdb
-rw-r--r-- 1 root root 145K Jul  7 10:14 FluentValidation.dll
-rw-r--r-- 1 root root  59K Jul  7 10:14 FluentValidation.dll.mdb
-rw-r--r-- 1 root root  50K Jul  7 10:14 Growl.Connector.dll
-rw-r--r-- 1 root root  17K Jul  7 10:14 Growl.CoreLibrary.dll
-rw-r--r-- 1 root root  20K Jul  7 10:14 Ical.Net.Collections.dll
-rw-r--r-- 1 root root 6.6K Jul  7 10:14 Ical.Net.Collections.dll.mdb
-rw-r--r-- 1 root root 170K Jul  7 10:14 Ical.Net.dll
-rw-r--r-- 1 root root  87K Jul  7 10:14 Ical.Net.dll.mdb
-rw-r--r-- 1 root root 196K Jul  7 10:14 ICSharpCode.SharpZipLib.dll
-rw-r--r-- 1 root root 205K Jul  7 10:14 ImageResizer.dll
-rw-r--r-- 1 root root  15K Jul  7 10:14 LogentriesCore.dll
-rw-r--r-- 1 root root 3.6K Jul  7 10:14 LogentriesCore.dll.mdb
-rw-r--r-- 1 root root  31K Jul  7 10:14 LogentriesNLog.dll
-rw-r--r-- 1 root root  12K Jul  7 10:14 LogentriesNLog.dll.mdb
-rw-r--r-- 1 root root 106K Jul  7 10:14 Marr.Data.dll
-rw-r--r-- 1 root root  55K Jul  7 10:14 Marr.Data.dll.mdb
-rw-r--r-- 1 root root 250K Jul  7 10:14 Microsoft.AspNet.SignalR.Core.dll
-rw-r--r-- 1 root root 117K Jul  7 10:14 Microsoft.AspNet.SignalR.Core.dll.mdb
-rw-r--r-- 1 root root  42K Jul  7 10:14 Microsoft.AspNet.SignalR.Owin.dll
-rw-r--r-- 1 root root  15K Jul  7 10:14 Microsoft.AspNet.SignalR.Owin.dll.mdb
-rw-r--r-- 1 root root  79K Jul  7 10:14 Microsoft.Owin.dll
-rw-r--r-- 1 root root  93K Jul  7 10:14 Microsoft.Owin.Host.HttpListener.dll
-rw-r--r-- 1 root root  64K Jul  7 10:14 Microsoft.Owin.Hosting.dll
-rw-r--r-- 1 root root  48K Jul  7 10:14 MonoTorrent.dll
-rw-r--r-- 1 root root  27K Jul  7 10:14 MonoTorrent.dll.mdb
-rw-r--r-- 1 root root 8.5K Jul  7 10:14 Nancy.Authentication.Basic.dll
-rw-r--r-- 1 root root  14K Jul  7 10:14 Nancy.Authentication.Forms.dll
-rw-r--r-- 1 root root 879K Jul  7 10:14 Nancy.dll
-rw-r--r-- 1 root root 6.0K Jul  7 10:14 Nancy.Owin.dll
-rw-r--r-- 1 root root 437K Jul  7 10:14 Newtonsoft.Json.dll
-rw-r--r-- 1 root root 667K Jul  7 10:14 NLog.dll
-rw-r--r-- 1 root root 404K Jul  7 10:14 NodaTime.dll
-rw-r--r-- 1 root root 234K Jul  7 10:14 NzbDrone.Api.dll
-rw-r--r-- 1 root root  95K Jul  7 10:14 NzbDrone.Api.dll.mdb
-rw-r--r-- 1 root root 232K Jul  7 10:14 NzbDrone.Common.dll
-rw-r--r-- 1 root root 102K Jul  7 10:14 NzbDrone.Common.dll.mdb
-rw-r--r-- 1 root root 1.4M Jul  7 10:14 NzbDrone.Core.dll
-rw-r--r-- 1 root root  367 Jul  7 10:14 NzbDrone.Core.dll.config
-rw-r--r-- 1 root root 605K Jul  7 10:14 NzbDrone.Core.dll.mdb
-rw-r--r-- 1 root root  25K Jul  7 10:14 NzbDrone.exe
-rw-r--r-- 1 root root 1.7K Jul  7 10:14 NzbDrone.exe.config
-rw-r--r-- 1 root root  834 Jul  7 10:14 NzbDrone.exe.mdb
-rw-r--r-- 1 root root  34K Jul  7 10:14 NzbDrone.Host.dll
-rw-r--r-- 1 root root  11K Jul  7 10:14 NzbDrone.Host.dll.mdb
-rw-r--r-- 1 root root  21K Jul  7 10:14 NzbDrone.Mono.dll
-rw-r--r-- 1 root root 7.2K Jul  7 10:14 NzbDrone.Mono.dll.mdb
-rw-r--r-- 1 root root 9.0K Jul  7 10:14 NzbDrone.SignalR.dll
-rw-r--r-- 1 root root 2.8K Jul  7 10:14 NzbDrone.SignalR.dll.mdb
-rw-r--r-- 1 root root  24K Jul  7 10:14 OAuth.dll
-rw-r--r-- 1 root root 7.6K Jul  7 10:14 OAuth.dll.mdb
-rw-r--r-- 1 root root  29K Jul  7 10:14 Org.Mentalis.dll
-rw-r--r-- 1 root root 4.5K Jul  7 10:14 Owin.dll
-rw-r--r-- 1 root root  14K Jul  7 10:14 Prowlin.dll
-rw-r--r-- 1 root root 165K Jul  7 10:14 RestSharp.dll
[... 59 lines cut ...]
-rwx------ 1 root root 110K May 27 21:41 NzbDrone.Common.dll.mdb
-rwx------ 1 root root 1.4M May 27 21:41 NzbDrone.Core.dll
-rwx------ 1 root root  367 May 27 21:41 NzbDrone.Core.dll.config
-rwx------ 1 root root 590K May 27 21:41 NzbDrone.Core.dll.mdb
-rwx------ 1 root root  16K May 27 21:41 NzbDrone.Mono.dll
-rwx------ 1 root root 4.4K May 27 21:41 NzbDrone.Mono.dll.mdb
-rwx------ 1 root root 9.0K May 27 21:41 NzbDrone.SignalR.dll
-rwx------ 1 root root 2.7K May 27 21:41 NzbDrone.SignalR.dll.mdb
drwxr-xr-x 2 root root 4.0K Jul 10 15:56 NzbDrone.Update
-rwx------ 1 root root  24K May 27 21:41 OAuth.dll
-rwx------ 1 root root 7.6K May 27 21:41 OAuth.dll.mdb
-rwx------ 1 root root  29K May 27 21:41 Org.Mentalis.dll
-rwx------ 1 root root 4.5K May 27 21:41 Owin.dll
-rwx------ 1 root root  14K May 27 21:41 Prowlin.dll
-rwx------ 1 root root  89K May 27 21:41 Radarr.exe
-rwx------ 1 root root 1.7K May 27 21:41 Radarr.exe.config
-rwx------ 1 root root  695 May 27 21:41 Radarr.exe.mdb
-rwx------ 1 root root 115K May 27 21:41 Radarr.Host.dll
-rwx------ 1 root root  11K May 27 21:41 Radarr.Host.dll.mdb
-rwx------ 1 root root 165K May 27 21:41 RestSharp.dll
-rwx------ 1 root root  11K May 27 21:41 SocksWebProxy.dll
-rwx------ 1 root root 268K May 27 21:41 System.Data.SQLite.dll
-rwx------ 1 root root 676K May 27 21:41 System.Data.SQLite.xml
drwxr-xr-x 3 root root 4.0K Jul 10 15:56 UI
```

Fourdee, 2018-07-10:
We need to purge mono from ```/tmp``` after its installed.
```
/DietPi/dietpi/func/dietpi-globals: line 1179: echo: write error: No space left on device
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

root@DietPi:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G  1.2G   13G   9% /
devtmpfs        943M     0  943M   0% /dev
tmpfs          1008M     0 1008M   0% /dev/shm
tmpfs          1008M  8.4M  999M   1% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs          1008M     0 1008M   0% /sys/fs/cgroup
tmpfs          1023M 1023M     0 100% /tmp
tmpfs            20M  104K   20M   1% /var/log
tmpfs            10M  1.4M  8.7M  14% /DietPi
/dev/mmcblk1p1   63M  8.6M   55M  14% /boot
root@DietPi:~# dietpi-backup -1; reboot^C
root@DietPi:~# ls -lha /tmp
total 1022M
drwxrwxrwt 13 root root 1.2K Jul 10 16:04 .
drwxr-xr-x 23 root root 4.0K Jun 28 16:27 ..
-rw-r--r--  1 root root   98 Jul  9 19:20 dietpi-available_cpu_freqs
drwxrwxrwt  2 root root   40 Jul  8 17:43 .font-unix
drwxr-xr-x  2  999  997   40 Jul  9 19:24 hsperfdata_airsonic
drwxr-xr-x  2 root root   40 Jul  9 19:28 hsperfdata_root
drwxr-xr-x  2  999  997   40 Jul  9 19:50 hsperfdata_ubooquity
drwxrwxrwt  2 root root   40 Jul  8 17:43 .ICE-unix
drwxrwxrwx  2  999  997   40 Jul  9 19:50 JarClassLoader
-rw-------  1 root root  32M Jul  9 21:18 mono_aot_0DC35Q
```

Fourdee, 2018-07-10:
Hmm:
[image]

Fourdee, 2018-07-10:
rtorrent requires access to:
```
Jul 10 18:47:39 DietPi systemd[29208]: rtorrent.service: Failed at step CHROOT spawning /usr/bin/screen: No such file or directory
```

Fourdee, 2018-07-15:
@MichaIng 

Great work keeping the DietPi ship running while I was away 👍 🥇 Really appreciate it.

My new glasses are on, -10.25 and -9.75 lol. I'll try and get this wrapped today. Once done, all items need install testing again just in-case.

Fourdee, 2018-07-15:
Ok done.

🈯️ Reinstalls of all items + patch testing passed

I'am not going to touch Docker, do not have enough experience with it, or even use it.

closed 2018-07-15 by Fourdee

MichaIng, 2018-07-16:
@Fourdee 
Yey, you resolved the wrong version number issue. I also wanted to open an issue about that since I just faced it again. Leads also to wrong https://dietpi.com/survey statistics, I believe.