MichaIng/DietPi #1878: Dietpi command files are empty after reboot

jimbroze, 2018-06-27:
**Info**
- DietPi version	  | 6.9
- Distro version	  | Stretch
- Kernel version	  | 4.4.77-rockchip-ayafun-136
- SBC device		  | Rock64
- Power supply used | Official Rock64 power supply
- SDcard used		  | SanDisk Ultra 32Gb

**Installed software:**
- Plex
- Samba
- Transmission
- Sickrage
- Couchpotato

- OpenVPN Client (Package installed via apt)

**Additional info**
- External USB3.0 hard drive mounted for use with Samba

**Problem**
Randomly after a reboot, most dietpi commands stop working and inspection reveals that many of the files in /DietPi/dietpi/ have been wiped. All of the files are still there but most are now empty and therefore the commands cannot be run. This has happened several times after fresh installs and appears to be random during a reboot. That is, on all occassions there were many reboots prior to the one that caused the issue with no obvious changes between. I'm planning on starting fresh again but would love to prevent this happening again, if possible

Fourdee, 2018-06-27:
@jimbroze 

Thanks for the report 👍 

I'am 99.9 sure its one of the following:
- Insufficient PSU
- Failing SD card
- Kernel bug/issue with mmc stability.

However, i've been running a Rock64 as a testing device for some time, and, i've not experienced any filesystem issues.

The only thing I can suggest is to:
- Full format the SD card prior to writing the DietPi image: https://github.com/Fourdee/DietPi/issues/769#issuecomment-283030267
- Try another SD card
- Verify voltage output of PSU does not go below 5V under load
- Check ```dmesg``` for any reported errors (```dmesg | grep mmc``` should pick up any filesystem issues)

jimbroze, 2018-06-27:
The power supply is rated to 3A so in theory should be fine.  I have a spare Samsung Evo 64Gb lying about but unfortunately it's a few years old. I can format it and give it a try.
Here is the output from dmesg:

[    0.578017] of_get_named_gpiod_flags: parsed 'gpio' property of node '/sdmmc-                                                                 regulator[0]' - status (0)
[    0.578081] reg-fixed-voltage sdmmc-regulator: Looking up vin-supply from dev                                                                 ice tree
[    0.578345] reg-fixed-voltage sdmmc-regulator: Looking up vin-supply from dev                                                                 ice tree
[    0.578384] reg-fixed-voltage sdmmc-regulator: vcc_sd supplying 3300000uV
[    0.578746] reg-fixed-voltage sdmmc-regulator: Looking up vin-supply from dev                                                                 ice tree
[    2.203724] reg-fixed-voltage sdmmc-regulator: Looking up vin-supply from dev                                                                 ice tree
[    2.206857] reg-fixed-voltage sdmmc-regulator: Looking up vin-supply from dev                                                                 ice tree
[    2.209140] reg-fixed-voltage sdmmc-regulator: Looking up vin-supply from dev                                                                 ice tree
[    2.211385] reg-fixed-voltage sdmmc-regulator: Looking up vin-supply from dev                                                                 ice tree
[    2.218200] vcc18_emmc: 1800 mV
[    2.758322] dwmmc_rockchip ff520000.rksdmmc: IDMAC supports 32-bit address mo                                                                 de.
[    2.772508] dwmmc_rockchip ff520000.rksdmmc: Using internal DMA controller.
[    2.786489] dwmmc_rockchip ff520000.rksdmmc: Version ID is 270a
[    2.800462] dwmmc_rockchip ff520000.rksdmmc: DW MMC controller at irq 37,32 b                                                                 it host data width,256 deep fifo
[    2.814905] dwmmc_rockchip ff520000.rksdmmc: Looking up vmmc-supply from devi                                                                 ce tree
[    2.815032] dwmmc_rockchip ff520000.rksdmmc: Looking up vqmmc-supply from dev                                                                 ice tree
[    2.815577] dwmmc_rockchip ff520000.rksdmmc: GPIO lookup for consumer wp
[    2.815585] dwmmc_rockchip ff520000.rksdmmc: using device tree for GPIO looku                                                                 p
[    2.815595] of_get_named_gpiod_flags: can't parse 'wp-gpios' property of node                                                                  '/rksdmmc@ff520000[0]'
[    2.815602] of_get_named_gpiod_flags: can't parse 'wp-gpio' property of node                                                                  '/rksdmmc@ff520000[0]'
[    2.815608] dwmmc_rockchip ff520000.rksdmmc: using lookup tables for GPIO loo                                                                 kup
[    2.815616] dwmmc_rockchip ff520000.rksdmmc: lookup for GPIO wp failed
[    2.826040] mmc_host mmc0: Bus speed (slot 0) = 400000Hz (slot req 400000Hz,                                                                  actual 400000HZ div = 0)
[    2.849106] dwmmc_rockchip ff520000.rksdmmc: 1 slots initialized
[    2.861915] dwmmc_rockchip ff500000.rksdmmc: IDMAC supports 32-bit address mo                                                                 de.
[    2.879314] dwmmc_rockchip ff500000.rksdmmc: Using internal DMA controller.
[    2.896593] dwmmc_rockchip ff500000.rksdmmc: Version ID is 270a
[    2.913812] dwmmc_rockchip ff500000.rksdmmc: DW MMC controller at irq 38,32 b                                                                 it host data width,256 deep fifo
[    2.931636] dwmmc_rockchip ff500000.rksdmmc: Looking up vmmc-supply from devi                                                                 ce tree
[    2.931816] dwmmc_rockchip ff500000.rksdmmc: Looking up vqmmc-supply from dev                                                                 ice tree
[    2.931900] dwmmc_rockchip ff500000.rksdmmc: GPIO lookup for consumer cd
[    2.931909] dwmmc_rockchip ff500000.rksdmmc: using device tree for GPIO looku                                                                 p
[    2.931932] of_get_named_gpiod_flags: parsed 'cd-gpios' property of node '/rk                                                                 sdmmc@ff500000[0]' - status (0)
[    2.931969] dwmmc_rockchip ff500000.rksdmmc: Got CD GPIO
[    2.946163] dwmmc_rockchip ff500000.rksdmmc: GPIO lookup for consumer wp
[    2.946173] dwmmc_rockchip ff500000.rksdmmc: using device tree for GPIO looku                                                                 p
[    2.946185] of_get_named_gpiod_flags: can't parse 'wp-gpios' property of node                                                                  '/rksdmmc@ff500000[0]'
[    2.946193] of_get_named_gpiod_flags: can't parse 'wp-gpio' property of node                                                                  '/rksdmmc@ff500000[0]'
[    2.946201] dwmmc_rockchip ff500000.rksdmmc: using lookup tables for GPIO loo                                                                 kup
[    2.946211] dwmmc_rockchip ff500000.rksdmmc: lookup for GPIO wp failed
[... cut ...]
[   15.929423] EXT4-fs (mmcblk1p7): mounted filesystem with writeback data mode.                                                                  Opts: (null)
[   21.194922] EXT4-fs (mmcblk1p7): re-mounted. Opts: (null)
[   24.970463] FAT-fs (mmcblk1p6): utf8 is not a recommended IO charset for FAT                                                                  filesystems, filesystem will be case sensitive!
[   25.032539] FAT-fs (mmcblk1p6): Volume was not properly unmounted. Some data                                                                  may be corrupt. Please run fsck.

Fourdee, 2018-06-28:
@jimbroze 

Yep, there is some filesystem failures and corruptions, hence the missing files:
```
[ 15.900360] EXT4-fs (mmcblk1p7): warning: mounting fs with errors, running e2 fsck is recommended
[ 25.032539] FAT-fs (mmcblk1p6): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
```

Any power cuts/unexpected loss of power recently?

--- 

You can try to recover the filesystem, however, it may be quicker and more efficient to simply re-write the DietPi image and start from scratch.

To repair ```/boot```:
- ```umount /dev/mmcblk1p6; fsck.fat -y /dev/mmcblk1p6```
- ```mount -a```

jimbroze, 2018-06-29:
I'll try formatting the SD card, starting fresh and running for a while. As far as I know there haven't been any power loss. Just out of curiosity, does the reset button on a Rock64 perform a graceful reboot or does it essentially cut power?

Fourdee, 2018-06-29:
@jimbroze 

>  Just out of curiosity, does the reset button on a Rock64 perform a graceful reboot or does it essentially cut power?

I'd imagine its a hard reset. Avoid using it during operation.

Fourdee, 2018-06-29:
@jimbroze 

I'll mark as closed as I believe this a hardware related issue (we are unable to replicate). Please reopen if required, and feel free to leave additional posts.

closed 2018-06-29 by Fourdee