MichaIng/DietPi #1879: waiting for "valid' internet connection

unclehowell, 2018-06-27:
https://ibb.co/ndPY9o

Today it just won't connect, says its waiting for a valid internet connection. Internet's fine. Something must have changed your end. Can you help me diagnose this please ?

Fourdee, 2018-06-28:
@unclehowell 

Thanks for the report 👍 

I'am unable to replicate our end. We'll need to debug.

Please mount the SD card using a SD card > USB reader, and set the following in ```/boot/dietpi.txt```
- ```CONFIG_BOOT_WAIT_FOR_NETWORK=1```

Boot the system back up, then run the following commands (paste/picture results please):
```
ip a
ping google.com -c 1
ping raspbian.raspberrypi.org -c 1
ping dietpi.com -c 1
cat /etc/network/interfaces #Please remove any WiFi keys before pasting
```

Fourdee, 2018-06-29:
@unclehowell 

Marking as closed, I believe this is a temp issue with networking, however, please reopen if required.

closed 2018-06-29 by Fourdee