MichaIng/DietPi #1880: DietPi-Software | MineOS installation breaks on npm

MichaIng, 2018-06-28:
## Creating a bug report/issue:
- Reference: https://github.com/Fourdee/DietPi/issues/1875#issuecomment-400653407

#### Required Information:
- DietPi version	| `v6.9`
- Distro version	| `Stretch`
- Kernel version	| `4.9.88`
- SBC device		| VirtualBox + reported as well on RPi3: https://github.com/Fourdee/DietPi/issues/1875#issuecomment-400653407

#### Additional Information (if applicable):
- Software title	| MineOS
- Was the software title installed freshly or updated/migrated? Fresh
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce:

1. Install MineOS

#### Expected behaviour:

- `npm --install` runs through without error.

#### Actual behaviour:

- `npm --install` throws a lot of warnings and finally error messages.

#### Extra details:

<details>
<summary>npm install log</summary>

```
> posix@4.1.2 install /mnt/0e1fef85-e95c-4777-83ea-4339d1d043e5/dietpi_userdata/mineos/minecraft/node_modules/posix
> node-gyp rebuild

make: Entering directory '/mnt/0e1fef85-e95c-4777-83ea-4339d1d043e5/dietpi_userdata/mineos/minecraft/node_modules/posix/build'
  CXX(target) Release/obj.target/posix/src/posix.o
../src/posix.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE node_chroot(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/posix.cc:121:51: warning: ‘v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)’ is deprecated: Use Isolate version [-Wdeprecated-declarations]
     String::Utf8Value dir_path(info[0]->ToString());
                                                   ^
In file included from /root/.node-gyp/10.5.0/include/node/v8.h:26:0,
                 from /root/.node-gyp/10.5.0/include/node/node.h:63,
                 from ../node_modules/nan/nan.h:51,
                 from ../src/posix.cc:1:
/root/.node-gyp/10.5.0/include/node/v8.h:2852:28: note: declared here
                   explicit Utf8Value(Local<v8::Value> obj));
                            ^
/root/.node-gyp/10.5.0/include/node/v8config.h:324:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
../src/posix.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE node_getrlimit(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/posix.cc:177:54: warning: ‘v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)’ is deprecated: Use Isolate version [-Wdeprecated-declarations]
     String::Utf8Value rlimit_name(info[0]->ToString());
                                                      ^
In file included from /root/.node-gyp/10.5.0/include/node/v8.h:26:0,
                 from /root/.node-gyp/10.5.0/include/node/node.h:63,
                 from ../node_modules/nan/nan.h:51,
                 from ../src/posix.cc:1:
/root/.node-gyp/10.5.0/include/node/v8.h:2852:28: note: declared here
                   explicit Utf8Value(Local<v8::Value> obj));
                            ^
/root/.node-gyp/10.5.0/include/node/v8config.h:324:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
../src/posix.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE node_setrlimit(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/posix.cc:217:54: warning: ‘v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)’ is deprecated: Use Isolate version [-Wdeprecated-declarations]
     String::Utf8Value rlimit_name(info[0]->ToString());
                                                      ^
In file included from /root/.node-gyp/10.5.0/include/node/v8.h:26:0,
[... 246 lines cut ...]
npm ERR! userid@0.3.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the userid@0.3.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-06-28T16_55_07_748Z-debug.log
```
</details>
- I switched to Node 8: `npm install -g n && n 8`
- But faced another error message then:
```
root@VM-Stretch:/mnt/dietpi_userdata/mineos/minecraft# npm install --unsafe-perm
npm ERR! path /mnt/0e1fef85-e95c-4777-83ea-4339d1d043e5/dietpi_userdata/mineos/minecraft/node_modules/nyc/node_modules/amdefine
npm ERR! code ENOENT
npm ERR! errno -2
npm ERR! syscall rename
npm ERR! enoent ENOENT: no such file or directory, rename '/mnt/0e1fef85-e95c-4777-83ea-4339d1d043e5/dietpi_userdata/mineos/minecraft/node_modules/nyc/node_modu                                                                             les/amdefine' -> '/mnt/0e1fef85-e95c-4777-83ea-4339d1d043e5/dietpi_userdata/mineos/minecraft/node_modules/nyc/node_modules/.amdefine.DELETE'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-06-28T17_13_05_913Z-debug.log
```

symbios24, 2018-06-28:
hello what about use the installation as described on MineOS github or its already like that?

or just use another method not the supervisor

Thanks

LexiconCode, 2018-07-02:
Reach out to the author `hexparrot` Will Dizon [MineOS Forum](https://discourse.codeemo.com/). He is very approachable and will tweak the source code of MineOS if needed.

MichaIng, 2018-07-17:
Retest on VM Jessie:
<details>
<summary>Terminal output</summary>

```
Cloning into 'minecraft'...
remote: Counting objects: 5803, done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 5803 (delta 1), reused 0 (delta 0), pack-reused 5801
Receiving objects: 100% (5803/5803), 4.95 MiB | 972.00 KiB/s, done.
Resolving deltas: 100% (3407/3407), done.
Checking connectivity... done.

> posix@4.1.2 install /mnt/dietpi_userdata/mineos/minecraft/node_modules/posix
> node-gyp rebuild

make: Entering directory '/mnt/dietpi_userdata/mineos/minecraft/node_modules/posix/build'
  CXX(target) Release/obj.target/posix/src/posix.o
../src/posix.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE node_chroot(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/posix.cc:121:51: warning: ‘v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)’ is deprecated (declared at /root/.node-gyp/10.6.0/include/node/v8.h:2851): Use Isolate version [-Wdeprecated-declarations]
     String::Utf8Value dir_path(info[0]->ToString());
                                                   ^
../src/posix.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE node_getrlimit(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/posix.cc:177:54: warning: ‘v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)’ is deprecated (declared at /root/.node-gyp/10.6.0/include/node/v8.h:2851): Use Isolate version [-Wdeprecated-declarations]
     String::Utf8Value rlimit_name(info[0]->ToString());
                                                      ^
../src/posix.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE node_setrlimit(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/posix.cc:217:54: warning: ‘v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)’ is deprecated (declared at /root/.node-gyp/10.6.0/include/node/v8.h:2851): Use Isolate version [-Wdeprecated-declarations]
     String::Utf8Value rlimit_name(info[0]->ToString());
                                                      ^
../src/posix.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE node_getpwnam(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/posix.cc:288:52: warning: ‘v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)’ is deprecated (declared at /root/.node-gyp/10.6.0/include/node/v8.h:2851): Use Isolate version [-Wdeprecated-declarations]
         String::Utf8Value pwnam(info[0]->ToString());
                                                    ^
../src/posix.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE node_getgrnam(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/posix.cc:333:52: warning: ‘v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)’ is deprecated (declared at /root/.node-gyp/10.6.0/include/node/v8.h:2851): Use Isolate version [-Wdeprecated-declarations]
         String::Utf8Value pwnam(info[0]->ToString());
                                                    ^
../src/posix.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE node_initgroups(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/posix.cc:373:47: warning: ‘v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)’ is deprecated (declared at /root/.node-gyp/10.6.0/include/node/v8.h:2851): Use Isolate version [-Wdeprecated-declarations]
     String::Utf8Value unam(info[0]->ToString());
                                               ^
../src/posix.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE node_openlog(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/posix.cc:449:48: warning: ‘v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)’ is deprecated (declared at /root/.node-gyp/10.6.0/include/node/v8.h:2851): Use Isolate version [-Wdeprecated-declarations]
     String::Utf8Value ident(info[0]->ToString());
                                                ^
../src/posix.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE node_syslog(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/posix.cc:481:50: warning: ‘v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)’ is deprecated (declared at /root/.node-gyp/10.6.0/include/node/v8.h:2851): Use Isolate version [-Wdeprecated-declarations]
     String::Utf8Value message(info[1]->ToString());
                                                  ^
../src/posix.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE node_sethostname(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/posix.cc:591:34: warning: ‘v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)’ is deprecated (declared at /root/.node-gyp/10.6.0/include/node/v8.h:2851): Use Isolate version [-Wdeprecated-declarations]
     String::Utf8Value str(info[0]);
                                  ^
../src/posix.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE node_swapon(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/posix.cc:618:34: warning: ‘v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)’ is deprecated (declared at /root/.node-gyp/10.6.0/include/node/v8.h:2851): Use Isolate version [-Wdeprecated-declarations]
     String::Utf8Value str(info[0]);
                                  ^
../src/posix.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE node_swapoff(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/posix.cc:639:34: warning: ‘v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)’ is deprecated (declared at /root/.node-gyp/10.6.0/include/node/v8.h:2851): Use Isolate version [-Wdeprecated-declarations]
     String::Utf8Value str(info[0]);
                                  ^
  SOLINK_MODULE(target) Release/obj.target/posix.node
  COPY Release/posix.node
make: Leaving directory '/mnt/dietpi_userdata/mineos/minecraft/node_modules/posix/build'

> userid@0.3.1 install /mnt/dietpi_userdata/mineos/minecraft/node_modules/userid
> node-gyp rebuild

[... cut ...]
npm ERR! enoent ENOENT: no such file or directory, rename '/mnt/dietpi_userdata/mineos/minecraft/node_modules/nyc/node_modules/amdefine' -> '/mnt/dietpi_userdata/mineos/minecraft/node_modules/nyc/node_modules/.amdefine.DELETE'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-07-17T16_30_52_204Z-debug.log
```

Using alternative method to install NodeJS via repo:
```
curl -sL https://deb.nodesource.com/setup_8.x | bash -
apt-get -y install nodejs
```
- Fails with same error

MichaIng, 2018-07-17:
Restart from scratch, using official Node v8 install method directly:
```
curl -sL https://deb.nodesource.com/setup_8.x | bash -
apt-get -y install nodejs
```
🈯️ Install succeeds. Above install method creates node binary at expected location, thus `ln -s /usr/bin/node /usr/local/bin/node` is needed or `sed -i '/^command=/c\command=/usr/local/bin/node webui.js' /etc/supervisor/conf.d/mineos.conf` needs to be removed from configuration script.

@Fourdee 
Since this is at least the second software title that needs Node 8 (Koel as well), maybe we should either:
- Fully revert to Node 8
- Offer Node 8 and most current Node version as separate software titles to use most recent version where possible and old stable where needed.
- Reason for second e.g. is that several security warnings show up on npm install with Node 8.

With Java we could btw. do the same, as Java 10 is available.

MichaIng, 2018-07-18:
@Fourdee 
For now maybe it's easier to add Node 8 as new install option and just verify again that it's working (duet to different install method) with Koel.
As long as all other software titles work with the current version/install method, we should not change something there.

We could call the new install NodeJS legacy e.g., then it's clear that it should only be seen as a compatibility fallback.

MichaIng, 2018-07-18:
Okay, actually the Koel solution works for MineOS as well. Node 8 just needs to be installed directly instead of failed Node 10 install attempt 😄.

Will test on all Distros:
🈯️ Jessie
- [ ] Stretch
- [ ] Buster

Fourdee, 2018-07-18:
ARMv7 Stretch, MineOS + NetData:
```
writing new private key to '.tmpkey.pem'
-----
writing RSA key
Enter new UNIX password: Retype new UNIX password: passwd: password updated successfully
cp: cannot stat '/var/games/minecraft/*': No such file or directory
error: <class 'socket.error'>, [Errno 2] No such file or directory: file: /usr/lib/python2.7/socket.py line: 228

root@DietPi:~# ls -lha /var/games/minecraft
lrwxrwxrwx 1 mineos mineos 38 Jul 18 17:46 /var/games/minecraft -> /mnt/dietpi_userdata/mineos/serverdata

```
Doesn't support symlinks?

Ah ```./generate-sslcert.sh``` runs before ```mkdir```

```linux-x86.tar.gz```?
```
/usr/local/bin/n -> /usr/local/lib/node_modules/n/bin/n
+ n@2.1.12
added 1 package from 4 contributors in 1.739s

     install : node-v8.11.3
       mkdir : /usr/local/n/versions/node/8.11.3
       fetch : https://nodejs.org/dist/v8.11.3/node-v8.11.3-linux-x86.tar.gz
######################################################################## 100.0%
/usr/local/bin/n: line 590: /usr/local/bin/node: cannot execute binary file: Exec format error
   installed :

/usr/local/bin/node: 1: /usr/local/bin/node: Syntax error: word unexpected (expecting ")")
```

MichaIng, 2018-07-18:
@Fourdee 
Ah, I was already wondering why the error about missing dir still occurs after pre-creating it. So we move to in front of `./generate-sslcert.sh` 👍.

Strange, but Node 10 is the correct ARMv7 version?

Fourdee, 2018-07-18:
@MichaIng 
> Strange, but Node 10 is the correct ARMv7 version?

Yep:
```
root@DietPi:~# ./node_install.sh
Node Linux Installer by www.github.com/taaem
Running as Root User
Get Latest Version Number...
Done
Downloading latest stable Version node-v10.6.0-linux-armv7l.tar.gz...
```

MichaIng, 2018-07-18:
Okay, retested successfully here. Hmm Node installs/builds stay a bid unreliable, but it's the best we can do?

Fourdee, 2018-07-18:
@MichaIng 

> Okay, retested successfully here. Hmm Node installs/builds stay a bid unreliable, but it's the best we can do?

Yep, outside our control.

Just trying to run MineOS as non-root, will send commit if successful.

DeathIsUnknown, 2019-02-15:
> 🈴 `linux-x86.tar.gz`?
>
> ```
> /usr/local/bin/n -> /usr/local/lib/node_modules/n/bin/n
> + n@2.1.12
> added 1 package from 4 contributors in 1.739s
>
>      install : node-v8.11.3
>        mkdir : [...]

I'm having this same issue when attempting to install MineOS on a RockPro64 (aarch64) with Dietpi  v6.21.1, albeit with node v8.15.0 instead of v8.11.3. Could you please expand on how you were able to get n to install the correct architecture here:

> 🈺 Christ, 2nd test is fine, weird. We have no G_HW_ARCH usage here, so appears the nodejs install script is to blame.
> 
> ```
> [  OK  ] DietPi-Software | G_AGI: python python3 supervisor rdiff-backup screen rsync
> Cloning into 'minecraft'...
> remote: Counting objects: 5803, done.
> remote: Compressing objects: 100% (2/2), done.
> remote: Total 5803 (delta 1), reused 0 (delta 0), pack-reused 5801
> Receiving objects: 100% (5803/5803), 4.95 MiB | 2.65 MiB/s, done.
> Resolving deltas: 100% (3407/3407), done.
> /usr/local/bin/n -> /usr/local/lib/node_modules/n/bin/n
> + n@2.1.12
> added 1 package from 4 contributors in 0.933s
> 
>      install : node-v8.11.3
>        mkdir : /usr/local/n/versions/node/8.11.3
>        fetch : https://nodejs.org/dist/v8.11.3/node-v8.11.3-linux-armv7l.tar.gz
> ```

MichaIng, 2019-02-15:
@DeathIsUnknown 
Hmm Node install, `npm install -g n` and `n 8` is done within the install code already. Not sure if `n -a arm64 8` so setting the architecture was the required step in your case, e.g. for ARMv8 in general?

DeathIsUnknown, 2019-02-15:
@MichaIng 

> so setting the architecture was the required step in your case, e.g. for ARMv8 in general?

I believe so, yes. Even now, if I try `n 11`, I get:
```
root@dietpi:~# n 11

     install : node-v11.10.0

  Error: invalid version 11.10.0
```

According to [this issue's comment](https://github.com/tj/n/issues/517#issuecomment-450586230), this error crops up when n can't find a corresponding pre-built binary from nodejs. However, arm64 binaries are available, and specifying `n -a arm64 11` confirms n is requesting the wrong architecture.
```
root@dietpi:~# n -a arm64 11

     install : node-v11.10.0
       mkdir : /usr/local/n/versions/node/11.10.0
       fetch : https://nodejs.org/dist/v11.10.0/node-v11.10.0-linux-arm64.tar.gz
```

I am assuming that, as Fourdee found in his test, n is attempting to target x86 on my ARMv8 system - unfortunately I don't have the hardware to confirm if it's common across all systems.

MichaIng, 2019-02-15:
@DeathIsUnknown 
> I am assuming that, as Fourdee found in his test, n is still attempting to target x86 (of which Node has dropped Linux support since v10) on my ARMv8 device

Now that you say it I remember this issue. AFAIK it was indeed only for ARMv8 devices, but perhaps to be failsafe we should simply always add the `-a` option.

DeathIsUnknown, 2019-02-17:
Looks like a pull request was made to add support for 64-bit architectures back in 2017, but it still hasn't been approved. https://github.com/tj/n/pull/448

MichaIng, 2019-02-17:
@DeathIsUnknown 
Thanks for linking this issue. If shows very clearly where the issue relies. A pain that the PR is opened for years and already approved by all two reviewers but still not merged. It is so simple and clear that there is not even any "review" required. ARMv8 has simply been forgotten or not yet added ...

closed 2019-02-17 by MichaIng

DeathIsUnknown, 2019-03-29:
Bit of a follow-up, `n` has been updated to support ARM64: https://github.com/tj/n/pull/448#issuecomment-477898698

MichaIng, 2019-03-29:
@DeathIsUnknown 
Many thanks for the information. So we can remove the workaround then, although a quick test should be good:
```
npm i -g n
n 8
```

DeathIsUnknown, 2019-03-31:
@MichaIng My SBC is offline at the moment until I figure out a kernel issue, so I won't be able to test on an ARM64 platform. Sorry!